JSBSim-Team/jsbsim #685: module 'jsbsim' has no attribute 'FGRocket'

IMHinnG, 2022-07-26:
attribute 'FGRocket' non-existent in python

Traceback (most recent call last):
  File "JSBSim_1.py", line 6, in <module>
    jsbsim.FGRocket('..\\')
AttributeError: module 'jsbsim' has no attribute 'FGRocket'

seanmcleod, 2022-07-26:
What are you trying to do? Share the contents of `JSBSim_1.py` or least the relevant sections that are trying to use `FGRocket`.

IMHinnG, 2022-08-05:
> What are you trying to do? Share the contents of `JSBSim_1.py` or least the relevant sections that are trying to use `FGRocket`.

its appears in c++ code, but not in python  of this attribute,

seanmcleod, 2022-08-05:
Not every single C++ class and method is exposed 1 for 1 in the python API.

Hence my question:

> What are you trying to do? Share the contents of `JSBSim_1.py` or least the relevant sections that are trying to use `FGRocket`.

If you share that sort of information we can potentially help in terms of possibly pointing you to an alternative API to achieve the same goal, or if there is a good reason for exposing something via the python API then we can look at that as well.

But we need more information.

IMHinnG, 2022-08-05:
> Not every single C++ class and method is exposed 1 for 1 in the python API.
>
> Hence my question:
>
>
> If you share that sort of information we can potentially help in terms of possibly pointing you to [...]

# utf-8

import jsbsim

jsbsim.FGRocket('..\')

#
that is all

seanmcleod, 2022-08-05:
That's not enough information for me to make any useful comments or suggestions.