JSBSim-Team/jsbsim #687: Aircraft's `Systems` sub-directories are missing in the wheel packages.

bcoconni, 2022-07-27:
### Discussed in https://github.com/JSBSim-Team/jsbsim/discussions/683

<div type='discussions-op-text'>

<sup>Originally posted by **cjdupuis1** July 21, 2022</sup>
I have been having trouble with a lot of the example scripts that come with the download. For example, the Short-S23 is missing the Short_S23-hydrodynamics system file. The ah1s helicopter model is missing files. It seems like only about half of the example scripts run without errors. I can provide more examples if necessary.

Are all of the example scripts supposed to run?</div>

seanmcleod, 2022-07-27:
@bcoconni in addition to aircraft specific `Systems` sub-directories some aircraft also include an `Engines` sub-directory, e.g. the `ah1s` that I checked when I looked into this issue.

bcoconni, 2022-07-27:
Oups ! You're absolutely right @seanmcleod. Thanks for the heads up 👍

seanmcleod, 2022-07-27:
The other thing I noticed last night when looking into this and finding the relevant code in `setup.py.in` was whether `os.path.exists()` etc. are case sensitive or not across Windows and Unix. 

For example we have `aircraft\MB339\systems`, i.e. all lower case for `systems` whereas other aircraft have `\Systems` etc.

So I didn't submit a pull request last night to fix the missing sub-directories for aircraft since I wanted to double-check the case sensitivity issue. But you then pushed a commit, so I'll let you double-check :wink:

In general Windows has a case preserving but case insensitive file system, but in general Unix file systems are case preserving and case sensitive.

dpculp, 2022-07-27:
Some historical background on the capitalization.  The lower case is 
JSBSim-style whereas the upper case is FlightGear-style. Originally the 
aircraft model was to be spread across three directories so that the 
engines and systems could be shared.  In FlightGear it made more sense 
to package the model in a single directory using subdirectories for the 
engines and systems, to ease the distribution of models.  So the loading 
algorithm was changed to look for the subdirectories as well as the 
external directories.

bcoconni, 2022-07-27:
> The other thing I noticed last night when looking into this and finding the relevant code in `setup.py.in` was whether `os.path.exists()` etc. are case sensitive or not across Windows and Unix.

Regarding the case sensitivity topic, I'd say we are in a mixed situation since the default `systems` folder is all lower case:
https://github.com/JSBSim-Team/jsbsim/blob/16bb2705de1cb74d5896e8eb1c25600fda3b5ca6/src/FGFDMExec.cpp#L99-L101

while the `Systems` folder under the `aircraft` has a capital letter :

https://github.com/JSBSim-Team/jsbsim/blob/16bb2705de1cb74d5896e8eb1c25600fda3b5ca6/src/models/FGFCS.cpp#L611-L620

The same issue applies to `engines` vs `Engines`, by the way:

https://github.com/JSBSim-Team/jsbsim/blob/16bb2705de1cb74d5896e8eb1c25600fda3b5ca6/src/models/FGPropulsion.cpp#L439-L448

So strictly speaking the code that has been committed to `master` (commits 3016b7c951987130eb0318e881f4a364690e276f and 16bb2705de1cb74d5896e8eb1c25600fda3b5ca6) is correct 😄. However we might want to allow both `systems` and `Systems` as `aircraft` subdirectories (as well as `engines` and `Engines`) ?

> For example we have `aircraft\MB339\systems`, i.e. all lower case for `systems` whereas other aircraft have `\Systems` etc.

I can't find any `MB339` aircraft in JSBSim, are you referring to an aircraft from FlightGear ?

> So I didn't submit a pull request last night to fix the missing sub-directories for aircraft since I wanted to double-check the case sensitivity issue. But you then pushed a commit, so I'll let you double-check 😉

Yes, sorry. I was so eager to fix my error that I pushed a fix as soon as I could.

seanmcleod, 2022-07-28:
> Yes, sorry.

Definitely no problem :wink:.

> I can't find any MB339 aircraft in JSBSim, are you referring to an aircraft from FlightGear ?

Hmm, just double-checked via `git status` and I see that `MB339` is something I added locally at some stage. Can't remember when or from where. So, good chance it could be from FlightGear.

commit 77d2df24 (2022-08-05): Allow multiple spellings for `Systems` and `Engines` sub-directories. (#690)

Fixes #687.
commit 4530539f (2022-08-20): Allow multiple spellings for `Systems` and `Engines` sub-directories. (#690)

Fixes #687.
closed 2022-08-05 by bcoconni