JSBSim-Team/jsbsim #688: JSBSim Python Module not functioning with M1 chip

NA1384, 2022-07-28:
I am trying to use the gym-jsbsim library and installed JSBSim first as directed by the installation guide in the gym-jsbsim repository. I wanted to test to see if the JSBSim installation was successful, so I ran the example in the readme.md for Python below:

```
import jsbsim

fdm = jsbsim.FGFDMExec(None)  # Use JSBSim default aircraft data.
fdm.load_script('scripts/c1723.xml')
fdm.run_ic()

while fdm.run():
  pass
```

When the program runs, it provides the following error:

```
Traceback (most recent call last):
  File "/Users/user/Documents/Project/Production/main.py", line 1, in <module>
    import jsbsim
ImportError: dlopen(/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/jsbsim.cpython-310-darwin.so, 0x0002): tried: '/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/jsbsim.cpython-310-darwin.so' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e'))

Process finished with exit code 1
```

I was wondering whether there is another version of the module for arm computers that I am unaware of that needs to be installed separately or whether JSBSim doesn't support arm computers. In this sense, I am unsure if this would be a bug or feature request for support for arm computers. Any guidance would be greatly appreciated.

Environment
 - macOS Monterey 12.4
 - MacBook Air (M1, 2020)
 - JSBSim version 1.1.12
 - Python module, installed via PyCharm package manager (uninstalled and tried pip, same issue).

seanmcleod, 2022-07-28:
Hmm, I sort of thought Apple's Rosetta would kick-in to support the execution of the x64 code on M1 Arm CPUs.

I guess the first question is whether you have Rosetta installed? See - https://support.apple.com/en-us/HT211861

But I wonder if it's more complicated in terms of mixed CPU architecture support, i.e. whether in this case the main python executable you have installed and have running is built for Arm and now it's trying to load an x64 shared library. Maybe Rosetta only works for apps that are completely x64? I haven't double-checked.

NA1384, 2022-07-28:
I have Rosetta installed, but the version of PyCharm I downloaded is for the M1 Chip. I could try to use the Intel version of the application and have it run through Rosetta to see if that could fix the issue. I will also try running the file through the terminal and see if the same error occurs. Would you have any other ideas for getting JSBSim to run? 

Thanks for the quick response and suggestions.

NA1384, 2022-07-28:
I ran the program through terminal instead of PyCharm, and thankfully it works flawlessly, so the issue would probably be on how PyCharm runs the program instead of with JSBSim. I am going to check the Interpreter settings in PyCharm as that would probably be the source of the issue.

NA1384, 2022-07-28:
Tried running it in DataSpell, another IDE by the same company, and the program works perfectly here. It seems like this is a PyCharm issue at this point, and I will just use DataSpell instead. If you think there is nothing else to add, I will go ahead and close the issue so other people who run into this issue or something similar know the root cause.

Thanks for your help.

closed 2022-07-28 by NA1384

seanmcleod, 2022-07-28:
Maybe you have two versions of python installed, an x64 version and an ARM version and PyCharm is launching the ARM version whereas DataSpell and the terminal is defaulting to using the x64 version?

NA1384, 2022-07-28:
Looked into it a bit more and found the issue is something like what you described. I checked the version of Python that DataSpell was running, and it was on 3.9, while PyCharm was running 3.10. Changed the python interpreter in PyCharm to 3.8, and now the program works in PyCharm. This means the issue is probably based on something in python 3.10 that is causing the issue.