JSBSim-Team/jsbsim #698: Input sockets are broken on Windows

bcoconni, 2022-08-07:
### Discussed in https://github.com/JSBSim-Team/jsbsim/discussions/693

<div type='discussions-op-text'>

<sup>Originally posted by **achodoglugil** August  4, 2022</sup>
I set the input parameters like this in c1723.xm
[image]

I run the command `.\JSBSim --script=scripts/c1723.xml --realtime --suspend` and then from another terminal i run the command `telnet localhost 1137` and doesnt get the message `connected to jsbsim server`. I just get a blank terminal and cant type anything</div>

bcoconni, 2022-08-07:
As reported by @seanmcleod in https://github.com/JSBSim-Team/jsbsim/discussions/693#discussioncomment-3336344, the issue is due to an improper use of the `FGfdmSocket` members `sckt` and `sckt_in` as variables of the type `SOCKET`on Windows.

seanmcleod, 2022-08-07:
So looking at `socket()` and `accept()` documentation for Linux and presumably for other Unix based OS, it looks like -1 is returned for both to indicate failure.

Linux socket() - https://www.man7.org/linux/man-pages/man2/socket.2.html
Linux accept() - https://www.man7.org/linux/man-pages/man2/accept.2.html

So one option to potentially avoid a bunch of extra `#ifdefs` around every call to `socket()`, `accept()` etc. is to introduce:

```C++
#ifndef WINDOWS
#define INVALID_SOCKET -1 
#endif
``` 

So resulting in the following sorts of changes:

```c++
// Currently
sckt = sckt_in = 0;
// To
sckt = sckt_in = INVALID_SOCKET;

// Currently
  if (sckt) shutdown(sckt,2);
  if (sckt_in) shutdown(sckt_in,2);
// To
  if (sckt != INVALID_SOCKET) shutdown(sckt,2);
  if (sckt_in != INVALID_SOCKET) shutdown(sckt_in,2);

// Currently 
if (sckt_in > 0) {
// To
if (sckt_in != INVALID_SOCKET 0) {

// Currently
  if (sckt_in <= 0 && Protocol == ptTCP) {
// To
  if (sckt_in != INVALID_SOCKET && Protocol == ptTCP) {

// Currently
sckt_in = -1;
// To
sckt_in = INVALID_SOCKET;
```

I also looked into the validity of 0 as a valid file handle on Linux. Looks to me like some of inconsistency in the JSBSim code base in terms of using 0 for some of the comparisons etc.

https://stackoverflow.com/questions/22367920/is-it-possible-that-linux-file-descriptor-0-1-2-not-for-stdin-stdout-and-stderr

Basically 0 by default is the FD for StdIn, but someone could call close on StdIn and a subsequent call to `socket()` could then return 0 as a valid FD for the socket.

bcoconni, 2022-08-08:
Great idea @seanmcleod !
It indeed simplifies much the code (see the commit https://github.com/bcoconni/jsbsim/commit/432a181b6d669c3386ff9ca06bb774f1ce540fb8).

bcoconni, 2022-08-15:
Fixed by the PR #700.

closed 2022-08-15 by bcoconni