JSBSim-Team/jsbsim #711: Flight behaviour issue

ImKogane, 2022-08-29:
**I'm submitting a ...**
  - [X] bug report
  - [ ] feature request
  - [ ] support request => **Please do not submit support request here**, see note at the top of this template.

**Describe the issue**

We're having a disorientation issue after a few minutes of flight, causing the aircraft to have random movements and then crash.

**What is the current behavior?**
 After a few minutes of flight, the aircraft gets disorientated while being way above the ground in a stable state. The strength of this disorientation is not constant, sometimes it doesn't affect much the trajectory of the aircraft, but it also can be very violent and moves the aircraft below the ground level in an instant. Video showing the issue below.

**Please tell us about your environment:**
 - Windows 11 Pro. Ryzen 7 5700G, 48Go RAM, RTX 3080Ti
 - JSBSim version : Unreal Engine v1.01
 - Simulating in mixed reality with a Varjo XR-3
 - Language if applicable : Using Blueprints to interact with JSBSim plugin on Unreal Engine

**Other information**
Vidéo of the issue : https://www.youtube.com/watch?v=qiI9f86q_c0&feature=youtu.be

seanmcleod, 2022-08-29:
Hmm, so based on that video how sure are you that it isn't a simple example of a stall and then spin?

Use JSBSim's logging facilities to log useful state information, e.g. airspeed, attitude, altitude, alpha (angle of attack), beta (angle of sidelip), control positions, g, pitch rate, roll rate, yaw rate etc. Given this sort of state information you can work out what is happening.

ImKogane, 2022-08-29:
Hi,

It is neither a problem of stall or spin, since it happened also during a normal level flight or a simple visual circuit. 
Even if it was a stall or spin, the earth woudnt appear chaotic like that, and unfortunately that happend during a demonstration of our simulator when a General of the french air force previously test pilot tried it! 
i am also a aerodynamics engineer and airline pilot, so I know exactly what parameters to look for when it is aircraft related. here it is just unbelivable what is happening, and I think it is related to the plugin and the way it is implemented. It is why we contacted you to help us figure out the possibilities related to our pb.
Thanks in advance for your consideration and help
Regards
Jeff
link to the issue: https://youtu.be/nRTNc35lMx8

gallonmate, 2022-08-29:
Here's a few things to confirm/check in order to eliminate potential issues:
-check frame rate of the game and see if it's dropping/losing frames before/while issue happens.
-confirm the frame rate is locked
-look at the unreal engine output log and take note of any warnings or errors or anything unusual.

I also suggest you do the following to eliminate any other issues first:
-test your game without VR, without earth model/ground/buildings, without other plugins/features you added etc. This is to verify that the other features aren't causing problems.

My thought is that something else is broken/not configured and is causing frame rate drop after a few minutes. Frame rate drop will introduce instability/jumping in JSBSim,  currently the plugin will run JSB sim steps even when Unreal drops framerate. So this will cause the simulation to "jump" because unreal slowed down, while JSB stays running in real time. This can easily be fixed, as the original plugin v1.0 ran JSB together with Unreal engine, ie Unreal Engine runs one frame, JSB then runs one simulation step. v1.01 was updated to run JSB at 120hz, no matter what Unreal Engine runs at. So if you run Unreal Engine at 10fps or 200fps, JSB still goes 120hz.

seanmcleod, 2022-08-29:
> v1.01 was updated to run JSB at 120hz, no matter what Unreal Engine runs at. So if you run Unreal Engine at 10fps or 200fps, JSB still goes 120hz.

I remember seeing this, so sort of assumed this was the default, but I guess they could be using V1.0. Also a JSBSim log output will confirm what rate JSBSim is being run at by looking at the time column.

dpculp, 2022-08-29:
It looks like the problem is not aerodynamic but rather mathematical.  
The extreme and instant changes in orientation look like an equation is 
"blowing up".  Some factor is going to near infinity.

Does this flight model work OK outside of Unreal?

-- Dave

jeffneosim, 2022-08-30:
Hi guys,

first of all thanks for your contribution to solve our pb, we truly appreciate! 
We are trying to figure out what is wrong with jsbsim with a log of all the parameters, since I do agree that one equation should go near infinity suddenly and then crash the simulator. We have also to understand what is the impact of jsbsim running at 120hz and UE at around 40-60fps. We are using the varjo XR-3, doing mixed reality and using cesium as terrain all at the same time, and trying to get more fps from UE!
We will come back here as soon as we have some more information 
Thanks again guys
Regards
Jeff

seanmcleod, 2022-08-30:
> We have also to understand what is the impact of jsbsim running at 120hz and UE at around 40-60fps. 

There shouldn't really be an issue. The UE screen refreshes should really just be at a half to a third of what JSBSim is running at. Maybe control inputs if they're sampled at the same rate as the UE rendering will only be fed into JSBSim at 40-60Hz which again I don't see that being an issue.

Let's see what the JSBSim logs show.

gallonmate, 2022-08-30:
I may have misdirected about checking for FPS issues, I just finished testing various unsteady and poor FPS conditions in UE5/JSB plugin and could not reproduce any issues. UE5 JSBsim plugin is actually more stable that I thought with bad frame rates. When I initially updated plugin to be a fixed 120hz, I only tested a steady range of FPS and not unsteady FPS. So I am happy to report unsteady and low FPS is not a problem.

It appears you are using custom controller input. Possibly this is the source of strange flight behaviour? As Sean said, logs will help.

ImKogane, 2022-08-31:
Hello.

> So I am happy to report unsteady and low FPS is not a problem.

We see that with our tests and it's a good point to know !

> It appears you are using custom controller input. Possibly this is the source of strange flight behaviour? As Sean said, logs will help.

After your message, we tried with a lot of different controller (simple keyboard, simple xbox controller) without our plugin for custom inputs and the result is the same, the issue continue to appear.

After a long flight sessions with activated logs, we took some screenshots on a video (frame by frame) to see the difference in all datas. And we see a radical difference at the exact issue moment.
There is there screen :

### First flight :
[image]
[image]
[image]

### Second flight : 
[image]
[image]
[image]

Do you have an explaination for this changement ?

Thanks for your help !

seanmcleod, 2022-08-31:
Can you not plot the relevant data from the JSBSim output log showing say from 2s before the issue to 2s after the issue appears? And attach a snippet of the log as well.

It's not ideal in terms of trying to help to try and squint at this sort of text overlaid on the outside world.

[image]

Plus there is a lot of useful information missing, e.g. what are the exact sim times that JSBSim is seeing, is there something going wrong where instead of a dT of 1/120 JSBSim is being fed a massive dT when things go wrong? This would show up in the logs.

But squinting through the screenshots it looks like CAS increases massively and obviously unrealistically, e.g. from 167kt to 3977kt. Again a wild guess without seeing the logs, maybe a sudden massive dT is passed, possibly due to some memory corruption somewhere so that when integrating the acceleration you end up with a massive change in CAS velocity.

ImKogane, 2022-08-31:
We couldn't find a specific logs directory for JSBSim infos, is there any ?
The following pastebin is the end of the logs file of a session where the issue occured. Hoping this will help you :

https://pastebin.com/71MDCfqA

Thanks by advance

seanmcleod, 2022-08-31:
Did you notice the massive increase in CAS between the video frames you posted? Did you notice anything else that I haven't spotted in the video frame data when you analysed them?

> After a long flight sessions with activated logs....

I assumed that meant you had enabled JSBSim output logs. 

> We couldn't find a specific logs directory for JSBSim infos, is there any ?

You have to activate output logging. There are multiple ways, the easiest for you is to probably add an `<output>` element to your aircraft file, configured to output to CSV. For example here is what a CSV output element looks like, from the C172x aircraft:

https://github.com/JSBSim-Team/jsbsim/blob/4195e16f83f41e2b510dbb7062ce3cba733fcdef/aircraft/c172x/c172x.xml#L1197-L1216

In general there are categories of properties you can output, e.g. `<rates>`, or you can go more fine-grained and list individual properties, and you can mix both types.

For example here are a number of individual properties, example from the 737 model aircraft:

https://github.com/JSBSim-Team/jsbsim/blob/4195e16f83f41e2b510dbb7062ce3cba733fcdef/aircraft/737/737.xml#L845-L894

dpculp, 2022-08-31:
I "replied to list", but I don't see the message here (maybe it takes a long time) so I'll repost:

It looks like the problem is that landing gear 1 is on the ground and generated a great force:

 
[image]

dpculp, 2022-08-31:
Maybe an inches/meters units error in the landing gear definition in the configuration file?

seanmcleod, 2022-08-31:
@dpculp well spotted. Visually and looking at the altitude values it appears the aircraft is a fair distance from the ground. However I wonder if there is a bug in their code in terms of feeding in terrain elevation to JSBSim from their terrain data.

> and using cesium as terrain

seanmcleod, 2022-08-31:
Although in Flight 2, none of the 3 screenshots show a WOW for any of the gears at least displayed. Is this because there was a brief WOW in-between the video frames they've shared? Plus does the `(7)` indicate that there are 7 gears and they're only printing out values for 2 of the gears? Which is really why I've been pushing for proper JSBSim output logs rather than haphazard screenshots with difficult to read text etc.

dpculp, 2022-08-31:
I wonder if four of those landing gear are actually structural contact points?  In any case it appears some part of the airplane model is contacting the ground while the airplane is at 300 feet AGL altitude, and this is generating a crash.  

I remember a time when crash detection and subsequent simulation freeze was added to JSBSim, and then removed because one user wanted to animate crashes.  The problem is that without crash detection the model doesn't know it has crashed, so it behaves wildly until some factor reaches NaN.

Likely causes of ground contact are:

1)  Ill-defined contact points (wrong units?)
2)  A problem in the terrain interaction routine with cesium  (@seanmcleod)

seanmcleod, 2022-08-31:
@ImKogane so definitely make sure you log gear data and also elevation data so that we can see whether the issue you're seeing is specifically due to JSBSim thinking that one of the gears is suddenly underground causing a massive spring force. 

Have you written your own ground callback to feed elevation data into JSBSim from the Cesium terrain DEM you're using?

gallonmate, 2022-08-31:
You have multiple serious errors in your Unreal output log that you pasted a link to, it's a surprise the simulation didn't entirely crash.

I would re-read everything from https://dev.epicgames.com/community/learning/tutorials/mmL/a-diy-flight-simulator-tutorial to see if you missed some setting. It's like there's issues with the cesium/level/ue world.

From your logs, I do not think there is any bug in the plugin, and has something to do with cesium and the level/map of your game. Unfortunately I do not use cesium.  

First error of your log is saying that you've gone outside the Unreal Physics World bounds. (has nothing to do with JSBSim, but it could be causing JSBSim to malfunction indirectly) `[2022.08.31-14.23.24:670][980]LogOutputDevice: Error: AABBTree encountered invalid bounds input. Forcing element to global payload. Min: X=-113527076612833200.000 Y=-26845629848152164.000 Z=-155918927914946592.000 Max: X=-113527076612833200.000 Y=-26845629848152156.000 Z=-155918927914946592.000. If Bounds are valid but large, increase FAABBTreeCVars::MaxNonGlobalElementBoundsExtrema.

You can try finding a setting for physics world bounds or searching for some solution, and see if a larger physics bounds will solve your problems. But to be honest why isn't cesium solving this for you? I would imagine there is some kind of orgin re-basing that should be taking place. Anyways try fixing the physics problem and see if your issue goes away.
`

ImKogane, 2022-09-01:
Hello,

> I would post to the Cesium community forum and ask if anyone has issues with UE5 and physics going out of bounds

Thank you for your answers, we will soon be in touch with the Cesium team. However, we tested to fly in an empty level, without Cesium, and the issue still occurs.

**More news :** After testing, we currently have 2 types of problems : 
- One giving the aircraft some random rotations/speed for a brief period, sometimes minor enough to take back the control of the aircraft (with normal fly infos values). 
- Another one causing the entier project to crash suddenly, with the extreme log values you saw earlier _(with the WOW of gears before crash)_.

We use to try with the C172p or c182 aircrafts, ande we have the problem. But when we launched with the 737 *(in a Cesium level)*, we could fly for more than an hour without any physics/stability issue. Maybe the problem is related to the Cessna files ?

> You have to activate output logging.

Sadly we couldn't find any C172 or C182 logs in our system after removing the commentary in the xml, is there another parameter to enable or a very specific path to find the files ?

_To precise, we didn't touch the JSBSIM source code. We use the native code for Unreal provided on the repository._

Hope this will help, and thank you again by advance.

seanmcleod, 2022-09-01:
> specific path to find the files

I don't think JSBSim does anything specific to the path, so I'm guessing it will be in the `current` directory for your process. You can double-check by setting a breakpoint in `FGOutputTextFile` etc.

ImKogane, 2022-09-01:
Okay, we found the logs and we can show it ! _(but we don't understand a lot of things in it to be honest)_

[UEReferenceApp.log](https://github.com/JSBSim-Team/jsbsim/files/9471343/UEReferenceApp.log) 
Pastebin if you want : https://pastebin.com/9hSjyV1m

_There are the last 2 seconds of the simulation, just when the problem changes the behavior and makes the plane crash._

With the logs, I hope this time, it will help.

seanmcleod, 2022-09-01:
> Okay, we found the logs.... 

Although what you've found isn't the CSV file that the `output` element should be creating. What you've found is some log generated by the UE plug-in code. This is very limited compared to what can be output via the `output` element, it only reports the position, attitude, translation velocity and angular velocity.

dpculp, 2022-09-01:
> We use to try with the C172p or c182 aircrafts, ande we have the problem. But when we launched with the 737 _(in a Cesium level)_, we could fly for more than an hour without any physics/stability issue. Maybe the problem is related to the Cessna files ?

I think the 737 is flying without problems because the landing gear is retracted.  Retracted landing gear don't interact with the terrain elevation.  Try flying it with the gear down.

gallonmate, 2022-09-02:
I can't reproduce any issues with the c172p in UE5 while using the original demo map from the plugin. I am using all default settings, nothing change from demo map and plane. I use the demo's BP Airliner, type in c172p in the level editor, hit play, and fly for many minutes.

ImKogane, 2022-09-02:
> I think the 737 is flying without problems because the landing gear is retracted. Retracted landing gear don't interact with the terrain elevation. Try flying it with the gear down.

We fly for many inutes with the 737 and the gear down and everything is great, the probleme not come back.
Actually it's just with the Cessna aircrafts.

> I can't reproduce any issues with the c172p in UE5 while using the original demo map from the plugin. I am using all default settings, nothing change from demo map and plane.

We will try to take the basic demo map to try this in our side too.

Thanks in advance.

seanmcleod, 2022-09-02:
Add some code around this line in `FGLGear.cpp` to conditionally break if the force calculated for the gear is greater than some fairly large magnitude. 

https://github.com/JSBSim-Team/jsbsim/blob/375f5be0fdfeb210f4cf66f938e3e6fd426c2966/src/models/FGLGear.cpp#L418

As @dpculp noticed there was an extremely large force being generated by one of the gears. So when the breakpoint is hit then you can inspect the code up and down the stack to try and figure out why at this sort of altitude you're getting contact. In particular what is the ground elevation at that point in time in JSBSim?

[image]

seanmcleod, 2022-09-02:
Also log these properties in the `output` element:

```xml
        <property> position/terrain-elevation-asl-ft </property>
        <property> position/h-agl-ft </property>
        <property> position/h-sl-ft </property>
```

ImKogane, 2022-09-02:
>Add some code around this line in FGLGear.cpp to conditionally break if the force calculated for the gear is greater than some fairly large magnitude.

We found where is the problem with the JSBSIM Logs in our project. It's because all files in 
`...\Plugins\JSBSimFlightDynamicsModel\Source\ThirdParty\JSBSim\Include\input_output` are ignored in the compilation and so, they never be called in the flight process.
We will find how to add and compile this and after we can normaly have JSBSIM logs.

jeffneosim, 2022-09-07:
Hi Guys,

I am coming back to you after a few days of break, since the 2 students working on the project have to go back home to 2 weeks. In our exchange with Epic Game, They mentioned that the pb didnt appear with a B737. I decided to give it a shot and effectively the simulator worked perfectly, keeping the 3D model of the Cessna 172. I confess that I really enjoyed flying a Cessna at 250 kts at 3000 feet :-) I did 3 Visual circuits and absolutely no pb at all. 
Tonight I am going to Study in détail the différences between the 2 flight modele, since it worked also perfectly with the piper cub. 
You all did a freaking good job in the development of JSBSIM, which impresses me a LOT! for an Airline Pilot, I can tell you that I love it sooo much that it is difficult to stop flying every time I try the simulator :-)

I will be back to you soon to tell you if I discovered something.

Again thanks to all of you for your gréat support and your amazing job! 

Regards
Jeff

seanmcleod, 2022-09-07:
If you can reliably reproduce the issue with the C-172 model then I suggest you run that and set the breakpoint I suggested above and get the logging working. We'll then be able to have a good chance to confirm what is causing the issue.

At the moment we only really have 1 clue in terms of recorded data and that's noticing the extremely massive force being generated by the gear/contact point in one of the video screenshots. So again if it's easily reproducible in the C-172 then the first thing to confirm is whether in each case when you reproduce the issue there is always a massive force from the gear involved. That will immediately narrow things down.

jeffneosim, 2022-09-07:
My guys activated the log of jsbsim, it should be in a .csv file, but unfortunately nothing is logged out!!! I think there is a pb with the jsbsim log when it is integrated into UE, and so far we cant figure out why. It is why they created their own logg to get at least something out. 

I will dig into it those days guys.

 

Thanks again.

 

Regards

Jeff

Reply to: JSBSim-Team/jsbsim ***@***.***>
Date: Wednesday 7 September 2022 at 20:06
To: JSBSim-Team/jsbsim ***@***.***>
Cc: jeffneosim ***@***.***>, Comment ***@***.***>

seanmcleod, 2022-09-07:
What about trying the breakpoint as I suggested, independent of any logging issues.

What does **pb** stand for?

jeffneosim, 2022-09-07:
Pb stands for problem.
They also tried the break point they realised that the code never reached
it!!!!!!

seanmcleod, 2022-09-07:
Just to confirm, are you saying that the breakpoint for the gear force calculation is never hit?

Or are you referring to breakpoints that were placed to figure out what was going wrong with the logging?

jeffneosim, 2022-09-07:
As far as I understood, because I was back from a flight before they left, it was the breakpoints that were placed to figure out what was going wrong with the logging. 

Date: Wednesday 7 September 2022 at 20:51

seanmcleod, 2022-09-07:
Yep, but the breakpoint I've always been referring to is the one I suggested you place in `FGLGear.cpp` as mentioned above.

jeffneosim, 2022-09-07:
I will try on my side. I am trying other aircrafts tonight and will give a shot for the breakpoint again later. 

Date: Wednesday 7 September 2022 at 21:00

seanmcleod, 2022-09-07:
Why try a bunch of other aircraft when you have a smoking gun in the sense that you said the C-172 exhibits the issue pretty reliably?

jeffneosim, 2022-09-08:
Good morning,

 

The issue is now known, it comes from the landing gear. So far I checked the differences between the xml files of the c172p and others which are working and I couldn’t find out the geometry definition which generates the crash.  I am on it but unfortunately I have only few time to do that, having a lot to do as the ceo of my startup plus my job as airline pilot, and the developers being on break for 2 weeks…it is why it is slow. 

Date: Wednesday 7 September 2022 at 21:35

seanmcleod, 2022-09-08:
Yep, so set the breakpoint for when the landing gear generates a massive force and then when it's hit you can start looking at the other state, things like ground elevation etc. to try and understand why this is happening.

jeffneosim, 2022-09-08:
Ok thanks

Date: Thursday 8 September 2022 at 08:24

gallonmate, 2022-09-08:
I already mentioned that I could find no bug in the default demo level/plane that comes with the plugin. Only change I made was set aircraft to c172p, and found no issues. The only way I could get jsbsim plugin to crash was to fly 12,000 km away from geo origin. 

-Check yourself that the default demo level/demo plane - with only changing aircraft name to c172p, has no bugs.
-Then Compare the differences from your level/plane setup to the demo level/demo plane blueprint and you will find your problem.

seanmcleod, 2022-09-10:
From @jeffneosim 

> Hi guys, I tried many other aircrafts since it is the fastest and easiest way to know if it is only the FDM of Cessna 172p or jsbsim which is not working. The only aircrafts always crashing are the Cessna!!! I tried the J3Cub and it is wonderful! Very stable in its flight behavior and such a pleasure to fly! Unfortunately too slow when you are flying an A330 in real life 😊I tried the pc7, works good but there is a lack of dampening in the roll mode when I release the yoke, very annoying and difficult to spot out why. I have to study JSBSIM completely and figure out what are the fsc parameters. I tried to get the Cessna 172p from jsbsim GitHub, working with FG, but it didn’t work so far. Need to spend more time on that.  So my conclusion is:JSBSIM is working perfectly with UE 5.0The crash is only due to the FDM of the Cessna models, Cesium has not impact on JSBSIM. There is no pb of altitude in our simulator, otherwise it would have happened with all the models.  Thanks guys Regards Jeff

seanmcleod, 2022-09-10:
@jeffneosim JSBSim isn't closed source like Microsoft Flight Simulator for example, where your only option is to try and narrow things down by changing aircraft etc. and run them against the 'black box' to try and understand an issue. JSBSim is open source so you can set a breakpoint and then inspect the state etc. to really understand what is going wrong.

So I'll recommend it again 😉 set a breakpoint and let's see what we can deduce from the state you observe at the time.

jeffneosim, 2022-09-10:
I do understand that but as an airline pilot + ceo I do not have time for that and will let my students work on that in 2 weeks. As a pilot I can tell if an aircraft is flying correctly and as per the real feeling, I can also understand and work out all the flight mechanics parameters, but flying and trying is faster for me, and we have very very few time to find a solution because we have to move on about mixed reality and hand-tracking integration, the whole thing should be on a 6 DOF working in 6 months!!!! 

Date: Saturday 10 September 2022 at 13:51
To: JSBSim-Team/jsbsim ***@***.***>
Cc: jeffneosim ***@***.***>, Mention ***@***.***>

gallonmate, 2022-09-10:
When your students get back, maybe they can tell what settings were changed from the demo level/demo plane. Because I still haven't been able to replicate this "bug". Also to fix aircraft output logging, your students can edit the c172p aircraft file by just adding a folder location similar to this:
```
   <output name="F:\Downloads\JSBout172C.csv" type="CSV" rate="60">
      <property> aero/qbar-psf </property>
      <property> accelerations/udot-ft_sec2 </property>
      <property> accelerations/vdot-ft_sec2 </property>
      <property> accelerations/wdot-ft_sec2 </property>
      <property> accelerations/a-pilot-x-ft_sec2 </property>
      <property> accelerations/a-pilot-y-ft_sec2 </property>
      <property> accelerations/a-pilot-z-ft_sec2 </property>
      <property> accelerations/n-pilot-x-norm    </property>
      <property> accelerations/n-pilot-y-norm    </property>
      <property> accelerations/n-pilot-z-norm    </property>
      <rates> ON </rates>
      <velocities> ON </velocities>
      <forces> ON </forces>
      <moments> ON </moments>
      <position> ON </position>
      <fcs> OFF </fcs>
      <propulsion> OFF </propulsion>
      <aerosurfaces> OFF </aerosurfaces>
      <fcs> OFF </fcs>
      <ground_reactions> OFF </ground_reactions>
    </output>

```

seanmcleod, 2022-09-10:
@gallonmate after I mentioned how to go about adding an `<output>` element earlier in this thread to help with debugging they came back and said that they noticed an issue with the output code in their build, see - https://github.com/JSBSim-Team/jsbsim/issues/711#issuecomment-1235284201. Which I thought might be related to the issue with `FGOutput` not having a `JSBSIM_API` decoration.

Can you confirm that in your builds that output logging via an `output` element works?

gallonmate, 2022-09-10:
@seanmcleod Yes I can confirm a nice csv file is generated. But a real folder location needs to be specified.  (I actually added & built bcoconni suggested fix first but that didn't work, that's when I tried with a folder location and that did the trick!)

seanmcleod, 2022-09-10:
I'm pretty sure what is happening if you don't specify a full path is that JSBSim is trying to create the file in the current directory for the process. So either the current directory is some unusual directory that you haven't searched when looking for the .csv file, or the current directory is ending up as a read-only directory, e.g. maybe `\windows`.

jeffneosim, 2022-09-10:
Hi Guys,

i have finally the Jsbsim log. As advise, I specified the folder  and then done it worked. I had searched before for the csv file all over the computer and didnt find anything. now all good. Here we go, the last line is when it crashed: 
[JSBout172C.csv](https://github.com/JSBSim-Team/jsbsim/files/9540665/JSBout172C.csv)

seanmcleod, 2022-09-10:
Just had a quick glance, will look in more detail a bit later.

There is some gear issue at time 169.0667.

[image]

jeffneosim, 2022-09-10:
yes. and exploring it I saw also that at 34.9s the flap position is increasing without using it!
this log: 
[JSBout172C.csv](https://github.com/JSBSim-Team/jsbsim/files/9540738/JSBout172C.csv)

dpculp, 2022-09-10:
@jeffneosim  Can you post the configuration XML file for the broken model here?  We might be able to spot the problem.

seanmcleod, 2022-09-10:
So in terms of the gear compression:

https://github.com/JSBSim-Team/jsbsim/blob/64560ab422eb446eca17b07638d57bb5e1025ea9/src/models/FGLGear.h#L96-L113

Hence my assumption that there may be a bug/issue with the terrain elevation. However according to the logfile there doesn't appear to be a sudden jump in the terrain elevation.

The compression is calculated here:

https://github.com/JSBSim-Team/jsbsim/blob/64560ab422eb446eca17b07638d57bb5e1025ea9/src/models/FGLGear.cpp#L328-L333

Where `height` is calculated a little bit before in the method:

https://github.com/JSBSim-Team/jsbsim/blob/64560ab422eb446eca17b07638d57bb5e1025ea9/src/models/FGLGear.cpp#L298-L302

And for the compression to be calculated the `height` has to be negative:

https://github.com/JSBSim-Team/jsbsim/blob/64560ab422eb446eca17b07638d57bb5e1025ea9/src/models/FGLGear.cpp#L317-L318

And here is the default implementation that gets called which returns the `height`.

https://github.com/JSBSim-Team/jsbsim/blob/64560ab422eb446eca17b07638d57bb5e1025ea9/src/input_output/FGGroundCallback.cpp#L41-L57

seanmcleod, 2022-09-10:
So another breakpoint suggestion. Modify the code in `FGGroundCallback.cpp` so that you can set a breakpoint when `l.GetGeodAltitude() - mTerrainElevation < 0`. 

So fly around at altitude, then set the breakpoint, and then when the breakpoint is hit you can check a couple of things.

What does `loc` which is the gear's location look like? Does it look corrupted/way off compared to what it should be and compared to the other gears?

Also double-check the `latitude` and 'longitude` returned to make sure they make sense.

Then what does `l.GetGeodAltitude()` return?

Lastly what is `mTerrainElevation`.

jeffneosim, 2022-09-10:
the xml file: 
[c172p.zip](https://github.com/JSBSim-Team/jsbsim/files/9541274/c172p.zip)

jeffneosim, 2022-09-10:
I have to see with the breakpoints also

dpculp, 2022-09-10:
> the xml file: [c172p.zip](https://github.com/JSBSim-Team/jsbsim/files/9541274/c172p.zip)

I see that this C-172 model has a complex set of ground interactions including hydrodynamic forces from floats.  It's possible that this complexity works well within the FlightGear environment, but operating the model outside of FlightGear may cause some variable to not be set or controlled properly.  I don't know that this is definitely the case, but I would be wary of using this model  without first examining all the supporting systems, how they are initialized and controlled, and how they interact with terrain provided by the simulation.  That would take a long time.  I think it may be better for you to avoid this model for now.

jeffneosim, 2022-09-11:
Hi,

As for the Cessna model, I deleted all parts related to ground interaction excepted the main gear. I tried many models but ONLY the Cessna crashed!!!!! 
That is just unbelievable and I will hire a student in aero just for jsbsim to find out why it behaved this way and also to design our next model, a Diamond DA42. 
I can use an other model to replace the Cessna, having the same behavior, no worry, as it is for our POC, but I have to also figure out the very slow damping in the roll mode for any model I used. They work gréât, especially the pc7, the t37, but as soon as you release the Yoke, they tend to oscillate ( bit of Dutch roll) before keeping the heading. I looked at the coefficients for roll and it didnt change a lot.....

[8 more replies]

closed 2022-12-27 by bcoconni