JSBSim-Team/jsbsim #727: calculate wind effect on UAV using JSBSIM

sorgan191, 2022-09-19:
I am working on generate path planning for UAVs, I have a data set containing wind direction and speed. I want to know how much the wind will shift the drone from its path (meters) using JSBSIM.
Note that I am a very new user of JSBSIM.

thank you in advance

seanmcleod, 2022-09-19:
So are you saying that the drone won't adjust it's heading to follow the pre-planned route (course) when it detects via GPS etc. that it is drifting off-route?

Here is an example of the sort of heading change required given the wind vector, the aircraft's airspeed in order to fly the required course.

[image]

If the UAV isn't going to change it's heading then it will drift over time.

[image]

The calculation is basic vector maths.

seanmcleod, 2022-09-19:
So, you want to fly the UAV in JSBSim following some heading commands etc. and input a wind vector and then you want to track the 3D position and the acceleration in each axis?

If so, that's fairly easy to do in JSBSim since there are logging facilities, e.g. see the `<output>` element that will allow you to log lots of different state information including 3D position, accelerations etc.

bcoconni, 2022-09-21:
This topic is neither a bug nor a feature request.
This is transferred to "Discussions".

closed 2022-09-21 by bcoconni