JSBSim-Team/jsbsim #740: Unreal 5, crash on play in editor instantly with default editor starting map.

nonlin, 2022-09-30:
**I'm submitting a ...**
  - [X ] bug report
  - [ ] feature request
  - [ ] support request => **Please do not submit support request here**, see note at the top of this template.

**Describe the issue**
Getting crash on play

**What is the current behavior?**
Fresh clone, build solution, dll made, build unreal project, launch, play in editor, instant crash

**What is the expected behavior?**
Play in editor doesn't cause a crash 

**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

**Other information**
Stack trace 

Unhandled Exception: 0x80000003

```
ucrtbased
ucrtbased
ucrtbased
ucrtbased
JSBSim
JSBSim
JSBSim
JSBSim
UnrealEditor_JSBSimFlightDynamicsModel!UJSBSimMovementComponent::InitializeJSBSim() [J:\Git\jsbsim\UnrealEngine\Plugins\JSBSimFlightDynamicsModel\Source\JSBSimFlightDynamicsModel\Private\JSBSimMovementComponent.cpp:410]
UnrealEditor_JSBSimFlightDynamicsModel!UJSBSimMovementComponent::LoadAircraft() [J:\Git\jsbsim\UnrealEngine\Plugins\JSBSimFlightDynamicsModel\Source\JSBSimFlightDynamicsModel\Private\JSBSimMovementComponent.cpp:151]
UnrealEditor_JSBSimFlightDynamicsModel!UJSBSimMovementComponent::BeginPlay() [J:\Git\jsbsim\UnrealEngine\Plugins\JSBSimFlightDynamicsModel\Source\JSBSimFlightDynamicsModel\Private\JSBSimMovementComponent.cpp:351]
UnrealEditor_Engine!AActor::BeginPlay() [D:\build\++UE5\Sync\Engine\Source\Runtime\Engine\Private\Actor.cpp:3887]
UnrealEditor_Engine!AActor::DispatchBeginPlay() [D:\build\++UE5\Sync\Engine\Source\Runtime\Engine\Private\Actor.cpp:3849]
UnrealEditor_Engine!AWorldSettings::NotifyBeginPlay() [D:\build\++UE5\Sync\Engine\Source\Runtime\Engine\Private\WorldSettings.cpp:283]
UnrealEditor_Engine!AGameStateBase::HandleBeginPlay() [D:\build\++UE5\Sync\Engine\Source\Runtime\Engine\Private\GameStateBase.cpp:205]
UnrealEditor_Engine!UWorld::BeginPlay() [D:\build\++UE5\Sync\Engine\Source\Runtime\Engine\Private\World.cpp:4908]
UnrealEditor_Engine!UGameInstance::StartPlayInEditorGameInstance() [D:\build\++UE5\Sync\Engine\Source\Runtime\Engine\Private\GameInstance.cpp:500]
UnrealEditor_UnrealEd!UEditorEngine::CreateInnerProcessPIEGameInstance() [D:\build\++UE5\Sync\Engine\Source\Editor\UnrealEd\Private\PlayLevel.cpp:3052]
UnrealEditor_UnrealEd!UEditorEngine::OnLoginPIEComplete_Deferred() [D:\build\++UE5\Sync\Engine\Source\Editor\UnrealEd\Private\PlayLevel.cpp:1568]
UnrealEditor_UnrealEd!UEditorEngine::CreateNewPlayInEditorInstance() [D:\build\++UE5\Sync\Engine\Source\Editor\UnrealEd\Private\PlayLevel.cpp:1830]
UnrealEditor_UnrealEd!UEditorEngine::StartPlayInEditorSession() [D:\build\++UE5\Sync\Engine\Source\Editor\UnrealEd\Private\PlayLevel.cpp:2798]
UnrealEditor_UnrealEd!UEditorEngine::StartQueuedPlaySessionRequestImpl() [D:\build\++UE5\Sync\Engine\Source\Editor\UnrealEd\Private\PlayLevel.cpp:1148]
UnrealEditor_UnrealEd!UEditorEngine::StartQueuedPlaySessionRequest() [D:\build\++UE5\Sync\Engine\Source\Editor\UnrealEd\Private\PlayLevel.cpp:1051]
UnrealEditor_UnrealEd!UEditorEngine::Tick() [D:\build\++UE5\Sync\Engine\Source\Editor\UnrealEd\Private\EditorEngine.cpp:1665]
UnrealEditor_UnrealEd!UUnrealEdEngine::Tick() [D:\build\++UE5\Sync\Engine\Source\Editor\UnrealEd\Private\UnrealEdEngine.cpp:474]
UnrealEditor!FEngineLoop::Tick() [D:\build\++UE5\Sync\Engine\Source\Runtime\Launch\Private\LaunchEngineLoop.cpp:5215]
UnrealEditor!GuardedMain() [D:\build\++UE5\Sync\Engine\Source\Runtime\Launch\Private\Launch.cpp:183]
UnrealEditor!GuardedMainWrapper() [D:\build\++UE5\Sync\Engine\Source\Runtime\Launch\Private\Windows\LaunchWindows.cpp:147]
UnrealEditor!LaunchWindowsStartup() [D:\build\++UE5\Sync\Engine\Source\Runtime\Launch\Private\Windows\LaunchWindows.cpp:283]
UnrealEditor!WinMain() [D:\build\++UE5\Sync\Engine\Source\Runtime\Launch\Private\Windows\LaunchWindows.cpp:330]
UnrealEditor!__scrt_common_main_seh() [d:\a01\_work\6\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288]
kernel32
ntdll
```

gallonmate, 2022-10-01:
Can you share the last/newest Log file after a crash? It's in project folder Saved/Logs.

nonlin, 2022-10-01:
@gallonmate same crash even with the release dll.

seanmcleod, 2022-10-01:
> Unhandled Exception: 0x80000003

Hmm, that's the breakpoint exception, which is typically used when a debugger sets breakpoints, which the debugger will then handle when the exception is raised. It will also be raised by a call to the DebugBreak() Windows API, so there may be some conditional code that is calling DebugBreak() under certain conditions, and since you don't have a debugger attached to handle the breakpoint exception the process exits.

nonlin, 2022-10-01:
Solved by cleaning project, regenerating project files, building solution all while making sure I had the proper released DLL compiled for use inside thirdparty.

closed 2022-10-01 by nonlin