JSBSim-Team/jsbsim #759: Unit conversion round off can prevent tanks from being emptied.

bcoconni, 2022-11-06:
**I'm submitting a ...**
  - [x] bug report
  - [ ] feature request
  - [ ] support request => **Please do not submit support request here**, see note at the top of this template.

### Describe the issue

As reported in the issue c172p-team/c172p#1408 the engine of the c172p did not run out of fuel during a "long haul" flight while it should have.

### What is the current behavior?

The amount of fuel was stuck to a value slightly higher than the unusable amount of fuel (i.e. 1.500000000000001 vs 1.50000000000000). Hence the tank never emptied and the engine did not stop.

### What is the expected behavior?

This is most likely due to a conversion between the volume of unusable fuel (in gal) and the current content of fuel (in lbs).

As shown in the code below (lines 275-276), the amount of fuel in a tank can never go below the unusable amount of fuel (in lbs). If the conversion of the variable `Contents` from *lbs* to *gal* rounds the value to a volume slightly higher than the unusable volume then the condition `remaining < unusable` is never reached, the tank never empties and the engine never runs out of fuel.

https://github.com/JSBSim-Team/jsbsim/blob/10d0ca9e248d3a8ced7a5b5c065d8a52a281945c/src/models/propulsion/FGTank.cpp#L268-L285

### Suggested fix

When the unusable content is reached, narrow down the content to a value slightly lower than the unusable amount:

```diff
    if (Contents > GetUnusable())
-      Contents = GetUnusable();
+      Contents = 0.9999*GetUnusable();
```

bcoconni, 2022-11-06:
@dany93, @wlbragg, @ wkitty42
As discussed, the issue is now open.
Please comment if I did not convey our discussion correctly.

bcoconni, 2022-11-06:
For the record, the feature for the unusable amount of fuel was introduced following the feature request #141 (see commits b8f4a4c5ef6d8d540c8ed7a530f92c54602bf04c and c8c5c6e698ccd368d4b29cccdcc46678b1c101fa).

wkitty42, 2022-11-06:
it looks to me like you described the issue the c172p is seeing just fine, @bcoconni...

gallonmate, 2022-11-07:
I think the suggested fix is a quick and easy solution! But I'm confused. Isn't the problem somewhere else? (Unfortunately my debugger is broke other I'd just trace and watch the numbers myself.) Following the code manually:

Drain says:
```
   if (Contents > GetUnusable()) 
       Contents = GetUnusable();
```
Which means if 1.50001 > 1.5, then contents will become equal to unusable. So fuel should be starved according to this:

[FGPropulsion.cpp](https://github.com/JSBSim-Team/jsbsim/blob/77d2df249829796a7688a5cfbdf03d970461967e/src/models/FGPropulsion.cpp)
```   
 if ((Tank->GetContents() > Tank->GetUnusable()) && Tank->GetSelected() &&....

            Starved = false;
```

dany93, 2022-11-08:
@gallonmate wrote
> Drain says:
> 
> ```
>    if (Contents > GetUnusable()) 
>        Contents = GetUnusable();
> ```
> 
> Which means if 1.50001 > 1.5, then contents will become equal to unusable. So fuel should be starved according to this:

I was asking myself the same question, but I didn't dare taking into account my almost ignorance of C++.

@bcoconni 
Please don't merge your fix too fast.

As I already wrote in https://github.com/c172p-team/c172p/issues/1408#issuecomment-1304814674, I"m more and more suspecting a conflict between JSBSim stopping the fuel from the main tank and the switch in the float chamber which is expected to do the same thing. The first which reaches the limit wins the game. The 'first one' depends on some last digit.

I'm still having to do more tests, but I feel that the way would be to use the float chamber, with its switches and own limits to do the job, totally independently from the JSBSim unusable fuel feature.

If I'm correct, @bcoconni, you would have nothing to change in JSBSim.

dany93, 2022-11-10:
I think that the issue has been fixed in the c172p, just by changing the code in the float chamber.
https://github.com/c172p-team/c172p/issues/1408#issuecomment-1307381746

From my current understanding,
@dany93 wrote
> a conflict / competition between JSBSim stopping the fuel from the main tank and the switch in the float chamber which is expected to do the same thing. The first one which reaches the limit wins the game. The 'first one' depends on some last digit.

If the main tank stops feeding with the last digit like 1.500000000000001, the float chamber switch never reaches the unusable content (1.5) condition for the main tank. The float chamber goes on filling in, even though (or because) the main tank level is no longer decreasing.

The error was **not** in JSBSim, it was in the float-chamber code of the c172p.
@bcoconni, you have nothing to change in JSBSim. Anyway, thank you for your maintenance.

With a float chamber, I think the best (only?) way is to use a code independently from the JSBSim one. With property names different from the hard-coded JSBSim ones.

Not more complicated (less code lines than initially in the c172p),
No competition between both features,
No last digit rounding issues, rounding doesn't matter,
Same flexibility for the unusable fuel content.

_If I have not missed something..._
________________

dany93, 2022-11-10:
@bcoconni
What do you think about this piece of code? 
https://jsbsim.sourceforge.net/JSBSim/FGTank_8cpp_source.html (line 263)
```
double FGTank::Drain(double used)
{
  double remaining = Contents - used;

  if (remaining >= 0) { // Reduce contents by amount used.
    Contents -= used;
```
Remnant from _property_tree_ (a.k.a. _youdontneedtoknow_) commit?
`if (remaining >= 0` looks to me like it sets unusable-fuel = 0.

_(my lack of understanding of C++? Sorry...)_

seanmcleod, 2022-11-10:
The `if (remaining >= 0)` statement doesn't change/update `remaining`, it simply tests it to decide whether to execute the code in the first set of braces just below.

And all it's doing in the example you posted from Sourceforge is preventing `Contents` from going negative. If `used > Contents` then in the code you posted `Contents` is set to 0 rather than a negative number.

dany93, 2022-11-10:
Thank you @seanmcleod 

However, in my local FGTank.cpp (download, FlightGear  64ae52438 Tue, 11 Oct 2022) I see:
(line 273)
```
double FGTank::Drain(double used)
{
  double remaining = Contents - used;

  if (remaining >= GetUnusable()) { // Reduce contents by amount used.
    Contents -= used;
  } else { // This tank must be empty.
    if (Contents > GetUnusable())
      Contents = GetUnusable();

    remaining = Contents;
  }

  PctFull = 100.0*Contents/Capacity;
  CalculateInertias();

  return remaining;
}
```
Something has been changed, and it fits with [youdontneedtoknow (a.k.a. propertytree) commit](https://sourceforge.net/u/propertytree/flightgear_fuel_fix/ci/infinite_fuel_fix/).

seanmcleod, 2022-11-10:
@dany93 okay so I've just taken a look. So first off, in terms of the JSBSim repo there has been no commit to `FGTank.cpp` since Feb 5th 2022 - https://github.com/JSBSim-Team/jsbsim/commits/master/src/models/propulsion/FGTank.cpp

The commit you're referring to by `youdontneedtoknow` is to the copy of the JSBSim source code in the FlightGear repo.

Typically fixes/updates are done in the main JSBSim repo and then periodically pushed to FlightGear. But in this case someone has made a change to the JSBSim code in FlightGear's copy. So any subsequent pushes from JSBSim will involve a merge around this change.

The original snippet you posted from SourceForge is much older, before there was any concept of `GetUnusable()`, so it simply prevented `Contents` from going negative, rather clamping it to 0.

The version you just posted in terms of what you have for your local version of `FGTank.cpp` matches the current version of the code in the JSBSim repo, i.e.

https://github.com/JSBSim-Team/jsbsim/blob/486e1f061664e6340e62e78900a8698f300ba040/src/models/propulsion/FGTank.cpp#L268-L285

dany93, 2022-11-10:
Many Thanks @seanmcleod  for this research and your explanations.
I'm reassured.
And sorry for the hassle.

bcoconni, 2022-11-11:
The root of this issue is that *"unusable fuel"* is a volume (units = gal) while fuel content is a mass (units = lbs). As a consequence a comparison between these 2 values need a unit conversion (either gal -> lbs or the other way).

On one one hand, JSBSim performs a comparison between mass values i.e. it converts the unusable volume to a mass:
https://github.com/JSBSim-Team/jsbsim/blob/486e1f061664e6340e62e78900a8698f300ba040/src/models/propulsion/FGTank.h#L290-L292
and the fuel content is compared to the result of this conversion (as per the code in @seanmcleod's comment above).

On the other hand, the code in *c172p* (that I authored, ahem) performed the comparison the other way (i.e. the fuel content was converted to a volume):
```xml
<!-- from Tank 0 (to Collector Tank 4) -->
<fcs_function name="propulsion/tank[0]/contents-volume-gal">
    <function>
        <quotient>
            <p> propulsion/tank[0]/contents-lbs </p>
            <p> propulsion/tank[0]/density-lbs_per_gal </p>
        </quotient>
    </function>
</fcs_function>
<switch name="fuel/from-tank0-to-tank4">
    <default value="0"/>
    <test logic="AND" value="0.1">
        propulsion/tank[0]/priority EQ 1 
        propulsion/tank[0]/contents-volume-gal GT propulsion/tank[0]/unusable-volume-gal
        /consumables/fuel/tank[0]/level-gal_us GT /consumables/fuel/tank[0]/unusable-vol-gal
        /consumables/fuel/tank[4]/level-lbs LT 0.05
        propulsion/tank[4]/priority EQ 1
        accelerations/Nz GE 0
        fuel/tank EQ 0
    </test>
</switch>
```
[Floating point arithmetic](https://en.wikipedia.org/wiki/Floating-point_arithmetic) being what  it is, these 2 ways of comparing `unusable` and `contents` albeit mathematically equivalent might not be perfectly interchangeable when performed with floating point arithmetic. Hence this discussion.

IMHO the patch that I have suggested in my original post might still make sense in this context. Simply rounding down `Contents` a tiny little bit when it reaches the *unusable* amount can make XML code insensitive to round off errors whether the comparison is made in volume units or in mass units. What do you all think ?

dany93, 2022-11-11:
@bcoconni wrote
>The root of this issue is that "unusable fuel" is a volume (units = gal) while fuel content is a mass (units = lbs)

Now, I'm not that certain that it was the root issue.

After my first attempt using weight, lbs (below)
```
    <consumables>
        <fuel>
            <tank n="0">
                (....)
                <!-- <unusable-gal_us type="double">1.5</unusable-gal_us> -->
                <unusable-fuel-lbs type="double">9.03</unusable-fuel-lbs>
            </tank>

        <switch name="fuel/from-tank0-to-tank4">
            <default value="0"/>
            <test logic="AND" value="0.1">
                (...)
                propulsion/tank[0]/contents-lbs GT /consumables/fuel/tank[0]/unusable-fuel-lbs
                (...)
            </test>
        </switch>
```

Everything worked correctly and I thought having found the root issue by avoiding, in c172p and JSBSim, the gal <--> lbs conversion.

But what I didn't realize at this time, and what I (think to) understand now is that, by changing the property names for unusable-fuel, I also avoided the use of the JSBSim feature. JSBSim no longer stopped the main tank from decreasing its level. Thus, the float chamber switch comparison worked, it reached the unusable value and stopped the engine (and the main tank consumption). And I realize now that it would have, independently from any floating point arithmetic rounding, if I had set the values in volume, gal. The internal JSBSim conversion disturbs nothing.
And finally, it also works for the same reason in my last version with the property names in gal units like
`/consumables/fuel/tank[0]/level-gal_us GT /consumables/fuel/tank[0]/unusable-vol-gal`
(different from the JSBSim coded property names for unusable fuel)

With my last code, even by using volume gal units, the c172p internal unit conversion is merely useless, it is deleted. But I do not think that it was the root cause.

> IMHO the patch that I have suggested in my original post might still make sense in this context. Simply rounding down Contents a tiny little bit when it reaches the unusable amount can make XML code insensitive to round off errors whether the comparison is made in volume units or in mass units

I do not say that the patch that you have suggested in your original post does not make sense.
But I don't see its interest. You have to add a few lines in the JSBSim code, adjust the factor in `Contents = 0.9999*GetUnusable()`. This last (although not critical) can be uneasy to test because the bug was erratic. It happened for 1.5 gal, and I found later that it didn't for 1 gal and 0.1 gal. And @wlbragg wrote that it didn't with the integral tank. We kind of were unlucky with the 1.5 gal value, but it enabled revealing the (perfectly repeatable) bug.

I don't see its interest also because the last code that I propose for the c172p is shorter than the initial one (no internal conversion gal <--> lbs), not more complicated: we just have to choose any new property name for the unusable fuel. Rather more simple.
My last code:
`/consumables/fuel/tank[3]/unusable-vol-gal`
instead of
`<unusable-gal_us type="double">1.5</unusable-gal_us>`
and
`propulsion/tank[3]/unusable-volume-gal`

This patch https://github.com/c172p-team/c172p/pull/1409/commits/952956f29b8a64f83f35343db81c3dd58325b35a

The unusable feature in JSBSIm is perfect for the usual configuration (engine directly fed by a tank). But, with a float chamber, we already need a code to stop the fuel consumption (done by the switch). I don't see the interest of adding an extra feature to do the same by JSBSim, with extra lines in JSBSim to circumvent the competition issues. And it does not simplify the code aircraft side.

bcoconni, 2022-11-11:
The following code is returning `1.499999999999999778` (tested on both Linux and Windows) while it should theoretically return `1.5`.

```c++
#include <iostream>
#include <iomanip>

int main(void) {
    double unusable_volume =  1.5;
    double density = 6.6;
    double unusable_contents = unusable_volume * density;
    double volume_from_contents = unusable_contents / density;
    std::cout << std::setprecision(20) << volume_from_contents << std::endl;
}
```

Now if you change `density` to `6.6001`, the same code returns `1.5` which is the exact mathematical result.

> [...] because the bug was erratic. It happened for 1.5 gal, and I found later that it didn't for 1 gal and 0.1 gal.

Indeed the occurrence of round off errors is extremely sensitive to input values as I have tried to show above.

> I don't see its interest also because the last code that I propose for the c172p is [...]

We have to look at the bigger picture: the mere fact that the problem is fixed in the c172p does not mean someone else could not stumble upon the same problem.

Now if everyone is happy with the current state, then let's close this issue.

bcoconni, 2022-11-11:
I have also investigated the code in FlightGear and basically the property `/consumables/fuel/tank[*]/unusable-gal_us` is a copy of `propulsion/tank[*]/unusable-volume-gal`:

File [src/FDM/JSBSim/JSBSim.cxx lines 254-285](https://sourceforge.net/p/flightgear/flightgear/ci/next/tree/src/FDM/JSBSim/JSBSim.cxx#l254):
```c++
    // Set initial fuel levels if provided.
    for (unsigned int i = 0; i < Propulsion->GetNumTanks(); i++) {
      double d;
      SGPropertyNode * node = fgGetNode("/consumables/fuel/tank", i, true);
      FGTank* tank = Propulsion->GetTank(i);
      SGPropertyNode * prop = node->getNode( "density-ppg", true );

      d = prop->getDoubleValue();
      if( d > 0.0 ) {
        tank->SetDensity( d );
      } else {
        prop->setDoubleValue( SG_MAX2<double>(tank->GetDensity(), 0.1) );
      }

      prop = node->getNode( "level-lbs", true );
      d = prop->getDoubleValue();
      if( d > 0.0 ) {
        tank->SetContents( d );
      } else {
        prop->setDoubleValue( tank->GetContents() );
      }

      prop = node->getNode( "unusable-gal_us", true );
      d = prop->getDoubleValue();
      if ( d > 0.0 ) {
        tank->SetUnusableVolume( d );
      } else {
        prop->setDoubleValue(tank->GetUnusableVolume());
      }
      /* Capacity is read-only in FGTank and can't be overwritten from FlightGear */
      node->getNode("capacity-gal_us", true )->setDoubleValue( tank->GetCapacityGallons() );
    }
```

and FlightGear computes `/consumables/fuel/tank[0]/level-gal_us` with a whole lot of conversions (lbs are converted to kg, pounds per gallon to kg per m3 and finally m3 back to gallons):

1. FlightGear copies the tank content using lbs:
```c++
    if ( ! Propulsion->GetFuelFreeze() ) {
      for (i = 0; i < Propulsion->GetNumTanks(); i++) {
        SGPropertyNode * node = fgGetNode("/consumables/fuel/tank", i, true);
        FGTank* tank = Propulsion->GetTank(i);
        double contents = tank->GetContents();
        double temp = tank->GetTemperature_degC();
        double fuelDensity = tank->GetDensity();

        if (fuelDensity < 0.1)
          fuelDensity = 6.0; // Use average fuel value

        node->setDoubleValue("density-ppg" , fuelDensity);
        node->setDoubleValue("level-lbs", contents);
```

2. Contents and density are updated by the methods `TankProperties::setContent_lbs` and `TankProperties::setDensity_ppg`
```c++
  _tiedProperties.Tie("density-ppg", this, &TankProperties::getDensity_ppg, &TankProperties::setDensity_ppg );
  _tiedProperties.Tie("level-lbs", this, &TankProperties::getContent_lbs, &TankProperties::setContent_lbs );
```

3. Contents are converted from lbs to kg
```c++
void TankProperties::setContent_lbs( double value )
{
  _content_kg = SG_MAX2<double>(value * KG_PER_LBS, 0.0);
}
```

4. Density is converted from pounds per gallon to kg per cubic meters
```c++
void TankProperties::setDensity_ppg( double value )
{
  _density_kgpm3 = SG_MAX2<double>(value * KG_PER_LBS / M3_PER_USGAL, 0.0);
}
```
5. The content is computed in cubic meters by dividing the content mass by the fuel density:
```c++
double TankProperties::getContent_m3() const
{
  return _density_kgpm3 > SGLimitsd::min() ? _content_kg / _density_kgpm3 : 0.0;
}
```

6. Finally the content volume is returned in gallons by converting cubic meters to gallons
```c++
double TankProperties::getContent_gal_us() const
{
  return getContent_m3() * USGAL_PER_M3;
}
```

So there is no magic bullet: as I mentioned previously, contents and unusable volume are not using the same units so some conversion must take place. JSBSim performs the conversion with a single multiplication, whereas FlightGear needs 3 multiplications (if we assume that the density is converted only once during initialization).

So your code is now working with 1.5 gallon but I'd say that this is mostly coincidental. Your code is still susceptible to round off errors but it's just that the values that can fail are now different than 1.5 gallons. Of course, as far as the c172p is concerned, the problem is solved as all you need is the code to succeed for 1.5 gallons.

dany93, 2022-11-12:
@bcoconni wrote
> the mere fact that the problem is fixed in the c172p does not mean someone else could not stumble upon the same problem

I'm wondering if I've not understood something...

With my last code in the c172p, there should not be rounding problems because the only, single comparison is done by the float chamber switch.
Which is fed when
```
/consumables/fuel/tank[*]/level-gal_us GT /consumables/fuel/tank[*]/unusable-vol-gal
```
 When the fuel content in the main tank reaches < or = `unusable-vol-gal,` (even with a rounding error) it is enough to trigger a 'stop' in the float chamber feeding (thus a 'stop' in the main tank level decreasing). I don't see where it can be sensitive to rounding.
Differently from the JSBSim feature combined with the float chamber switch condition where a race happens between both, making this way sensitive to rounding. 

Many apologies if this is me... )-:

dany93, 2022-11-12:
@bcoconni wrote
>basically the property `/consumables/fuel/tank[*]/unusable-gal_us` is a copy of `propulsion/tank[*]/unusable-volume-gal`

I'd rather have understood the inverse: 
`/consumables/fuel/tank[*]/unusable-gal_us` is set in the *-set or *-main.xml aircraft file. 
Then, `/fdm/jsbsim/propulsion/tank[*]/unusable-volume-gal` is set (copied from `unusable-gal_us`) at FG start.

bcoconni, 2022-11-13:
I have just added a new test `tests\TestUnusableFuel.py` that checks if the conditions that have been discussed in this issue lead to an error. (Un)fortunately, this test could not make JSBSim fail so I agree there is indeed no need for a fix. This topic will now be monitored by the new test and this issue can now be closed, at least until someone can make `TestUnusableFuel.py` fail.

Thanks all for your contribution.

closed 2022-11-13 by bcoconni

bcoconni, 2022-11-13:
> With anything else in the *-set or *-main .xml file, no extra property.

What I tried to explain is that the property `/consumables/fuel/tank[*]/level-gal_us` uses JSBSim's property `propulsion/tank[0]/contents-lbs` under the hood. So basically you are still comparing the same values.

> Differently from the JSBSim limitation combined with the float chamber switch condition where the JSBSim limitation and the float chamber switch try do do the same operation. A race happens between both, making this way sensitive to rounding.
From my point of view, the JSBSim limitation of the tank level is useless with a float chamber. Worse, it may bring malfunctions.

Could you substantiate this statement ? Especially the race thing ?

bcoconni, 2022-11-13:
> > basically the property `/consumables/fuel/tank[*]/unusable-gal_us` is a copy of `propulsion/tank[*]/unusable-volume-gal`
> 
> I'd rather have understood the inverse: `/consumables/fuel/tank[*]/unusable-gal_us` is set in the *-set or *-main.xml aircraft file. Then, `/fdm/jsbsim/propulsion/tank[*]/unusable-volume-gal` **value** is set (copied from `unusable-gal_us`) at FG start.
> 
> _(not important)_

Agreed.

So as long as you set the property `/consumables/fuel/tank[*]/unusable-gal_us`, you also set the property `/fdm/jsbsim/propulsion/tank[*]/unusable-volume-gal` and that basically triggers the unusable fuel feature in JSBSim. Said otherwise, the modification you made to your code did not disable the unusable fuel feature. Hence following your reasoning, the 2 features are still competing. Yet, if I understood your explanation correctly, the fix you committed to your code avoid the race from taking place. How do you explain that ?

dany93, 2022-11-13:
> So as long as you set the property /consumables/fuel/tank[*]/unusable-gal_us, you also set the property /fdm/jsbsim/propulsion/tank[*]/unusable-volume-gal and that basically triggers the unusable [...]

Sorry, I have no time, I will respond later to give more.

I haven't set `/consumables/fuel/tank[*]/unusable-gal_us` I have set `/consumables/fuel/tank[*]/unusable-vol-gal`,  a new property that I invented for this in -main.xml (l. 684). Used as it is in fuel.xml. It comes down to write `/consumables/fuel/tank[1]/level-gal_us GT 1.5` in the float chamber switch. This code did disable the 'JSBSim unusable fuel feature'.

dany93, 2022-11-15:
>Could you please substantiate this statement ? Especially the race thing ?

(1)- The 'JSBSim unusable fuel feature' lets the main tank level decrease if 
`/consumables/fuel/tank[*]/level-gal_us GT propulsion/tank[*]/unusable-volume-gal`. 
With level-gal_us calculated by JSBSim.
(2)- The float chamber switch lets float chamber be fed if 
(/fdm/jsbsim properties)
`propulsion/tank[0]/contents-volume-gal GT propulsion/tank[*]/unusable-volume-gal`

With contents-volume-gal converted by the float chamber fcs_function.

If (1) is fulfilled first (rounding error for this value), the float chamber never reaches its condition, it never stops from filling in. Despite (rather because) the main tank has stopped from decreasing.
Like this
[image]

`propulsion/tank[*]/contents-volume-gal` can stop decreasing at a value slighly greater than 1.5.

To display theses values, I often have to click several times in the property inside the black box. It looks like a numerical last digit, more or less erratic occurrence, every one out of a few frames.
When this happens, it triggers the float chamber switch, which makes that the float chamber never empties: the engine never runs out of fuel.

tweaking the values like your 'Suggested fix' in  https://github.com/JSBSim-Team/jsbsim/issues/759#issue-1437504047
```
    if (Contents > GetUnusable())
-      Contents = GetUnusable();
+      Contents = 0.9999*GetUnusable();
```
(although I don't understand the code)
might be a solution, but why apply a circumventing if we can avoid this feature?

I tried to decouple the JSBSim limit from the float chamber limit regarding the main tanks.
- 'float chamber limit': set a value for tank[*]/contents-volume-gal = 1.5
by writing, in the float chamber switch for tanks 0 and 1:
 `propulsion/tank[*]/contents-volume-gal GT 1.5`
- JSBSim limit: stop the tank[*] content from decreasing
by setting a **value** in `propulsion/tank[*]/unusable-volume-gal` ('JSBSim limit', via the -main.xml file or by setting it in the property, in the simulator).
If I set 1.8, the engine never stops, as expected, despite (or because) the main tank level had stopped decreasing.
If I set 1.5 in the simulator, the engine stops slightly below 1.5 (surprisingly)
If I set 1.3 or 0, the engine stops for 1.4999 gal.

This shows that the JSBSim unusable-volume-gal feature is useless here, the float chamber switch works by itself with this feature disabled

Up to now, I could not have my current code to fail.
I also tried with the default density (6.6). No problem.
Everything I observe goes in the direction of (or is compatible with) a last digit fluctuation competition **when a double test in done**.

If I don't miss something in my attempts... I understand the code too poorly, that's rather reverse engineering.

Difficult to write... Please tell me if Ive made mistakes or poor descriptions before loosing your time.