JSBSim-Team/jsbsim #787: Negative pointmass is calculated

MariuszXC, 2022-12-22:
**I'm submitting a ...**
  - [x] bug report
  - [ ] feature request
  - [ ] support request => **Please do not submit support request here**, see note at the top of this template.

**Describe the issue**
A negative pointmass Payload is calculated with a comment (in output file):
- in web aeromatic: "- xxx LBS + full (35202 LBS) fuel should bring model up to entered max weight",
- in command line aeromatic: "- xxx LBS should bring model up to entered max weight".

This should not happen, because a negative mass is:
- not allowed by schema, and
- physically impossible.

**What is the current behavior?**
Relevant input data in my case was:
MTOW: 34473 kg,
EmptyW: 23288 kg.
Question about max. allowed fuel mass is not asked.
As a result negative pointmass is calculated (which is bad).

**What is the expected behavior?**
I would expect to be able to enter maximum allowed fuel capacity (in LBS or kg). 
As it is now, aeromatic makes an assumption, which produces bad result.

**What is the motivation / use case for changing the behavior?**
Better user experience :)

**Please tell us about your environment:**

- Linux (Ubuntu 22.04)
- AeromatiC++ version 3.3.20
- https://jsbsim-team.github.io/aeromatic/ (as currently available)

**Other information**

I can provide full input data used to help replicate this issue in a comment, if so requested.

bcoconni, 2022-12-26:
@ermarch, this issue is related to `aeromatic++` could you please investigate ?

ermarch, 2023-03-06:
Full input data would be welcome as I don't see this behavior on this data alone.
I 'm only able to reproduce it if I switch the values for max weight and empty weight.

MariuszXC, 2023-03-12:
It has been a while (almost 4 months) since I submitted this report, so I'll have to check if I still have my notes about full input data. If not, I'll try to re-create the issue. Sorry about that, but it's been a while before you asked..

ermarch, 2023-03-13:
No problem, at least it triggered me to make AeromatiC++ more robust.
For one I switch max weight and empty weight is the later is larger than the first and it doesn't crash with a segmentation fault anymore if not a single entry is filled in.

If your problem still exist I can fix it. And then I'll d another pull request.

closed 2023-03-24 by bcoconni

bcoconni, 2023-03-24:
The PR #860 has just been merged.
Thanks @MariuszXC for the bug report and @ermarch for the bug fix :+1:

PR #860 (2023-03-24): Aeromatic fix inputs
Harden input handling and report warnings and alerts if something seems wrong.
This might fix #787. If not it could always be opened again.