JSBSim-Team/jsbsim #794: cannot change property value at run time using telnet set command

pbecchi, 2022-12-27:
I try to modify property value at run time using an external TELNET command but the actual value dont change.

Having added to model XML file this line:
`<input type="SOCKET" port="1155"/>`
I can succesfully connect with a telnet session: `telnet localhost 1155`
I can get property values by the `get \property\` command , but if i try the command :`set \property\ new value` I get a `set successfull` answer but the property value is not modified.  
Regardless where JSBSim is running or is on hold , the property value never change.
I dont understand what i am doing wrong.  Can somebody help me?

seanmcleod, 2022-12-27:
What property are you trying to set? Have you confimed that it isn't a read-only property?

```sh
JSBSim.exe --aircraft=737 --catalog`

  Property Catalog for 737

    inertial/sea-level-radius_ft (R)
    simulation/gravity-model (RW)
    simulation/integrator/rate/rotational (RW)
    simulation/integrator/rate/translational (RW)
    simulation/integrator/position/rotational (RW)
    simulation/integrator/position/translational (RW)
    simulation/write-state-file (W)
    simulation/channel-dt (R)
    simulation/gravitational-torque (RW)
    simulation/force-output (W)
    simulation/do_simple_trim (W)
    simulation/reset (W)
    simulation/disperse (R)
    simulation/randomseed (RW)
    simulation/terminate (RW)
    simulation/pause (RW)
    simulation/sim-time-sec (R)
    simulation/dt (R)
```

pbecchi, 2022-12-27:
Thanks for your fast answer -:).
I try to change the properties related to local wind , like `atmosphere/psiw-rad` that should be RW.
This is what i get:
`JSBSim> iterate 1

Iterations performed

JSBSim> set atmosphere/psiw-rad 1

set successful

JSBSim> get atmosphere/psiw-rad

atmosphere/psiw-rad =            0

JSBSim> iterate 1

Iterations performed

JSBSim> get atmosphere/psiw-rad

atmosphere/psiw-rad =            0

JSBSim> resume

Resuming

JSBSim> set atmosphere/psiw-rad 0.5

set successful

JSBSim> get atmosphere/psiw-rad

atmosphere/psiw-rad =            0`

seanmcleod, 2022-12-27:
Have you set a wind speed? While updating `atmosphere/psiw-rad` the `SetWindSpeed()` method is called.

https://github.com/JSBSim-Team/jsbsim/blob/9cb22d7d56d3841dc799420cc387e27f75361acf/src/models/atmosphere/FGWinds.cpp#L183-L190

And `psiw` can be reset to 0 in `SetWindSpeed()`.

https://github.com/JSBSim-Team/jsbsim/blob/9cb22d7d56d3841dc799420cc387e27f75361acf/src/models/atmosphere/FGWinds.cpp#L162-L172

pbecchi, 2022-12-27:
It works now:

JSBSim> set atmosphere/wind-mag-fps 10
set successful
JSBSim> get atmosphere/wind-mag-fps
atmosphere/wind-mag-fps =           10
JSBSim> set atmosphere/psiw-rad 0.5
set successful
JSBSim> get atmosphere/psiw-rad
atmosphere/psiw-rad =          0.5

Now I can change wind speed and direction during simulation! -:)       Thanks so much!

closed 2022-12-27 by pbecchi