JSBSim-Team/jsbsim #805: Online Python API documentation not working

Dobid, 2023-01-06:
Hello,
I'm facing the exact same issue as in [#373](https://github.com/JSBSim-Team/jsbsim/issues/373) : when I click on a class to see the docs, everything is blank. The issue seemed to be resolved back a year ago, but it seems to have resurfaced.

[image]

Thanks in advance !

Dobid, 2023-01-25:
Up !

bcoconni, 2023-01-25:
Thanks @Dobid for pointing this out.
I can reproduce the issue and I'll try to address this in the coming days.

bcoconni, 2023-01-28:
This is now fixed [FGFDMExec](https://jsbsim-team.github.io/jsbsim/python/FGFDMExec.html) is functional again.

Dobid, 2023-01-30:
Great, every section is now displaying something ! I'm closing the issue. Thank you very much.

closed 2023-01-30 by Dobid