JSBSim-Team/jsbsim #808: Compilation Failed While Opening in UE 5.1

junaidbinsarfraz, 2023-01-09:
**I'm submitting a ...**
  - [x] bug report
  - [ ] feature request
  - [ ] support request => **Please do not submit support request here**, see note at the top of this template.

**Describe the issue**
Cloned the master branch and try to open `jsbsim/UnrealEngine/UEReferenceApp.uproject` from Unreal Engine 5.1. But it didn't open the project rather threw a compilation error

```
The project could not be compiled. Would you like to open it in Visual Studio?

Running C:/Program Files/Epic Games/UE_5.1/Engine/Build/BatchFiles/Build.bat Development Win64 -Project="D:/jsbsim/UnrealEngine 5.1 - 2/UEReferenceApp.uproject" -TargetType=Editor -Progress -NoEngineChanges -NoHotReloadFromIDE
Running UnrealBuildTool: dotnet "..\..\Engine\Binaries\DotNET\UnrealBuildTool\UnrealBuildTool.dll" Development Win64 -Project="D:/jsbsim/UnrealEngine 5.1 - 2/UEReferenceApp.uproject" -TargetType=Editor -Progress -NoEngineChanges -NoHotReloadFromIDE
Log file: C:\Users\PC\AppData\Local\UnrealBuildTool\Log.txt
Using 'git status' to determine working set for adaptive non-unity build (D:\jsbsim).
Creating makefile for UEReferenceAppEditor (no existing makefile)
JSBSim.dll not found in D:\jsbsim\UnrealEngine 5.1 - 2\Plugins\JSBSimFlightDynamicsModel\Source\ThirdParty\JSBSim\Lib - Make sure that you have built JSBSimForUnreal.sln first
Unable to instantiate module 'JSBSim': JSBSim.dll not found in D:\jsbsim\UnrealEngine 5.1 - 2\Plugins\JSBSimFlightDynamicsModel\Source\ThirdParty\JSBSim\Lib - Make sure that you have built JSBSimForUnreal.sln first
(referenced via Target -> JSBSimFlightDynamicsModel.Build.cs)
```

**What is the current behavior?**
[image]

**Please tell us about your environment:**
 - OS: Windows 10
 - JSBSim version: clone of master branch

seanmcleod, 2023-01-09:
The error message mentions:

> JSBSim.dll not found in D:\jsbsim\UnrealEngine 5.1 - 2\Plugins\JSBSimFlightDynamicsModel\Source\ThirdParty\JSBSim\Lib - Make sure that you have built JSBSimForUnreal.sln first

You didn't specify whether you built `JSBSimForUnreal.sln` as the error message suggests.

junaidbinsarfraz, 2023-01-09:
I have completed the `JSBSimForUnreal.sln`. When I open the project it says

This project was made with a different version of the Unreal Engine. Converting to this version will rebuild your code projects.

```
This project was made with a different version of the Unreal Engine. Converting to this version will rebuild your code projects.

New features and improvements sometimes cause API changes, which may require you to modify your code before it compiles. Content saved with newer versions of the editor will not open in older versions.

We recommend you open a copy of your project to avoid damaging the original.
```

I am using UE5.1 and latest JSBSim master code.

gallonmate, 2023-01-09:
You can just click okay, it's just an Unreal Engine standard warning that the project cannot go back to version 5.0 once you open the project in engine 5.1.
If you still have problems, open and build UEReferenceApp.sln found in the Unreal folder.

junaidbinsarfraz, 2023-01-09:
@gallonmate I get your point, when I run `UEReferenceApp.sln` I get following errors

```
Build started...
1>------ Skipped Build: Project: UE5, Configuration: BuiltWithUnrealBuildTool Win64 ------
1>Project not selected to build for this solution configuration 
2>------ Build started: Project: UEReferenceApp, Configuration: Development_Editor x64 ------
2>Running UnrealBuildTool: dotnet "..\..\Engine\Binaries\DotNET\UnrealBuildTool\UnrealBuildTool.dll" UEReferenceAppEditor Win64 Development -Project="D:\jsbsim\UnrealEngine 5.1\UEReferenceApp.uproject" -WaitMutex -FromMsBuild
2>Log file: C:\Users\PC\AppData\Local\UnrealBuildTool\Log.txt
2>Using 'git status' to determine working set for adaptive non-unity build (D:\jsbsim).
2>Creating makefile for UEReferenceAppEditor (no existing makefile)
2>JSBSim.dll not found in D:\jsbsim\UnrealEngine 5.1\Plugins\JSBSimFlightDynamicsModel\Source\ThirdParty\JSBSim\Lib - Make sure that you have built JSBSimForUnreal.sln first
2>Unable to instantiate module 'JSBSim': JSBSim.dll not found in D:\jsbsim\UnrealEngine 5.1\Plugins\JSBSimFlightDynamicsModel\Source\ThirdParty\JSBSim\Lib - Make sure that you have built JSBSimForUnreal.sln first
2>(referenced via Target -> JSBSimFlightDynamicsModel.Build.cs)
2>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.MakeFile.Targets(45,5): error MSB3073: The command ""C:\Program Files\Epic Games\UE_5.1\Engine\Build\BatchFiles\Build.bat" UEReferenceAppEditor Win64 Development -Project="D:\jsbsim\UnrealEngine 5.1\UEReferenceApp.uproject" -WaitMutex -FromMsBuild" exited with code 6.
2>Done building project "UEReferenceApp.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 1 skipped ==========
```

gallonmate, 2023-01-09:
Open JSBSimForUnreal.sln and if it was Debug build, then change to Release and then can build it.
After that's done, UEReferenceApp.sln build should work.

herewego48, 2023-03-08:
> 你可以点击确定，这是一个虚幻引擎标准警告，一旦你在引擎5.0中打开项目，项目就无法返回到5.1版。如果仍有问题，请打开并构建UEReferenceApp.sln该应用程序位于虚幻文件夹中。
I use ubuntu and VSCODE. It seems that I can't use sln file.I simply copied the folder under plugin of JSBSim to the plugins folder of UE5. I don't know what to do next.

gallonmate, 2023-03-08:
@herewego48 The plugin is currently only setup for Windows. It does not currently work for Linux, but it might be possible.
See this discussion: https://github.com/JSBSim-Team/jsbsim/discussions/832

herewego48, 2023-03-08:
thank you very much !

herewego48, 2023-03-10:
thank you for your answer sir !

herewego48, 2023-03-10:
> @herewego48 The plugin is currently only setup for Windows. It does not currently work for Linux, but it might be possible. See this discussion: #832

Thank you for your answer sir !

gallonmate, 2023-04-03:
> Open JSBSimForUnreal.sln and if it was Debug build, then change to Release and then can build it. After that's done, build again the UEReferenceApp.sln
> 
> JSBSimForUnreal.sln has to be built in Release configuration in order to create JSBSim.dll If build as Debug config, it creates JSBSimDebug.dlll ,and the Unreal project cannot find it, unless also Unreal project is also built as Debug.

Please keep this issue Open. It's actually a common issue with the plugin for new users because the default build settings for Visual Studio aren't matching UE5 build settings. It's an easy fix to re-order the default build options so everything is matching when projects are opened for the first time. But the Debug Build for UE5 is broken anyways, as discussed here https://github.com/JSBSim-Team/jsbsim/discussions/761. In the near future I'll add a pull request which solves both issues.

closed 2023-05-12 by bcoconni