JSBSim-Team/jsbsim #809: JSBSim crashes after a while at FGTable::GetValue()

vranki, 2023-01-09:
**I'm submitting a ...**
  - [x] bug report
  - [ ] feature request
  - [ ] support request => **Please do not submit support request here**, see note at the top of this template.

**Describe the issue**
After some random time (tens of minutes) of flying, JSBSim crashes with assert.

**What is the current behavior?**
double JSBSim::FGTable::GetValue(double) const: assertion 'Factor >= 0.0 && Factor <= 1.0' failed.

**What is the expected behavior?**
Not crashing.

**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**
 - OS Ubuntu 20.04
 - JSBSim version 1.1.13
 - Language C++

**Other information**
This happens also often when the plane is crashed to the ground. But happens also when flying normally. Tested with f16 aircraft model from JSBSim sources.

vranki, 2023-01-10:
Stack trace:
```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff69f2859 in __GI_abort () at abort.c:79
#2  0x00007ffff69f2729 in __assert_fail_base (fmt=0x7ffff6b88588 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", assertion=0x7fffe8a59e10 "Factor >= 0.0 && Factor <= 1.0", 
    file=0x7fffe8a59d80 "/home/runner/work/jsbsim/jsbsim/src/math/FGTable.cpp", line=511, function=<optimized out>) at assert.c:92
#3  0x00007ffff6a03fd6 in __GI___assert_fail (assertion=assertion@entry=0x7fffe8a59e10 "Factor >= 0.0 && Factor <= 1.0", 
    file=file@entry=0x7fffe8a59d80 "/home/runner/work/jsbsim/jsbsim/src/math/FGTable.cpp", line=line@entry=511, function=function@entry=0x7fffe8a59de0 "double JSBSim::FGTable::GetValue(double) const")
    at assert.c:101
#4  0x00007fffe898fffa in JSBSim::FGTable::GetValue (this=<optimized out>, key=<optimized out>) at /usr/include/c++/9/bits/stl_vector.h:1058
#5  0x00007fffe8977786 in JSBSim::FGFunction::<lambda(const std::vector<SGSharedPtr<JSBSim::FGParameter>, std::allocator<SGSharedPtr<JSBSim::FGParameter> > >&)>::operator() (__closure=<optimized out>, 
    Parameters=...) at /home/runner/work/jsbsim/jsbsim/src/math/FGFunction.cpp:382
#6  JSBSim::aFunc<JSBSim::FGFunction::Load(JSBSim::Element*, JSBSim::FGPropertyValue*, JSBSim::FGFDMExec*, const string&)::<lambda(const std::vector<SGSharedPtr<JSBSim::FGParameter>, std::allocator<SGSharedPtr<JSBSim::FGParameter> > >&)>, 2>::GetValue(void) const (this=<optimized out>) at /home/runner/work/jsbsim/jsbsim/src/math/FGFunction.cpp:89
#7  0x00007fffe897766f in JSBSim::FGFunction::GetValue (this=0x55555788ae70) at /home/runner/work/jsbsim/jsbsim/src/math/FGFunction.cpp:936
#8  JSBSim::FGFunction::GetValue (this=0x55555788ae70) at /home/runner/work/jsbsim/jsbsim/src/math/FGFunction.cpp:932
#9  0x00007fffe89794b7 in JSBSim::FGFunction::cacheValue (this=0x55555788ae70, cache=cache@entry=true) at /home/runner/work/jsbsim/jsbsim/src/math/FGFunction.cpp:925
#10 0x00007fffe8918cb1 in JSBSim::FGAerodynamics::Run (this=0x55555663a290, Holding=<optimized out>) at /home/runner/work/jsbsim/jsbsim/src/models/FGAerodynamics.cpp:194
#11 0x00007fffe88e1346 in JSBSim::FGFDMExec::Run (this=0x555555b61a80) at /usr/include/c++/9/bits/stl_vector.h:1040
```

seanmcleod, 2023-01-10:
Can you set a breakpoint at this assertion to figure out which table and what the values are when this assertion fails?

vranki, 2023-01-11:
Looks like the library from debian package has compiled with optimization so I haven't been able to dig much out of it. GDB doesn't find symbol names or says they are optimized out.

I found out variable "temp" gets a NaN value when operation is "product". I haven't been able to figure out where and why it gets that. Testing this is difficult as reproducing can sometimes take a hour or two.

seanmcleod, 2023-01-11:
My assumption was that the crash wasn't specific to a release/optimized build, i.e. that you could build a debug build and set a breakpoint at the assert and then inspect the local values, the call stack etc. to figure out what table is in use that is causing the issue and we could then figure out the issue.

The assert you posted above is from line 511 below.

https://github.com/JSBSim-Team/jsbsim/blob/5ac73c2b555ac7b6b97a1ef43f3f47f1f68a6f0f/src/math/FGTable.cpp#L492-L515 

Which means that either `key < x0` or `key - x0 > Span`.

dpculp, 2023-01-11:
If you can post the XML file here I might be able to spot a typo in one of the tables.

seanmcleod, 2023-01-11:
@dpculp he mentioned:

> Tested with f16 aircraft model from JSBSim sources.

dpculp, 2023-01-11:
@seanmcleod Oops, I missed that bit.  

I just ran a cruise test with the F-16 model and it suffered a pitch excursion at about 90 seconds into cruise flight leading to a crash.  There have been reports in the past about pitch excursions with this model, and it's probably coming from integrator wind-up in a PID controller, although it could be coming from elsewhere in the pitch channel.

It might be best to set the status of this model to "beta" or "development" until it gets fixed.

The aircraft with simple pitch channels don't have this problem.

seanmcleod, 2023-01-12:
@dpculp when you say "leading to a crash" do you mean a crash into the ground or do you mean JSBSim crashes with a NaN or an assert etc.? Ah, I see your chart now. So it looks like you mean a crash into the ground as opposed to a JSBSim crash? Looks like it hits the ground around the 155s mark and then bounces once before coming to a rest?

vranki, 2023-01-12:
Correct, I'm using the f16 from jsbsim source distribution for testing. I haven't noticed any pitch excursions yet. The assert failure happens usually in stable level flight after random amount of time.

I'm traveling for a few days, back at computer next week so I can't do any further testing until that.

seanmcleod, 2023-01-12:
@vranki are you running JSBSim stand-alone, or are you using FlightGear or your own sim?

Glancing at the source code, given the range checking with lines 497-500 and given that the assert on line 509 `assert(Span > 0.0);` passes it's difficult to see how the assert on line 511 `assert(Factor >= 0.0 && Factor <= 1.0);` fires unless the table data gets corrupted at some point in memory, or maybe a `key == NaN`?

dpculp, 2023-01-12:
@seanmcleod Yes, in this case I mean crash as in "flew into ground", and it did so without generating any NaNs, which I find surprising.   In this test the airplane is trimmed at 20k feet altitude and let loose for 5 minutes with no control inputs.  After diving into the ground it does some unphysical things and eventual completes the 5 minute run.

The only reason I bring this test up is that it shows that the model is expected to "crash" anyway when run in a simple test using JSBSim stand-alone, so @vranki 's  trouble with it should also be expected.  I recommend @vranki switch to a different airplane.

As for the F-16 model it appears the pitch channel needs some work, and that could take a long time.

seanmcleod, 2023-01-14:
@dpculp I've started a separate F-16 FCS Issues discussion, see - https://github.com/JSBSim-Team/jsbsim/discussions/814 since I'm pretty sure that the issue you noticed with a simple level trim is separate to the issue @vranki is seeing with the assert in the `FGTable` lookup code.

vranki, 2023-01-18:
Ok I managed to do a debug build and got some new details from it:

 - Table "CDmach" gets NaN as key. 
 -  The FGTable's Data vector contains NaN's as first two values. This is probably normal (see FGTable() contructor). The remaining values match data in f16's Drag_due_to_mach table.
 - The lookupProperty[0].PropertyName is "velocities/mach"

My analysis is that "velocities/mach" seems to get NaN value and it's used for lookup key in the table. Now why does that happen? I'll try debugging more.

seanmcleod, 2023-01-18:
Hmm, `Mach` is simply:

https://github.com/JSBSim-Team/jsbsim/blob/5ac73c2b555ac7b6b97a1ef43f3f47f1f68a6f0f/src/models/FGAuxiliary.cpp#L180

So either `Vt` and/or `in.SoundSpeed` have gone haywire.

vranki, 2023-01-18:
I added an assert there to check for NaN, now waiting for the bug to repeat itself.

vranki, 2023-01-19:
Looks like NaNs are leaking from in variables - for example in.vPQR, in.CosTht/CosPhi/etc, in.Tb2l and various other variables have NaN's. The NaN explosion happens somewhere earlier. I wish the code had asserts to check for them to catch them as early as possible.

seanmcleod, 2023-01-22:
@vranki in terms of environment, are you running JSBSim stand-alone, or via FlightGear or via your own flight simulator?

seanmcleod, 2023-01-22:
For comparison I'm running JSBSim stand-alone for 25 mins and I've run it many times without seeing the assert.

The only change I made was to increase the P gain in the PID for roll control to ensure that the aircraft doesn't roll over and descend into the ground before the 25mins are up.

[image]

Losing about 3,000ft in altitude over the 25mins.

```sh
Repeating Notify (Event 2) executed at time: 1500.006664
    propulsion/engine[0]/n2 = 91.305264
    propulsion/engine[0]/thrust-lbs = 6421.272962
    velocities/vc-kts = 356.594497
    position/h-agl-ft = 17179.881806
    attitude/theta-deg = -1.722904
    attitude/roll-rad = 0.492895
    flight-path/gamma-deg = -2.571614
    aero/alpha-deg = 0.225443
    fcs/elevator-cmd-limiter = -0.059655
    fcs/elevator-scheduler = -0.059238
    fcs/pitch-trim-error = 0.020803
    fcs/roll-trim-error = 0.013752
    fcs/aileron-cmd-norm = -0.000000
```

vranki, 2023-01-23:
I have found a probable cause for this issue. Due to timestamp roll-over elsewhere I set Delta-T to about -4000. This moved the aircraft to outer space (or wherever) which triggered #815 . This caused other calculated values to become NaN's until the assertion failed later in code.

I'd suggest using more defensive programming in JSBSim code. Use asserts to check input and return values and in some cases verify that temporary variables have sane values. This would act as documentation itself and improve code quality a lot. Catching invalid values as early as possible  makes debugging problems like this much easier.

seanmcleod, 2023-01-23:
> Due to timestamp roll-over elsewhere I set Delta-T to about -4000. 
> Use asserts to check input and return values and in some cases verify that temporary variables have sane values.

So your assert for checking the value of `delta-T` that you calculated didn't fire? 😉

closed 2023-01-23 by vranki