JSBSim-Team/jsbsim #818: FDM returns NaN values in properties (not for a crash)

Dobid, 2023-01-26:
**I'm submitting a ...**
  - [x] bug report
  - [ ] feature request
  - [ ] support request => **Please do not submit support request here**, see note at the top of this template.

**Describe the issue**
JSBSim newbie here, feel free to move this issue to "Discussions". The FDM returns NaN values for (at least) the following properties : position/lat-gc-deg, position/long-gc-deg, position/h-sl-meters, attitude/roll-rad, attitude/pitch-rad, attitude/heading-true-rad

**What is the current behavior?**
I have the .xml files describing the Skywalker X8 UAV and its engine (you'll find the files attached, as well as the initial conditions xml file and the small python script I wrote). When I leave everything uncontrolled (no changes to fcs/aileron,elevator,throttle-cmd-norm), the simulation works well. However, when I set the fcs/aileron-cmd-norm property with +/- 0.3, I quickly get NaN values filling up my csv table.  The problem does not occur when I input 0.1. 

I tried reproducing the bug with the c172p, and I get NaNs only when there is a crash, and only for a value of -0.3 for the aileron command (+0.3 doesn't make the bug appear).
I stumbled upon this issue #243 that describes the same kind of problem I'm currenlty facing, and the answer (JSBSim not being very good at handling hard impacts) seems to explain the behaviour when testing with the c172p. However, it doesn't for my x8 UAV since the NaNs appear when the aircraft is still well above the ground (~173m for the last non aberrant value).

Is this a bug of JSBSim ? Does it come from my .xml files ? Is +/- 0.3 considered as "too extreme" of a command ?

**What is the expected behavior?**
The csv table to fill up with real values.

**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**
 - Ubuntu 20.04
 - JSBSim version : 1.1.13 (installed as a python package via conda)
 - Language if applicable : Python 3.8.15

Thanks in advance and sorry if this isn't a JSBSim bug, I can't really tell by myself at this point.

**Other information**
For your information, the xml files describing the x8 UAV and its engine directly come from this repository : [AOS55/Fixedwing-Airsim](https://github.com/AOS55/Fixedwing-Airsim).

x8 xml files : [x8_xml.zip](https://github.com/JSBSim-Team/jsbsim/files/10511662/x8_xml.zip)
python script that launches everything and the initial conditions.xml file : [script_and_ic.zip](https://github.com/JSBSim-Team/jsbsim/files/10511680/script_and_ic.zip)

seanmcleod, 2023-01-26:
You're starting the aircraft stationary with no velocity based on initial conditions file.

Any reason you're changing the default timestep?

What do the plots show for angular rates, attitude angles, airspeed, altitude etc.?

Setting the aileron will generate a rolling moment, now if the moment of inertia for example is way too small given the moment generated then it is possible that once JSBSim integrates the resulting moment to get an angular rate, integrates it again to get an attitude angle etc. that if there is a massive mismatch you could end up with NaNs.

dpculp, 2023-01-27:
You have a "direct" thruster attached to your electric motor.  You'll need to replace the thruster with either a propeller or a rotor in order to get thrust.

As currently configured your model can only fall to Earth.

Dobid, 2023-01-27:
Hello @seanmcleod, thanks for the quick reply,

> You're starting the aircraft stationary with no velocity based on initial conditions file.

Yes, I wanted to leave everything at 0, except the aileron command in order to isolate the problem. I just tested it with ubody = 5.0 m/s and I have the same problem.

> Any reason you're changing the default timestep?

If you're talking about the `fdm.set_dt()` : No, no particular reason, apart that I've seen it repetetively in other repositories. I thought it was mandatory to set a simulation timestep. I just commented out this line, the problem resolves itself for the value +/-0.3 but when I increase it to 0.4, I get NaNs again :/

> What do the plots show for angular rates, attitude angles, airspeed, altitude etc.?

Is there an integrated utility program to jsbsim that allows for plot visulization ? Anyways, here are two plots I made : both of them : same initial conditions, no initial velocity, I let the default timestep : The first one is with an aileron command of -0.3 (this now works, by not changing the default timestep), the second one is with an aileron command of -0.4 (this one generates NaNs after ~630 timesteps) 
Sim with -0.3 (now working) :
[image]

Sim with -0.4 (gives NaNs after a bit) :
[image]

> Setting the aileron will generate a rolling moment, now if the moment of inertia for example is way too small given the moment generated then it is possible that once JSBSim integrates the resulting [...]

I'm sorry, but I don't understand what you've just said here. I get that setting the aileron will generate a rolling moment, and that JSBSim integrates from accelerations to positions but what about the mismatch and the moment of inertia being way too small ?

seanmcleod, 2023-01-27:
> I get that setting the aileron will generate a rolling moment, and that JSBSim integrates from accelerations to positions but what about the mismatch and the moment of inertia being way too small ?

Take a translational example, i.e. $F = ma$, now think about what happens if  $F = 1\mathrm{e}{20}$ and $m = 1\mathrm{e}{-20}$.

Although looking at your graphs it looks similar to some of the results you see when the aircraft hits the ground very hard, with massive excessive spring/gear forces etc.

How sure are you that you aren't hitting the ground? You seem to be logging the sea level altitude, as opposed to AGL and I see reference to an `elevation` element in your initial conditions file.

Is there no output in the console about ground contact etc.?

Dobid, 2023-01-27:
> How sure are you that you aren't hitting the ground? You seem to be logging the sea level altitude, as opposed to AGL and I see reference to an `elevation` element in your initial conditions file.
> 
> Is there no output in the console about ground contact etc.?

I don't have any `ground_reactions` set up in my xml file, when testing for other aileron values that don't break the simulation, my aircraft goes below 0 without a problem. I tried replacing the sea level altitude by the above ground altitude, I get the same numbers (except h-sl-m being in meters and h-agl-km being in km).

For the `elevation` element, I got it from the initial conditions file present in the repository I cited earlier that I follow as an example of what I want to do. No reason I know to why it's there, I tried removing it, doesn't resolve the problem.

seanmcleod, 2023-01-27:
Okay, so you're basically dropping a stationary aircraft from some altitude and it's entering a spin right? Getting to the point of angular rates in the region of thousands of degrees per second, I'm assuming the y-axis of your angular velocities graph is in rad/s right?

And after getting to rates in the order of a couple of thousands of deg/s JSBSim then starts spitting out NaNs?

Dobid, 2023-01-27:
> Okay, so you're basically dropping a stationary aircraft from some altitude and it's entering a spin right? Getting to the point of angular rates in the region of thousands of degrees per second, I'm assuming the y-axis of your angular velocities graph is in rad/s right?

rad/s, correct.

> And after getting to rates in the order of a couple of thousands of deg/s JSBSim then starts spitting out NaNs?

I think so.  In order to have readable graphs, since I stopped the simulation right before the NaNs error occur : here's the csv file that I plot the data from, just in case you want to check the last lines before it spits NaNs.

[flight_data.csv](https://github.com/JSBSim-Team/jsbsim/files/10520159/flight_data.csv)

seanmcleod, 2023-01-27:
So these are the last couple of entries, take 437 rad/s for the first roll rate in the snippet below, that's 25,038 deg/s.

Roll rate (rad/s)| Pitch rate (rad/s) | Yaw rate (rad/s)
-- | -- | --
437.3738449 | 26.77790139 | -578.405973
443.8719697 | -2214.108471 | -601.5700519
-8198.758529 | -5168.244707 | 10498.49683
176081.2867 | -504479.3471 | -240302.6891
-824439661.4 | -502235492.2 | 1056244540
1.78073E+15 | -5.09655E+15 | -2.43034E+15
-8.42E+28 | -5.14E+28 | 1.08E+29
1.86E+55 | -5.33E+55 | -2.54E+55
-9.20E+108 | -5.61E+108 | 1.18E+109
2.22E+215 | -6.36E+215 | -3.03E+215
nan | nan | nan

So you've dropped an aircraft at zero velocity, and it's almost immediately gone into a spin that looks unstable.

So I'm not sure what you're really expecting. Do you know based on some other real world data that this aircraft with those sorts of starting conditions has stable spin characteristics?

Or have you explicitly checked or edited the FDM to ensure that it would be stable in a spin by editing the relevant aerodynamic coefficients?

Dobid, 2023-01-31:
Ok I think I've got it, I made some other experiments with an initial velocity and different throttle and aileron commands. Some lead to stable trajectories, some other to unstable ones and NaNs... If I understand correctly, I should check the stability of my UAV to know precisely what conditions lead to unstable trajectories.

gallonmate, 2023-01-31:
The aircraft you are using seems experimental and was only designed/tested for use with the autopilot included in that repo. The model itself does not appear to stabilized in anyway for manual control without an autopilot/stabilization system. Are you trying to fly the craft outside the scope of that repo you linked? You'll need to test/tune/improve the model if you intend on manually controlling it. You can search/read about creating aircraft models for JSBSim or ask for help with the model specifically. There also exists quadcopters, drones, and possibly other miniature craft if you search around. From which you could compare and add improvements to the X8 to have better manual control flights and better stabilizing in it's roll/center axis.

Dobid, 2023-01-31:
@gallonmate Yup, I'm aiming to build a similar control system based on machine learning techniques. But before going in the proper dev of that part, I just wanted to play around with the UAV and check what are the control inputs that the plane has etc... I stumbled upon that error and I was unsatisfied not understanding those NaNs.

gallonmate, 2023-01-31:
Yes I manually test flew the craft. It's very unstable but I was able to successfully fly it with gentle control inputs and constant attention/adjustments. Large aileron inputs immediately cause the craft to enter rolling spin/destabilize. I don't think there's an actual JSBSim bug here, the model is just very unstable for manual flight and could use adjustment/improvement. Another thing I noticed is that the original repo is updating JSBSim at 240hz, which generally speaking higher rates improve stability. Good luck with your work!

Dobid, 2023-01-31:
@gallonmate Thanks, for your additionnal info ! Yup, I don't think there's a bug with JSBSim either now. With that said, @seanmcleod, I think we can close this issue, is that ok  for you ?

seanmcleod, 2023-01-31:
> Some lead to stable trajectories, some other to unstable ones and NaNs... If I understand correctly, I should check the stability of my UAV to know precisely what conditions lead to unstable trajectories.

So first thing to consider is that "garbage in = garbage out", in other words if the FDM has "garbage" data then you can't trust the results you see in JSBSim. So the first thing you need to check and inspect is the FDM data in terms of what sort of coverage it has to understand under what conditions it will provide useful data.

For example, take the lift versus AoA for the 737 model, you'll see that it only provides data for AoA in the range from -11.5 degrees to +26.4 degrees.

https://github.com/JSBSim-Team/jsbsim/blob/98535e88748db3939910bd6267bcf13d297c60bf/aircraft/737/737.xml#L648-L666

Now if the AoA is outside that range then JSBSim clips the output. 

Which means JSBSim continues executing and generating output but you can't trust the output since it's operating outside the range of the FDM data. In your example you dropped the UAV with zero velocity, so pretty much immediately the AoA would've been 90 degrees, and I haven't double-check but I very much doubt your model has FDM data for that AoA. 

So if you want to have any confidence in JSBSim's output first double-check that the aircraft is being operated within the parameters for which you have FDM data. This also assumes that the FDM data is somewhat accurate for the regions it covers.

Dobid, 2023-02-01:
Understood, the x8.xml is a lot less detailed than the 737 or the cessna files. I'm guessing the FDM is not very complete since it doesn't cover a lot of regions hence can turn unstable pretty quicky, but this unstability is also backed up by @gallonmate experience of piloting the UAV. Since I don't have a wind tunnel, nor the UAV to experiment on it and get more data, I'm kinda bounded to work with the existing FDM of the x8.

bcoconni, 2023-02-04:
Should we consider this topic resolved and close this issue ?

Dobid, 2023-02-08:
Sure, thanks for the information !