JSBSim-Team/jsbsim #870: Can't build Unreal Engine plugin on Linux

demonixis, 2023-03-28:
**I'm submitting a ...**
  - [x] bug report
  - [] feature request
  - [ ] support request => **Please do not submit support request here**, see note at the top of this template.

**Describe the issue**
The Unreal Engine plugin can't be compiled on Linux after adapting the JSBSim.build.cs because of error.

**What is the current behavior?**
Here is the error log 

```bash
Building 6 actions with 6 processes...
[1/6] Compile Module.JSBSimFlightDynamicsModel.cpp
In file included from /home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Intermediate/Build/Linux/B4D820EA/UnrealEditor/Development/JSBSimFlightDynamicsModel/Module.JSBSimFlightDynamicsModel.cpp:6:
In file included from /home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/JSBSimFlightDynamicsModel/Private/JSBSimMovementComponent.cpp:16:
In file included from /home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/ThirdParty/JSBSim/Include/FGFDMExec.h:46:
In file included from /home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/ThirdParty/JSBSim/Include/models/FGPropagate.h:43:
In file included from /home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/ThirdParty/JSBSim/Include/models/FGModel.h:44:
In file included from /home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/ThirdParty/JSBSim/Include/math/FGModelFunctions.h:40:
In file included from /home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/ThirdParty/JSBSim/Include/input_output/FGPropertyReader.h:43:
/home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/ThirdParty/JSBSim/Include/simgear/props/props.hxx:1895:14: error: use of typeid requires -frtti
      return typeid(T).name();
             ^
In file included from /home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Intermediate/Build/Linux/B4D820EA/UnrealEditor/Development/JSBSimFlightDynamicsModel/Module.JSBSimFlightDynamicsModel.cpp:6:
/home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/JSBSimFlightDynamicsModel/Private/JSBSimMovementComponent.cpp:409:20: error: no matching conversion for functional-style cast from 'const TCHAR *' (aka 'const char16_t *') to 'SGPath'
                Exec->SetRootDir(SGPath(*RootDir));
                                 ^~~~~~~~~~~~~~~
/home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/ThirdParty/JSBSim/Include/simgear/misc/sg_path.hxx:68:14: note: candidate constructor not viable: no known conversion from 'const TCHAR *' (aka 'const char16_t *') to 'SGPath::PermissionChecker' (aka 'SGPath::Permissions (*)(const SGPath &)') for 1st argument
    explicit SGPath(PermissionChecker validator = NULL);
             ^
/home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/ThirdParty/JSBSim/Include/simgear/misc/sg_path.hxx:71:5: note: candidate constructor not viable: no known conversion from 'const TCHAR *' (aka 'const char16_t *') to 'const SGPath' for 1st argument
    SGPath(const SGPath& p);
    ^
/home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/ThirdParty/JSBSim/Include/simgear/misc/sg_path.hxx:79:14: note: candidate constructor not viable: no known conversion from 'const TCHAR *' (aka 'const char16_t *') to 'const std::string' (aka 'const basic_string<char, char_traits<char>, allocator<char>>') for 1st argument
    explicit SGPath( const std::string& p, PermissionChecker validator = NULL );
             ^
/home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/ThirdParty/JSBSim/Include/simgear/misc/sg_path.hxx:81:14: note: candidate constructor not viable: no known conversion from 'const TCHAR *' (aka 'const char16_t *') to 'const std::wstring' (aka 'const basic_string<wchar_t, char_traits<wchar_t>, allocator<wchar_t>>') for 1st argument
    explicit SGPath(const std::wstring& p, PermissionChecker validator = NULL);
             ^
/home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/ThirdParty/JSBSim/Include/simgear/misc/sg_path.hxx:88:5: note: candidate constructor not viable: requires at least 2 arguments, but 1 was provided
    SGPath( const SGPath& p,
    ^
In file included from /home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Intermediate/Build/Linux/B4D820EA/UnrealEditor/Development/JSBSimFlightDynamicsModel/Module.JSBSimFlightDynamicsModel.cpp:6:
/home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/JSBSimFlightDynamicsModel/Private/JSBSimMovementComponent.cpp:410:25: error: no matching conversion for functional-style cast from 'const TCHAR *' (aka 'const char16_t *') to 'SGPath'
                Exec->SetAircraftPath(SGPath(*AircraftPath));
                                      ^~~~~~~~~~~~~~~~~~~~
/home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/ThirdParty/JSBSim/Include/simgear/misc/sg_path.hxx:68:14: note: candidate constructor not viable: no known conversion from 'const TCHAR *' (aka 'const char16_t *') to 'SGPath::PermissionChecker' (aka 'SGPath::Permissions (*)(const SGPath &)') for 1st argument
    explicit SGPath(PermissionChecker validator = NULL);
             ^
/home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/ThirdParty/JSBSim/Include/simgear/misc/sg_path.hxx:71:5: note: candidate constructor not viable: no known conversion from 'const TCHAR *' (aka 'const char16_t *') to 'const SGPath' for 1st argument
    SGPath(const SGPath& p);
    ^
/home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/ThirdParty/JSBSim/Include/simgear/misc/sg_path.hxx:79:14: note: candidate constructor not viable: no known conversion from 'const TCHAR *' (aka 'const char16_t *') to 'const std::string' (aka 'const basic_string<char, char_traits<char>, allocator<char>>') for 1st argument
    explicit SGPath( const std::string& p, PermissionChecker validator = NULL );
             ^
/home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/ThirdParty/JSBSim/Include/simgear/misc/sg_path.hxx:81:14: note: candidate constructor not viable: no known conversion from 'const TCHAR *' (aka 'const char16_t *') to 'const std::wstring' (aka 'const basic_string<wchar_t, char_traits<wchar_t>, allocator<wchar_t>>') for 1st argument
    explicit SGPath(const std::wstring& p, PermissionChecker validator = NULL);
             ^
/home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/ThirdParty/JSBSim/Include/simgear/misc/sg_path.hxx:88:5: note: candidate constructor not viable: requires at least 2 arguments, but 1 was provided
    SGPath( const SGPath& p,
    ^
In file included from /home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Intermediate/Build/Linux/B4D820EA/UnrealEditor/Development/JSBSimFlightDynamicsModel/Module.JSBSimFlightDynamicsModel.cpp:6:
/home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/JSBSimFlightDynamicsModel/Private/JSBSimMovementComponent.cpp:411:23: error: no matching conversion for functional-style cast from 'const TCHAR *' (aka 'const char16_t *') to 'SGPath'
                Exec->SetEnginePath(SGPath(*EnginePath));
                                    ^~~~~~~~~~~~~~~~~~
/home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/ThirdParty/JSBSim/Include/simgear/misc/sg_path.hxx:68:14: note: candidate constructor not viable: no known conversion from 'const TCHAR *' (aka 'const char16_t *') to 'SGPath::PermissionChecker' (aka 'SGPath::Permissions (*)(const SGPath &)') for 1st argument
    explicit SGPath(PermissionChecker validator = NULL);
             ^
/home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/ThirdParty/JSBSim/Include/simgear/misc/sg_path.hxx:71:5: note: candidate constructor not viable: no known conversion from 'const TCHAR *' (aka 'const char16_t *') to 'const SGPath' for 1st argument
[... 36 lines cut ...]
In file included from /home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/ThirdParty/JSBSim/Include/models/FGLGear.h:43:
In file included from /home/yann/Projects/jsbsim/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/ThirdParty/JSBSim/Include/models/propulsion/FGForce.h:62:
[... cut ...]
[5/6] Link (lld) libUnrealEditor-JSBSimFlightDynamicsModelEditor.so cancelled
[6/6] WriteMetadata UEReferenceAppEditor.target cancelled
``` 

**What is the expected behavior?**
The expected behaviour is to have the plugin compiling on Linux

**What is the motivation / use case for changing the behavior?**
Using JSBSim for Unreal on the Linux operating system

**Please tell us about your environment:**
 - OS Ubuntu 23.04
 - JSBSim version (Latest)

You can follow [my work on my fork](https://github.com/demonixis?tab=repositories)

demonixis, 2023-03-29:
I fixed many problems on [my branch](https://github.com/demonixis/jsbsim) now the problem is about the typeid keyword that breaks the compilation on Linux and Mac. It's not possible to easilly add rtti without rebuilding the game engine itself.
A quick fix to validate that the plugin can compile on mac/linux is to hack the prop.hxx file like this

```cplusplus
static const char* name() { return ""; }// typeid(T).name(); }
```

But it's not good... So I'm asking for help to find a way to replace typeid OR make it work with Unreal.

seanmcleod, 2023-03-29:
So are you saying that the Unreal engine on Windows is built with rtti enabled but not by default on Mac and Linux?

bcoconni, 2023-03-29:
@demonixis, actually you can safely delete the code lines 1881-1929 in `src/simgear/props/props.hxx`, these are unused by JSBSim.
https://github.com/JSBSim-Team/jsbsim/blob/10c6e52e74452c31cd8dc2b857e2abbfe7b46daa/src/simgear/props/props.hxx#L1881-L1929
Once you're happy with your code (including deleting the lines mentioned above), you can submit a PR.

closed 2023-04-03 by demonixis