JSBSim-Team/jsbsim #880: Missing support for castering nosewheel with angle limiter.

MariuszXC, 2023-04-05:
**I'm submitting a ...**
  - [ ] bug report
  - [x] feature request

**Describe the issue**
[quotes from documentation] Currently JSBSim defines a castering wheel as "free to rotate and their angle is computed by JSBSim at any time from the local tangential speed" and  "For any value of <max_steer> other than 0 and 360 degrees, the gear is assumed to be steerable". 

There is no support for cases, where nose wheel is free castering (not steerable), but has mechanical end-stops to limit the available rotation angle. Please add such support, in the simplest form it could be an additional xml tag, for example:

<max_steer>360</max_steer>
<l_r_limit>30</l_r_limit>

**What is the current behavior?**
Currently there is no way to specify rotation limits on a free castering whel.

**What is the expected behavior?**
Having a method to instruct JSBSim to limit the computed rotation angle.
This probably would have an impact on forces transferred from ground friction.

**What is the motivation / use case for changing the behavior?**
Better support for real use cases. An example is Diamond DA40NG.

**Please tell us about your environment:**
JSBSim version as included in current FlightGear sim.

**Other information**
In some aircraft the rotation axis is also slightly angled fore-aft. This, together with the force from supported mass creates an additional centering force-moment. It would be perfect to include this effect in the angle computation.

bcoconni, 2023-04-08:
> **What is the motivation / use case for changing the behavior?**
Better support for real use cases. An example is Diamond DA40NG.

Given the landing gears configuration of this aircraft I doubt very much that its nose leg has a caster wheel because that would make it unstable... and deadly during take off and landing.

[image]

Caster wheels are always at the rear of an aircraft for that very reason and I don't see much point in limiting the range of their rotating angle.

Are you sure you mean a [caster wheel](https://en.wikipedia.org/wiki/Caster) ?

MariuszXC, 2023-04-17:
It may be a language barrier, but I meant an unsteered front wheel, free to rotate around its (almost)vertical axis. 
Of the fact it is unsteered I am 100% certain. Ditto about +/- 30deg end-stops.

Please refer to DA40NG AMM: http://support.diamond-air.at/fileadmin/uploads/files/after_sales_support/DA40%20New%20Generation/Airplane_Maintenance_Manual/Basic_Manual/60215-r3-DA40-NG-AMM-complete.pdf
Section 32-20 (pdf document page 950 and onwards).

gallonmate, 2023-04-18:
JSBSim already supports caster wheels with a angle limit. At least it did. 
Try this in the aircraft file

```
<castered> 1.0 </castered>
<max_steer unit="DEG"> 30 </max_steer>
```

bcoconni, 2023-04-18:
> It may be a language barrier, but I meant an unsteered front wheel, free to rotate around its (almost)vertical rotation axis. Of the fact it is unsteered I am 100% certain. Ditto about +/- 30deg end-stops.
> 
> Please refer to DA40NG AMM: http://support.diamond-air.at/fileadmin/uploads/files/after_sales_support/DA40%20New%20Generation/Airplane_Maintenance_Manual/Basic_Manual/60215-r3-DA40-NG-AMM-complete.pdf Section 32-20 (pdf document page 950 and onwards).

Thanks for the link @MariuszXC. I was indeed wrong in my assumptions.

bcoconni, 2023-04-18:
> ```xml
> <castered> 1.0 </castered>
> <max_steer unit="DEG"> 30 </max_steer>
> ```
> 
> Edit: Nevermind, this doesn't work as a limit. I misunderstood some old code.

`<max_steer>` does not work as a limit for caster wheel **yet** but it could ! 😄 

Your suggestion provides the syntax that we need to address the feature requested by @MariuszXC. I'd say we only need to modify the method `FGLGear::ComputeSteeringAngle` so that is `fabs(SteerAngle)` is kept below `MaxSteerAngle`. Something like:

```c++
void FGLGear::ComputeSteeringAngle(void)
{
  if (Castered) {
      // Check that the speed is non-null otherwise keep the current angle
      if (vWhlVelVec.Magnitude(eX,eY) > 0.1)
        SteerAngle = atan2(vWhlVelVec(eY), fabs(vWhlVelVec(eX)));

      SteerAngle = Constrain(-MaxSteerAngle, SteerAngle, MaxSteerAngle);
  }
}
```

MariuszXC, 2023-04-21:
I would have to dive deeper into the unfamiliar code, so I'll better simply ask here.
What will happen when the nose wheel angle is constrained as in the code above, but the aircraft trajectory path is such, that a larger angle would have been computed? Will an additional force-moment due to increased ground friction be calculated and included in the sum of forces acting on the ac body?

bcoconni, 2023-05-09:
> What will happen when the nose wheel angle is constrained as in the code above, but the aircraft trajectory path is such, that a larger angle would have been computed?

The wheel will slip, a situation that JSBSim is perfectly capable of modeling.
> Will an additional force-moment due to increased ground friction be calculated and included in the sum of forces acting on the ac body?

Yes. JSBSim uses a friction model to compute the forces that are induced by landing gears and their resulting force and moment. The code is located in `src/models/FGAcceleration.cpp` if you'd like to review it:

https://github.com/JSBSim-Team/jsbsim/blob/db830c68c77d9b56e86bbe5811f40e22cb0d0acf/src/models/FGAccelerations.cpp#L219-L232

MariuszXC, 2023-06-02:
Noted and appreciated. Thanks.