JSBSim-Team/jsbsim #882: param name new name in the form "hostname:port/proto" is not right  in  FGOutputSocket::SetOutputName

fishpro, 2023-04-08:
**Describe the issue**
FGOutputSocket.cpp
```

void FGOutputSocket::SetOutputName(const string& fname)
{
  // tokenize the output name
  size_t dot_pos = fname.find(':', 0);
  size_t slash_pos = fname.find('/', 0);
  
  string name = fname.substr(0, dot_pos);
  
  string proto = "TCP";
  if(dot_pos + 1 < slash_pos)
    proto = fname.substr(dot_pos + 1, slash_pos - dot_pos - 1);//bug not right dot_pos 
  
  string port = "1138";
  if(slash_pos < string::npos)
    port = fname.substr(slash_pos + 1, string::npos); //bug not right slash_pos 
  
  // set the model name
  Name = name + ":" + port + "/" + proto;
  
  // set the socket params
  SockName = name;
  
  SockPort = atoi(port.c_str());
  
  if (to_upper(proto) == "UDP")
    SockProtocol = FGfdmSocket::ptUDP;
  else // Default to TCP
    SockProtocol = FGfdmSocket::ptTCP;
}
```
param name new name in the form "hostname:port/proto" is not right 
but  "hostname:proto/port" is right?

seanmcleod, 2023-04-08:
Okay, so what you're pointing out is that when parsing the input parameter `fname` the assumption is that it has the form:

`hostname:proto/port`

But when then generating the model name, `Name`, from the components based on parsing, it's generating the model name as:

`hostname:port/proto`

fishpro, 2023-04-08:
> 好的，所以你要指出的是，在解析输入参数时，`fname`假设它具有以下形式：
> 
> `hostname:proto/port`
> 
> `Name`但是当基于解析从组件生成模型名称时，它生成的模型名称为：
> 
> `hostname:port/proto`

yes,
That's what it means

seanmcleod, 2023-04-08:
Okay and looking at the code that calls `FGOutputSocket::SetOutputName()` the correct format is:

`hostname:proto/port`.

https://github.com/JSBSim-Team/jsbsim/blob/92fa1eaa29322dc3f6ab55587161e55a6b24d36d/src/input_output/FGOutputSocket.cpp#L115-L122

So the model name `Name` is incorrect. Not sure off-hand where the model name `Name` is used, possibly for display purposes somewhere.

I see the comment in the header file is incorrect, it has swapped `port` and `protocol` around.

So we should update the comment in the header file and updated the `Name`. 

Do you want to submit a pull request (PR) or do you want me to go ahead and generate a pull request?