JSBSim-Team/jsbsim #911: Crashes after adding an xml path to a component in the engine

lhy0108, 2023-05-11:
[image]
[image]
[image]
[image]

**What is the reason?**

lhy0108, 2023-05-11:
**What is the reason?**

gallonmate, 2023-05-11:
The title of the discussion says "Crashes after adding an xml path to a component in the engine" but the post shows nothing related. Can you please describe or show what changes you made that lead to the crash.

Also you started a duplicate Issue and Discussion. Please use one or the other, not both.

lhy0108, 2023-05-12:
Hello, crash occurred when I ran the engine after generating the JSBSimForUnreal.sln file and the JSBSim component could not read the xml file

lhy0108, 2023-05-12:
When my Blueprint role uses the JSBSimMovement component, it crashes because it can't read the xml file

gallonmate, 2023-05-12:
Well if you didn't change anything in the original project demo file, then it shouldn't crash. 
Make sure that JSBSimForUnreal.sln was built as Release and not Debug.

lhy0108, 2023-05-12:
You can see in the image above that after I generated JSBSimForUnreal.sln, it automatically added the JSBSim file, but it still crashed when I used UE to read the aircraft model

lhy0108, 2023-05-12:
Oh, thank you. I tried to rebuild  JSBSimForUnreal.sln

lhy0108, 2023-05-12:
The problem has been solved. Thank you for your patient reply

gallonmate, 2023-05-12:
Can you let us know what solved the issue? Was the problem with Debug build and Release solved it?
If so, it's a known issue and I'm working to fix the Debug build in the near future.

bcoconni, 2023-05-12:
@lhy0108 
> The problem has been solved. Thank you for your patient reply

Your issue may be resolved however patience is not a virtue that you've displayed as you've posted 5 messages in less than 24h asking the exact same question:
* [May 10th, 2023, 10:19UTC](https://github.com/JSBSim-Team/jsbsim/issues/626#issuecomment-1541876385) Issue #626
* [May 11th, 2023, 07:56UTC](https://github.com/JSBSim-Team/jsbsim/discussions/910#discussion-5184213)
* May 11th, 2023, 07:59UTC (deleted by yourself)
* [May 11th, 2023, 08:01UTC](https://github.com/JSBSim-Team/jsbsim/issues/911#issue-1705281258)
* [May 11th, 2023, 08:40UTC](https://github.com/JSBSim-Team/jsbsim/issues/911#issuecomment-1543582941)

And would like to stress the fact that the 4 last messages have been posted in less than 45 minutes. This is message bombing (and spamming our mailboxes) which by all measure is extremely rude. Not to mention that you've disregarded the caution messages that request not to use issues for support requests.

For your information, JSBSim is maintained by a small team of volunteers on their free time so we cannot be 24/7 online responding questions and support requests. So we would appreciate very much that in the future you use wisely the communication tools that are provided by the project and most of all be patient.

Thanks.

closed 2023-05-12 by bcoconni