JSBSim-Team/jsbsim #918: Trim Errors Happening In RHEL8 On Models That Worked On RHEL7

omidnabi, 2023-06-01:
We just transitioned to redhat 8, and all of our models started to throw trim errors. I was just wondering if anyone else has experienced this problem?

bcoconni, 2023-06-03:
Hi @omidnabi, if this is just an open question then this should be moved to [Discussions](https://github.com/JSBSim-Team/jsbsim/discussions). Otherwise we need more details to answer your question.

omidnabi, 2023-06-06:
Definitely not an open question. I'll ask my colleague to post the details. But in short, we have been running the same models on RHEL7 for years with different initial conditions and the trim has been working just fine. We rand an identical testcase that runs on RHEL7 on RHEL8 and all the trims fail. We also tried to update our JSBSim to the latest version from the first official release, and that didn't fix the problem either.

agodemar, 2023-06-07:
@omidnabi the partial tableData you mention above does not clarify the context of the error. Some refinement have been made on the way tabular functions are handled. You should provide us with a reproducible example in order to spot the problem.

omidnabi, 2023-06-07:
> @omidnabi the partial tableData you mention above does not clarify the context of the error. Some refinement have been made on the way tabular functions are handled. You should provide us with a reproducible example in order to spot the problem.

The problem is the same models run on RHEL7 using the same JSBSim version. I was just wondering if you have support for RHEL8 for  JSBSim version 1.0.0.

seanmcleod, 2023-06-07:
> I was just wondering if you have support for RHEL8

We don't specifically perform test builds and tests (which includes tests that perform trim) for RHEL8, but in general there should be no reason it shouldn't run successfully on RHEL8.

You can see the different test builds that we perform for each commit here:

https://github.com/JSBSim-Team/jsbsim/pull/908/checks

Two versions of Ubuntu, two builds on Windows using 2 different compilers, MacOSX (Intel and Arm).

So in general we cover a number of different OS versions and a number of different compilers.

omidnabi, 2023-06-07:
> @omidnabi if you're closing this does this mean you figured out the issue with RHEL8?

Not quite. But based on your answers,  I'm starting to think the problem is being caused but the JSBSim interface we wrote. We are building JSBSim using C++11 on RHEL7 and RHEL8 but the interface is being built using C++11 on RHEL7 and C++17 on RHEL8. I'm guessing something is messing with the initial conditions passed to JSBSim. I will send an update if we can figure out the problem.

bcoconni, 2023-06-07:
Our trim algorithm is known to be sensitive to the initial conditions set before trimming (see issue #360 for instance) so your issue might possibly be linked to an upgrade of some maths library (including `libstdc++` itself) ?

However I'm mostly writing code on a Linux platform and as far as I remember I never came across a trim error being raised following a compiler upgrade (and since I'm involved with JSBSim since 2008-ish I have seen quite a number of compiler and Linux distrib upgrades 😄). On the other hand, I've already seen trims failing on Windows while succeeding on Linux, or vice-versa. As I mentioned above, our trim algorithm is **very** sensitive to initial conditions so different roundoffs might make a difference.

bcoconni, 2023-06-07:
Also I guess you already tried rebuilding from scratch ? All sort of errors can result from linking files compiled with different versions of the same compiler. I might make a fool of myself by mentioning that but this has happened to me quite a number of times to waste time chasing a bug until I realize that I should rebuild JSBSim from scratch because the C++ compiler has had a major update.

omidnabi, 2023-06-07:
Yes, I'm hoping it's just a problem caused by the interface math
functions.  And yes, we always build from scratch to avoid silly mistakes.
I wrote a Simulink interface for JSBSim which I use it for fine tuning the
models I create for JSBSim and its autopilot,  and I have noticed how
sensitive the trim is over and over again 😅