JSBSim-Team/jsbsim #919: Updates to gitignore to tidy up git status after build

DA5H8Y, 2023-06-02:
**I'm submitting a ...**
  - [ ] bug report
  - [X] feature request
  - [ ] support request => **Please do not submit support request here**, see note at the top of this template.

**Describe the issue**
After building JSBSim from the master branch, if one performs a ```git status``` to see if there have been any changes then most of the interment build files appear as working tree changes.

**What is the current behavior?**
After building JSBSim from the master branch, performing a ```git status``` to check for changes results in many of the intermediate build files being listed as local changes.

**What is the expected behavior?**
Intermediate build files should be in the .gitignore so that this behavior is not observed.

**What is the motivation / use case for changing the behavior?**
Many user will include automated process to keep the simulation software up-to-date or to alert engineers to changes that they may wish to consider using.

**Please tell us about your environment:**
Ubuntu 22.04, JSBSim<master> branch

**Other information**
Not Applicable

bcoconni, 2023-06-03:
Could you please provide the message that git is returning when `git status` is run ?

DA5H8Y, 2023-06-06:
I have found out that the build was being conducted in the root folder hence, not being caught by the .gitignore pattern ```build*```

closed 2023-06-06 by DA5H8Y