JSBSim-Team/jsbsim #922: CMAKE Configure Error on MacOS (M1 aarch64)

cryptik, 2023-06-06:
***

**I'm submitting a ...**
  - [ X] bug report
  - [ ] feature request
  - [ ] support request => **Please do not submit support request here**, see note at the top of this template.

**Describe the issue**

When running the cmake step on jsbsim build, the configure step appears to generate all of the necessary files; however, it generates an error relative to a missing class function (`JSBSim::FGPropertyManager::GetName`) 

**What is the current behavior?**

Running the CMAKE command as follows:
`cmake -DCMAKE_CXX_FLAGS_RELEASE="-O3 -march=native -mtune=native" -DCMAKE_C_FLAGS_RELEASE="-O3 -march=native -mtune=native" -DCMAKE_BUILD_TYPE=Release ..`
results in the following errors:
```
-- Found Cython: /opt/homebrew/Caskroom/miniconda/base/envs/ardu/bin/cython  
Traceback (most recent call last):
  File "/Users/cryptik/Development/other/jsbsim/build/python/doxy2pydocs.py", line 201, in <module>
    raise IOError(
OSError: File classJSBSim_1_1FGPropertyManager.xml does not contain JSBSim::FGPropertyManager::GetName
-- Found CxxTest: /opt/homebrew/include  
CMake Warning at CMakeLists.txt:149 (message):
  -- Could NOT find LSB_Release (missing: LSB_RELEASE_EXECUTABLE)
-- Configuring done (42.9s)
-- Generating done (0.1s)
-- Build files have been written to: /Users/cryptik/Development/other/jsbsim/build

```

**What is the expected behavior?**

No OSError should be generated on build

**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

OS - macOS (Ventura, M1 arm)

**Other information**

bcoconni, 2023-06-10:
As a temporary fix, you should be able to work around this issue by disabling the generation of documentation. For that purpose, you need to `-DBUILD_DOCS=OFF` to the CMake command:
```bash
cmake -DCMAKE_CXX_FLAGS_RELEASE="-O3 -march=native -mtune=native" -DCMAKE_C_FLAGS_RELEASE="-O3 -march=native -mtune=native" -DBUILD_DOCS=OFF -DCMAKE_BUILD_TYPE=Release ..
```

bcoconni, 2024-03-03:
Sorry @cryptik, I had lost sight of this ticket.

The issue you have reported should now be fixed by the PR #1048 which has been merged in the `master` branch. JSBSim should now compile whether the flag `BUILD_DOCS` is set to `ON` or `OFF`.

closed 2024-04-28 by bcoconni