JSBSim-Team/jsbsim #931: JSBSim interface to Matlab on Windows: throttle command from joystick doesn't work

jevta23, 2023-06-25:
Hello guys!

I have downloaded and installed on Windows the [JSBSim release v1.1.13]. I have decided for this simulation framework because I can easily configure fixed-wing planes (with my own parameters) which I want to test without the need for modelling aircraft dynamics from the ground. Furthermore, I plan to use this software for the following tasks:

1. Take-off the plane from the ground and fly it by using the joystick.
2. Starting the flight from the desired altitude (will be set as initial value) and after that fly the plane by using joystick.
3. Generate global autonomous task such as --> take-off to the desired altitude and fly to the previously defined waypoints.

Firstly, I have placed some blocks necessary for receiving commands from the joystick.

[image]

After that, I have added blocks for simulating fixed-wing UAV within the Unreal Engine environment.

[image]

I have tried to simulate flight on c172p by using the following files within the S-function block:

[image]

In order to do this, I had to modify one row in c1772.xml file (script file) as this:

<use aircraft="c172p" initialize="reset01"/>

I have figured out that plane only responds to aileron and/оr elevator commands when I set vt value to be non zero, as well as initial altitude for example to be 1000 ft. However, model doesn't react on throttle input, and therefore state variables (position/long-gc-deg and position/lat-gc-deg) have the same values as given within the initial "reset01.xml" file. The mentioned variables can be seen bellow:

[image]

Commands from the joystick can be seen bellow:

[image]

Does anybody know why throttle commands cannot be seen by the model, or did I miss something in preparing my Simulink model?

Thanks in advance!

Kind regards,
Đorđe Jevtić
Ph.D. student & Junior Research Assistant

This repository's issues are reserved for feature requests and bug reports. Issues that are neither bug reports or feature requests will be transferred to [Discussions](https://github.com/JSBSim-Team/jsbsim/discussions) by the team and the issue will be locked and closed.

***

**I'm submitting a ...**
  - [ ] bug report
  - [ ] feature request
  - [ ] support request => **Please do not submit support request here**, see note at the top of this template.

**Describe the issue**

**What is the current behavior?**

**What is the expected behavior?**

**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

**Other information**

seanmcleod, 2023-06-25:
> I have figured out that plane only responds to aileron and/оr elevator commands when I set vt value to be non zero

If `vt == 0` why would you expect a moment to be generated by the aileron or elevator?

In terms of the throttle, double-check that the engine is running. Script `c1722.xml` does have an 'Engine Start' event, but there is also an 'Engine starved' event. 

You can also use the following in the reset file to ensure that the engines are all started and running.

```xml
<running> -1 </running>
```

jevta23, 2023-06-25:
Dear @seanmcleod 

Thanks for your answer! 

I thought that the moment will be generated when throttle command is for example set to 1 (after some time), for example when the plane have to take-off. In such case, all velocities are equal to zero at the beginning, and after some time, aileron and/оr elevator commands will affect plane's behavior.

I have tested now new script model c172_cruise_8K.xml (there is event engine start in it). As in previous model I have to modify one row: `<use aircraft="c172p" initialize="reset01"/>`

I have made modification within the reset xml file as you suggested.

```
<?xml version="1.0"?>
<initialize name="reset01">
  <!--
    This file sets up the aircraft to start off
    from the runway in preparation for takeoff.
  -->
  <ubody>       0.0 </ubody>
  <vbody>       0.0 </vbody>
  <wbody>       0.0 </wbody>
  <latitude unit="DEG">   47.0 </latitude>
  <longitude unit="DEG"> 122.0 </longitude>
  <phi>         0.0 </phi>
  <theta>       0.0 </theta>
  <psi unit="DEG">       150.0 </psi>
  <altitude>    1000.00 </altitude>
  <hwind>        0.0 </hwind>
  <xwind>        0.0 </xwind>
  <vt>           0.0 </vt>
  <gamma unit="DEG">        3.0 </gamma>

  <running> -1 </running>

</initialize>
```
You can see my S-function block now:

[image]

Joystick commands are given bellow:

[image]

And again, results are the same:

[image]

It seems like the engine still won't start.

Kind regards,
Đorđe

seanmcleod, 2023-06-25:
Have you seen it working if you use the default 737 example?

Also, is the State Vector you're showing definitely being updated as opposed to just showing the initial state? Look at the first 3 elements of the State Vector, they represent the UVW velocities in fps, and all 3 are non-zero. So if they're non-zero, whether the engine is running or not over time the aircraft's lat/lon will change.

jevta23, 2023-06-25:
Yes, I have checked with the default 737 example, but the results are the same.

I have looked the first 3 elements and it has sense what you are saying. There should be the change over time for lateral/lon as well. Maybe I used non-adequate values for plot, I really don't understand.

```
<?xml version="1.0"?>
<s_function_config>
    <input>
        <property> fcs/throttle-cmd-norm </property>
        <property> fcs/aileron-cmd-norm </property>
        <property> fcs/elevator-cmd-norm </property>
        <property> fcs/rudder-cmd-norm </property>
        <property> fcs/mixture-cmd-norm[0] </property>
        <property> propulsion/engine[0]/set-running </property>
        <property> fcs/flap-cmd-norm </property>
        <property> gear/gear-cmd-norm </property>
    </input>
    <outputs>
        <output name="State">
            <property> velocities/u-fps </property>
            <property> velocities/v-fps </property>
            <property> velocities/w-fps </property>
            <property> velocities/p-rad_sec </property>
            <property> velocities/q-rad_sec </property>
            <property> velocities/r-rad_sec </property>
            <property> position/h-sl-ft </property>
            <property> position/long-gc-deg </property>
            <property> position/lat-gc-deg </property>
            <property> attitude/phi-rad </property>
            <property> attitude/theta-rad </property>
            <property> attitude/psi-rad </property>
        </output>
        <output name="Control">
            <property> fcs/throttle-pos-norm[0] </property>
            <property> fcs/left-aileron-pos-rad </property>
            <property> fcs/right-aileron-pos-rad </property>
            <property> fcs/elevator-pos-norm </property>
            <property> fcs/rudder-pos-norm </property>
            <property> fcs/flap-pos-norm </property>
            <property> fcs/speedbrake-pos-norm </property>
            <property> fcs/spoiler-pos-norm </property>
            <property> gear/gear-cmd-norm </property>
        </output>
        <output name="Pilot">
            <property> forces/load-factor </property>
            <property> aero/alpha-rad </property>
            <property> aero/alphadot-rad_sec </property>
            <property> aero/beta-rad </property>
            <property> aero/betadot-rad_sec </property>
            <property> velocities/vc-fps </property>
            <property> velocities/vc-kts </property>
            <property> velocities/vtrue-fps </property>
            <property> velocities/vg-fps </property>
            <property> velocities/mach </property>
            <property> velocities/h-dot-fps </property>
            <property> aero/qbar-psf </property>
            <property> fcs/elevator-cmd-norm </property>
        </output>
    </outputs>
    <!--This is an optional parameter-->
    <!--The only properties in this should be atmosphere related-->
    <!--
    <weather>
        <property> atmosphere/wind-north-fps </property>
    </weather>
    -->
</s_function_config>
```
Maybe I have to express these two values in another reference frame to see these changes properly. As I can see these are long-gc-deg, and lon-gc-deg, while the reference frame within the Simulink block is inertial and it requires data in meters.

Yes, I have found that the problem is probably in the definition of reference frames.

bcoconni, 2023-07-15:
I'd suggest that you add the property `forces/fbx-prop-lbs` to your state: 
* If this property reports a value of 0 then the engine is not started and you should investigate why.
* If this property reports a non zero value then your engine is ignoring the throttle command and you should investigate why.

Also note that:
* There are as many pairs of throttle properties as your aircraft has engines.
* For each engine there is a property `fcs/throttle-pos-norm[#]` and a property `fcs/throttle-cmd-norm[#]`.
* Engines are using `fcs/throttle-pos-norm[#]` properties to drive their thrust so make sure these properties are correctly updated.

thunder-wizard, 2025-04-07:
Hello there.
I think you have a problem in display setting.
Set the Numerical display format to long. Your display is working at short so you can see the change when you flight for more than 10 meters!
Thanks

[image]

[image]