JSBSim-Team/jsbsim #963: Building for Unreal 5 on Linux not working

RomanBapst, 2023-09-15:
**I'm submitting a ...**
  - [ ] bug report
  - [ ] feature request
  - [ ] support request => **Please do not submit support request here**, see note at the top of this template.

**Describe the issue**
I tried to build an run JSBSim for Unreal 5.3.0 on Ubuntu 20.04. When launching `UnrealEditor` it complained about `libJSBSim.so` not being found. I saw on a recent PR that support for Linux might not quite be there and that there supposedly are some linking issues.

In my case I was able to resolve the issue by adding the path to  `libJSBSim.so` to `LD_LIBRARY_PATH`.

Just wanted to report this in case it's helpful.

RomanBapst, 2023-09-15:
[image]

bcoconni, 2023-09-16:
Hi @RomanBapst, thanks for reporting the issue.

@demonixis could you please comment on this ?

demonixis, 2023-09-16:
Do you compile it using clang? Can you share the logs please?

RomanBapst, 2023-09-19:
@demonixis Yes, I was using clang. Do you need anything verbose turned on or just the standard logs?

demonixis, 2023-09-23:
For a reason I don't understand, cmake produces 3 files

- libJSBSim.so(Symlink to file)
- libJSBSim.so.1(Symlink to file)
- libJSBSim.so.1.2.0.dev1

You've to copy those 3 files for Unreal to load properly. Renaming libJSBSim.so.1.2.dev1 into libJSBSim.so doesn't work.
I'll update the README to reflect that.

demonixis, 2023-10-25:
Hey, any news about this?

bcoconni, 2023-11-11:
This issue is being closed because the issuer is not responding.

closed 2023-11-11 by bcoconni