JSBSim-Team/jsbsim #989: Linearization and simplex trim from console executable and xml script

vtpasquale, 2023-11-20:
**I'm submitting a ...**
  - [ ] bug report
  - [x] feature request
  - [ ] support request => **Please do not submit support request here**, see note at the top of this template.

**Describe the issue**

- Previous versions of JSBSim appear to have supported linearization and simplex trim from the console executable and xml scripts.

- For example, this current xml script attempts to access both features in the same run: scripts/[737_cruise_steady_turn_simplex.xml](https://github.com/JSBSim-Team/jsbsim/blob/master/scripts/737_cruise_steady_turn_simplex.xml) but fails because the xml interface to the features has been discontinued. 

- The script interface to xml was removed after this version of [FGFDMExec.cpp](https://github.com/JSBSim-Team/jsbsim/blob/3b44b79e817a85966aedf800e8c185217a7c1ade/src/FGFDMExec.cpp); comments suggest that the features were not working at the time of removal. 

- JSBSim linearization features have since been resurrected (#227, #370) and exposed through the Python interface. I'm unsure if simplex trim features have been resurrected; the corresponding classes are still in the source. 

- I attempted to fork JSBSim and borrow a few lines of code from previous versions of FGFDMExec.cpp and FGFDMExec.h to bring the functionality back, but I was unsuccessful. I imagine this might be easy for an experienced JSBSim developer.

**What is the current behavior?**

When running:
```
JSBSim --script=737_cruise_steady_turn_simplex
```

The run fails with the following console error:
```
FATAL ERROR: JSBSim terminated with an exception.
The message was: No property, "simulation/do_simplex_trim" is defined.
```

Simple trim can be used instead of simplex trim by changing
`<set name="simulation/do_simplex_trim" value="1"/>`
to
`<set name="simulation/do_simple_trim" value="1"/>`
in
`737_cruise_steady_turn_simplex.xml`.
 
In that case, when running
```
JSBSim --script=737_cruise_steady_turn_simplex
```
The run fails with the following console error:
```
FATAL ERROR: JSBSim terminated with an exception.
The message was: No property, "simulation/do_linearization" is defined.
```

**What is the expected behavior?**

- Linearization: FGLinearization::WriteScicoslab to write the matrices to a file or to the console.
- Simplex trim to work like simple trim.

**What is the motivation / use case for changing the behavior?**
- Expose the features from console executable and xml script.

**Please tell us about your environment:**

- Windows and Linux
- JSBSim v1.2.0

**Other information**

- I don't currently have a need for simplex trim, so please prioritize linearization over simplex trim. However, I think there is a similar path to resurrecting both features, so it might make sense to add them together. 

- I understand that trim features can be problematic, so I understand if you want less exposure of the simplex trim feature.

- I can help provide more test cases, but I think `737_cruise_steady_turn_simplex.xml` is likely sufficient

agodemar, 2023-11-20:
@vtpasquale thank you for reaching out and for the detailed feature request.
Please have a look at [this discussion: Update FGTrimAnalysis wrt latest JSBSim Release #988](https://github.com/JSBSim-Team/jsbsim/discussions/988), where a user is working on something similar.

vtpasquale, 2023-11-20:
Thanks for the quick response, @agodemar . It's interesting to see the parallel work. The trim feature (`'simulation/do_simple_trim'`) being used by @rishi1906 is currently accessible from the console executable and with xml scripts. That specific trim feature has been working OK for me. I don't believe resolution of #988 will simultaneously resolve my request.

I'm most interested in the Linearization feature becoming available through the console executable and xml scripts, as it was previously. That feature is currently accessible through Python. 

I think the alternative _simplex_ trim capability is also currently available through Python. I don’t currently need the simplex trim capability, since the standard trim capability is working OK for me, but it might make sense to add console executable access to the simplex trim capability at the same time as the Linearization feature (since console executable access was removed simultaneously for both features).

vtpasquale, 2023-11-20:
I think the required changes may be quite small. You can see the diff of my attempt here:
https://github.com/JSBSim-Team/jsbsim/compare/master...vtpasquale:jsbsim:dev

The build is failing 
```
/home/aricciar/projects/jsbsimFork/jsbsim/src/FGFDMExec.cpp: In constructor ‘JSBSim::FGFDMExec::FGFDMExec(JSBSim::FGPropertyManager*, std::shared_ptr<unsigned int>)’:
/home/aricciar/projects/jsbsimFork/jsbsim/src/FGFDMExec.cpp:164:16: error: no matching function for call to ‘JSBSim::FGPropertyManager::Tie(const char [28], JSBSim::FGFDMExec*, void (JSBSim::FGFDMExec::*)(int))’
  164 |   instance->Tie("simulation/do_linearization", this, &FGFDMExec::DoLinearization);
      |   ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/aricciar/projects/jsbsimFork/jsbsim/src/input_output/FGPropertyReader.h:44,
                 from /home/aricciar/projects/jsbsimFork/jsbsim/src/math/FGModelFunctions.h:40,
                 from /home/aricciar/projects/jsbsimFork/jsbsim/src/models/FGModel.h:44,
                 from /home/aricciar/projects/jsbsimFork/jsbsim/src/models/FGPropagate.h:43,
                 from /home/aricciar/projects/jsbsimFork/jsbsim/src/FGFDMExec.h:46,
                 from /home/aricciar/projects/jsbsimFork/jsbsim/src/FGFDMExec.cpp:47:
```
I added a `&FGFDMExec::DoLinearization` function (modified from earlier versions), but I think there may be an issue with the input arguments lining up with what `FGPropertyManager::Tie` is specifying, and my inexperience is showing.

seanmcleod, 2023-11-20:
You're close, use the same setup as you have for `do_simplex_trim`, i.e.

```c++
instance->Tie("simulation/do_linearization", this, (iPMF)0, &FGFDMExec::DoLinearization);
```

Basically you left out the argument for the getter.

vtpasquale, 2023-11-20:
Thanks, @seanmcleod. I made some updates:
https://github.com/JSBSim-Team/jsbsim/compare/master...vtpasquale:jsbsim:dev

But I'm still getting a similar error:
```
/home/aricciar/projects/jsbsimFork/jsbsim/src/FGFDMExec.cpp: In constructor ‘JSBSim::FGFDMExec::FGFDMExec(JSBSim::FGPropertyManager*, std::shared_ptr<unsigned int>)’:
/home/aricciar/projects/jsbsimFork/jsbsim/src/FGFDMExec.cpp:164:16: error: no matching function for call to ‘JSBSim::FGPropertyManager::Tie(const char [28], JSBSim::FGFDMExec*, int (JSBSim::FGFDMExec::*)() const, void (JSBSim::FGFDMExec::*)())’
  164 |   instance->Tie("simulation/do_linearization", this, (iPMF)0, &FGFDMExec::DoLinearization);
      |   ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/aricciar/projects/jsbsimFork/jsbsim/src/input_output/FGPropertyReader.h:44,
                 from /home/aricciar/projects/jsbsimFork/jsbsim/src/math/FGModelFunctions.h:40,
                 from /home/aricciar/projects/jsbsimFork/jsbsim/src/models/FGModel.h:44,
                 from /home/aricciar/projects/jsbsimFork/jsbsim/src/models/FGPropagate.h:43,
                 from /home/aricciar/projects/jsbsimFork/jsbsim/src/FGFDMExec.h:46,
                 from /home/aricciar/projects/jsbsimFork/jsbsim/src/FGFDMExec.cpp:47:
```
Any suggestions?

seanmcleod, 2023-11-20:
>  I made some updates

Too many updates? :wink:

I see you also changed the setter to take no/void arguments?

There are no compile issues with your `DoSimplexTrim(int)` right?

seanmcleod, 2023-11-20:
To start off I suggest keeping it simpler. Comment out the simplex trim stuff. You had mentioned your primary goal was the linearization.

So you need:

```c++
instance->Tie("simulation/do_linearization", this, (iPMF)0, &FGFDMExec::DoLinearization);

void FGFDMExec::DoLinearization(int)
{
...
}
```

And confirm whether you can get that to compile and link.

seanmcleod, 2023-11-20:
I did a quick test and was able to compile and link without any issues.

```c++
instance->Tie("simulation/do_linearization", this, (iPMF)0, &FGFDMExec::DoLinearization);

void FGFDMExec::DoLinearization(int)
{
  FGLinearization lin(this);
  lin.WriteScicoslab();
}
```

vtpasquale, 2023-11-20:
Nice! I'll give that a shot. Was that without the simplex trim code included?

I had not tried commenting out this line:
`// instance->Tie("simulation/do_linearization", this, (iPMF)0, &FGFDMExec::DoLinearization);`

The previous error is passed over when that line is commented out (I'll try updating the function as you did instead). I get to linking errors, but after starting to chase those down it was seeming like the simplex trim option is not maintained and will be a harder fix.

seanmcleod, 2023-11-20:
Correct, I haven't done anything to try and add in the simplex trim code.

The only other line I added to `FGFDMExec.cpp` was:

```c++
#include "initialization/FGLinearization.h"
```

vtpasquale, 2023-11-20:
**Awesome!** That complies for me and let's me generate the `_lin.sc` file with the linearized matrices using the desired workflow. I'll run some check cases to confirm the matrices match what I expect.   

This is the [test xml script ](https://github.com/vtpasquale/jsbsim/blob/dac763e2e54c9edaa090654be944113e8ab0639e/scripts/737_cruise_steady_turn_linearize.xml)

This is the latest https://github.com/JSBSim-Team/jsbsim/compare/master...vtpasquale:jsbsim:dev

**One thing:** JSBSim is locking up and not completing the simulation following the linearization. That's probably fine for the linearization workflow (mission accomplished), but I don't think that is the expected or desired behavior. Would you expect the typical simulation to continue after the linearized matrices are exported?

seanmcleod, 2023-11-20:
Yep, I wouldn't expect the linearization to crash or hang JSBSim.

vtpasquale, 2023-11-20:
This is happening because both `FGFDMExec.dT `and `FGFDMExec.saved_dT` are being set to 0.0 during the Linearization process. 

It's like `void FGFDMExec::SuspendIntegration(void) {saved_dT = dT; dT = 0.0;}`is called twice in a row.`FGLinearization` is interacting `FGStateSpace`, and FGStateSpace adjusts dT quite a bit during assembly - it's hard to pinpoint the exact culprit. I'll have to revisit later.

This workaround allows the simulation to proceed:
```
void FGFDMExec::DoLinearization(int)
{
  double dt0 = this->GetDeltaT();
  FGLinearization lin(this);
  lin.WriteScicoslab();
  this->Setdt(dt0);
}
```

vtpasquale, 2023-11-24:
Linearization from console executable and xml script feature is merged via #991. I appreciate the team's helpful and fast inputs on this. 

I thought enabling simplex trim at the same time as linearization might be easy, but it turned out to be more involved. The standard trim capability is working OK for me, so I'll close this issue.

closed 2023-11-24 by vtpasquale