JSBSim-Team/jsbsim #1010: JSBSim Simulink Function

kvgkrish, 2024-01-08:
Hi all,
This is Gokul doing a Master of Aerospace Engineering at TUM Germany. Currently working on master thesis at Chalmers University of Technology. 
I am working on a Bio-Inspired UAV dynamics system. I am new to JSBSIM and I have set up all the files and worked fine with available flights and scripts. If I use a new flight configuration I have simulated the model. It shows trim failure and sometimes flight scripts are not used.
Can you help me to resolve this issue?

[image]

[image]

seanmcleod, 2024-01-08:
Are you sure the aircraft should be trimmable at the initial conditions you're specifying for the trim? Or there could be an issue with the FDM file you've created.

If you share the FDM file and the conditions you're using for trim others could potentially take a look and help.

kvgkrish, 2024-01-08:
I have attached the flight data, scrip initial conditon for your reference.  

 aircraft is **StratoSurfer**, Scipts/Test_StratoSurfer_Cruise, and the Initial condition is **initCruise**.

https://github.com/UASLab/OpenFlightSim/blob/master/Simulation/aircraft/StratoSurfer/StratoSurfer.xml

kvgkrish, 2024-01-08:
I have attached the flight data, script, and initial conditon for your reference.

aircraft is StratoSurfer, Scipts/Test_StratoSurfer_Cruise, and the Initial condition is initCruise.

https://github.com/UASLab/OpenFlightSim/blob/master/Simulation/aircraft/StratoSurfer/StratoSurfer.xml

seanmcleod, 2024-01-08:
Just had a very quick glance. Is the engine definitely on and running? I see the throttle being advanced, but it's not 100% clear that the engine is running.

What is the output in terms of thrust and power and h-agl-ft below?

```xml
    <event name="Trim">
      <condition> simulation/frame gt 1 </condition>
      <set name="simulation/do_simple_trim" value="1"/>
      <notify>
        <property>velocities/vtrue-kts</property>
        <property>position/h-agl-ft</property>
        <property>aero/alpha-deg </property>
        <property>aero/beta-deg </property>
        <property>propulsion/engine[0]/thrust-lbs</property>
        <property>propulsion/engine[0]/power-hp</property>
        <property>attitude/phi-rad</property>
        <property>attitude/theta-rad</property>
        <property>attitude/psi-rad</property>
      </notify>
    </event>
```

kvgkrish, 2024-01-09:
***@***.***

***@***.***

Model has fails again even after throttle set to 100%.
The above picture is for motor power.
Thanks and regards,
Gokul.

kvgkrish, 2024-01-09:
[image]
[image]
[image]

I have attached the engine power and new script for your review.
Even after this modification, model has failed du to trim error.

kvgkrish, 2024-01-09:
Thanks for the response. How do you make sure the engine is on or not?
I faced this issue while editing the aircraft file and getting this error.

seanmcleod, 2024-01-09:
Take a look at this example of using the `<running>` element.

https://github.com/JSBSim-Team/jsbsim/blob/4b3ec67e2fffde5797b7fc168a686f2f08c4971d/aircraft/737/rudder_kick_init.xml#L11

The -1 means all engines that the aircraft has.

kvgkrish, 2024-01-09:
[image]
Now I am getting Very high thrust and again trim failure.

seanmcleod, 2024-01-09:
Well then there is something wrong with your engine, thruster model etc. With that sort of maximum thrust and a vehicle with a mass of only 3lbf, the trim routine probably can't find a small enough throttle command like 0.000001 to us to get a trim solution.

kvgkrish, 2024-01-09:
[image]
[image]

I have reduced the command, but it is still getting very high value.

seanmcleod, 2024-01-09:
No, no, no.... As I mentioned, the trim routine will try different throttle commands when trying to calculate a trim solution, so it doesn't matter what you set the throttle command to in the Motor Start event!

The point is it appears the engine power is way too high for this sort of vehicle. And the trim routine has probably some minimum throttle increment that it tries when trying to find a solution which is way too much in terms of thrust.

Think about it in terms of a rough sanity check, should a 3lbf vehicle have a motor of 350W, producing some enormous amount of thrust?

kvgkrish, 2024-01-09:
[image]
 It is a very small motor. I have attached the details of it.

seanmcleod, 2024-01-09:
Compare the amount of thrust you're seeing to the thrust measured for this engine for different propellers etc.

https://www.innov8tivedesigns.com/images/specs/Cobra_2217-8_Specs.htm

kvgkrish, 2024-01-09:
Yes, it is very high compared to the tabulated value.
How to resolve this issue.

seanmcleod, 2024-01-09:
Step through the JSBSim code where it calculates the thrust and see if you spot what input is making it so large.

kvgkrish, 2024-01-09:
But the 737 and c172x are working fine with the Simulink.

kvgkrish, 2024-01-09:
The aircraft file, script, and initial condition we use are created for the FlightGear module. Is there any problem with Simulink? Can we please tell me how to run JSBSim standalone and visualize the results in FG.

seanmcleod, 2024-01-09:
So run JSBSim stand-alone with your script, look at the command line arguments for JSBSim and see if there is any difference with the trim stand-alone versus with Simulink.

seanmcleod, 2024-01-10:
Did you look at the Quick Start section of the README? https://github.com/JSBSim-Team/jsbsim?tab=readme-ov-file#quick-start

kvgkrish, 2024-01-10:
[image]
 I am getting this error.

seanmcleod, 2024-01-10:
Take a look at - https://www.technewstoday.com/fix-command-prompt-access-denied/

seanmcleod, 2024-01-10:
Did you download JSBSim.exe from the web, or did you clone the repo and build it yourself? If downloaded from the web take a look at the Properties and check the Unblock checkbox if it's there, it is by default for any files downloaded from the internet.

[image]

kvgkrish, 2024-01-10:
[image]
I have downloaded from the web

kvgkrish, 2024-01-15:
I have corrected the model. Working on the controls, here I want to change the control based on the altitude to demonstrate the dynamic soring "S" patter.
Can you help me write control logic for dynamic soaring?

seanmcleod, 2024-01-29:
Do you see any error message when the input properties are processed?

https://github.com/JSBSim-Team/jsbsim/blob/6f0c4d805c499cf18280bcf8d7af358aee494876/matlab/JSBSim_SFunction.cpp#L459-L473

kvgkrish, 2024-01-29:
Hi,
I haven´t get any errors but I cant use that as a input.

seanmcleod, 2024-01-29:
So in what way can't you use it?

kvgkrish, 2024-01-30:
The figure below is for the desired altitude and headings function.
[image]

I want to feed the autopilot to follow the desired trajectory. I have created a function for the desired altitude and desired heading angle with respect to time. I want to feed as an input to the autopilot. If sending this comment through input, the model doesnt't follow the desired altitude. 
If I write this desired altitude and heading in the script file as a piecewise input, the model won't take it as an input.
The figure below is a script with a piecewise function. If I go with the below script, it crashes the model.
[image]

seanmcleod, 2024-01-30:
Rather than posting screenshots of your XML etc. rather put them in a triple back tick ` block with xml as the type. Means they're easier for other users to find via a search in future etc.

In terms of running this in Simulink, are you sure you have enabled the auto-pilot? Most auto-pilot implementations have a separate property that you have to set to enable them. So maybe you're updating the altitude and heading setpoints, but if the autopilot isn't enabled then the autopilot will ignore the setpoints.

For your script example, it looks to me like you haven't set the `Desired Flight Path` event to be executed more than once. Also the same comment applies to the script case, make sure the autopilot is enabled.

kvgkrish, 2024-01-30:
```XML
  <use aircraft="GBSv02" initialize="ini03"/>
	<run start="0.0" end="1000" dt="0.008333333333333333333">

		<event name="Wind aloft">
			<condition>simulation/sim-time-sec  ge  0.01</condition>
			<set name="atmosphere/psiw-rad" value="1.57" />

			<set name="atmosphere/wind-mag-fps">
				<description>Change_of_wind_magnitute_wrt_altitute</description>
				<function>
					<table>
						<independentVar lookup="row">position/h-agl-ft</independentVar>
						<tableData>
							5		16.93490625
							10		18.40375301
							15		19.32134464
							20		20
							25		20.54277915
							30		20.99717881
							35		21.38920099
							40		21.73469725
							45		22.04407591
							50		22.32455428
							55		22.58135087
							60		22.81836623
							65		23.03859546
							70		23.2443904
							75		23.43763291
							80		23.61985323
							85		23.79231315
							90		23.9560658
							95		24.11199966
							100		24.26087142
							1000.0  20.9000
							2000.0  19.8000
							2500.0  18.9000
							3000.0  20.0000
							3500.0  20.0000
						</tableData>
					</table>
				</function>
			</set>
			<!--set name="atmosphere/psiw-rad" value="0" /-->
			<!--set name="atmosphere/wind-mag-fps" value="0" /-->
			<!--set name="atmosphere/wind-down-fps" action="FG_RAMP" value="1" tc="10.0"/-->
		</event>

		<event name="Desired Flight Path">
			<!-->condition> position/h-agl-ft ge 600 </condition-->
			<condition> simulation/sim-time-sec ge 0.1 </condition>

			<set name="ap/altitude_hold" value="1.0" />
			<set name="ap/heading_hold" value="1.0" />
			</event>

		
  </run>

</runscript>

```
I have enabled the autopilot for altitude hold and heading hold.
```XML
<s_function_config>
    <input>
        <property> fcs/throttle-cmd-norm[0] </property>
        <property> fcs/aileron-cmd-norm </property>
        <property> fcs/elevator-cmd-norm </property>
        <property> fcs/rudder-cmd-norm </property>
		<property>ap/altitude_setpoint</property>
		<property>ap/heading_setpoint</property>
		
    </input> 

```
I have included the autopilot setpoint as an input for the S-Function. 
[image]
The model works well for altitude hold, but if I enable the heading, it oscillates around the x-axis.

seanmcleod, 2024-01-30:
> The model works well for altitude hold, but if I enable the heading, it oscillates around the x-axis.

Then you'll need to look at improving the heading controller.

kvgkrish, 2024-02-01:
In the initial stages of the simulation, a predefined crosswind, characterized by a specified magnitude and direction, is introduced to the model. The simulation accurately captures the model's initial response, manifesting in dynamic changes to its trajectory and behavior. However, as the simulation progresses, a noticeable reduction in the visual impact of the crosswind becomes evident. This observation indicates a diminishing influence over time.
To address this, I am looking to incorporate a windshear model into the simulation. I have already introduced values representing wind velocity with respect to altitude. However, the current model generates same responses for all directions. I am seeking guidance on how to effectively implement the windshear phenomenon into the atmospheric conditions of the simulation.

```XML
<?xml version="1.0"?>
<initialize name="reset03">

  <vc unit="KTS"> 38.87689112646 </vc>
  <longitude unit="DEG"> 0  </longitude>
  <latitude unit="DEG">   0  </latitude>
  <phi unit="DEG">         0.0  </phi> <!-- Roll -->
  <theta unit="DEG">       0.0  </theta> <!-- Pitch -->
  <psi unit="DEG">       0.0  </psi> <!-- Yaw -->
  <altitude unit="FT">     100  </altitude>
	<winddir unit="DEG">   90.0  </winddir> 
  <hwind unit="FT/SEC"> 20 </hwind>
	
</initialize>

```

``` XML
<event name="Wind aloft">
	<condition>simulation/sim-time-sec  ge  0.01</condition>
	<set name="atmosphere/psiw-rad" value="1.57" />

	<set name="atmosphere/wind-mag-fps">
		<description>Change_of_wind_magnitute_wrt_altitute</description>
		<function>
			<table>
				<independentVar lookup="row">position/h-agl-ft</independentVar>
				<tableData>
					5		16.93490625
					10		18.40375301
					15		19.32134464
					20		20
					25		20.54277915
					30		20.99717881
					35		21.38920099
					40		21.73469725
					45		22.04407591
					50		22.32455428
					55		22.58135087
					60		22.81836623
					65		23.03859546
					70		23.2443904
					75		23.43763291
					80		23.61985323
					85		23.79231315
					90		23.9560658
					95		24.11199966
					100		24.26087142
					1000.0  20.9000
					2000.0  19.8000
					2500.0  18.9000
					3000.0  20.0000
					3500.0  20.0000
				</tableData>
			</table>
		</function>
	</set>
	<!--set name="atmosphere/psiw-rad" value="0" /-->
	<!--set name="atmosphere/wind-mag-fps" value="0" /-->
	<!--set name="atmosphere/wind-down-fps" action="FG_RAMP" value="1" tc="10.0"/-->
</event>
```

agodemar, 2024-02-19:
@kvgkrish did you progress in your study on this topic?

kvgkrish, 2024-02-19:
@agodemar yes, still working on this

bcoconni, 2024-09-08:
@kvgkrish Any progress on this topic ? Or should we consider this issue closed ?