JSBSim-Team/jsbsim #1017: Can autopilot feature come to jsbsim- unreal engine 5?

buseeryas, 2024-01-15:
Can autopilot feature come to jsbsim- unreal engine 5?

seanmcleod, 2024-01-17:
I don't see why it isn't available at the moment. Simply use an aircraft that includes an autopilot system, and when the autopilot is invoked ensure that your code in your Unreal app ignores pilot control inputs.

gallonmate, 2024-01-19:
Autopilot works in UE5. See this chat https://github.com/JSBSim-Team/jsbsim/discussions/973
And as Sean mentioned you need to use an aircraft model that has an autopilot system included.
The following aircraft have autopilot systems in their code:

c172x
c310
global5000

If you mean to ask if a general autopilot can come to jsbsim so that the autopilot can automatically work for any aircraft, then maybe that would be a "Feature Request" and not an "Issue". Or it could be posted in the discussion section.

closed 2024-01-19 by buseeryas