JSBSim-Team/jsbsim #1034: Wrong calculation of vPQRdot in FGAccelerations class

mathieuflament, 2024-02-05:
***

**I'm submitting a ...**
  - [x] bug report
  - [ ] feature request
  - [ ] support request

**Describe the issue**

As far as we understand, the vector `vPQRdot` in `FGAccelerations` class is the derivative of the angular velocity of the body w.r.t the ECEF frame, expressed in body frame coordinates, and the vector `vPQRidot` is the derivative of the angular velocity of the body w.r.t the ECI frame, also expressed in body frame coordinates.
In method `FGAccelerations::CalculatePQRdot`, the variable `vPQRdot` is derived from `vPQRidot` using the expression :
https://github.com/JSBSim-Team/jsbsim/blob/01765533bd3fae492f62ecddb66c79988e0d0196/src/models/FGAccelerations.cpp#L162

It should be `vPQRdot = vPQRidot + in.vPQRi * (in.Ti2b * in.vOmegaPlanet)` instead.

Proof :

[image]

The correct expression is the last equation above, expressed in body frame coordinates.

**What is the current behavior?**
To highlight the problem, we have slightly modified the `orbit` check-case, with an initial angular rate w.r.t the ECI frame of `vPQRi=[15deg/s, 0., 0.]`. The body is a spinning ball around its X-axis (pointing to the East), orbiting the Earth in the equatorial plane. In this scenario, the body X-axis and the angular velocity vector are constant in the ECI frame, i.e. `vPQRidot` is null. The angular velocity vector w.r.t ECEF frame, namely `vPQR`, is then simply equal to the Earth sidereal rate vector projected on the (rotating) Y-Z plane of the body frame.  

The following graph shows the Q and R components of the `vPQR` vector, and `vPQRdot` vector, straight from the `BallOut.csv` output file produced by the JSBSim program :
[image]

The sign error on `vPQRdot` is obvious.

Fortunately, as the equation of motion are integrated in the ECI frame using `vPQRidot` in `FGPropagate` class, this error has a limited effect on trajectory computations:
- the calculation of `vPQRidot` is OK,
- the calculation of `vPQRi` by integrating `vPQRidot` is OK,
- the calculation of `vPQR` from `vPQRi` is OK,
- the attitude computations are OK.

Only the outputs and further computations based on `vPQRdot` are affected by a (small) error.

**What is the expected behavior?**

Once the sign error on the term `in.vPQRi * (in.Ti2b * in.vOmegaPlanet)` is corrected in `FGAccelerations::CalculatePQRdot` (l.158 and l.162) and `FGAccelerations::SetHoldDown` (l.214), consistent results are restored, as shown by the following figure:
[image]

All the other output variables in the `BallOut.csv` file have remained exactly the same, confirming that a change in `vPQRdot` values has no adverse side-effect on the simulation loop. 

**Please tell us about your environment:**
 - OS : Linux Debian 12
 - JSBSim 1.2.0
 - Language C++

**Other information**
We would like to thank the JSBSim developer team for their invaluable contribution to the flight dynamics simulation community.

jonsberndt, 2024-02-06:
Hi, Mathieu,

Thanks for this detailed report. JSBSim is where it is today in part due to continuous feedback from the larger community. 

This issue got my attention - it's a very interesting issue. I was initially surprised to read of it partially due to the fact that JSBSim has been involved in comparisons against a number of other high-fidelity simulations and fared very well. But, as you pointed out, "this error has a limited effect on trajectory computations" so that could be why it has gone unnoticed if it is indeed an error. 

I'll take a closer look at this myself (this one is going to need some quiet time and a few cups of coffee ;-), and I'm sure that Bertrand and Sean and others may also take a close look. I just wanted to post a note of appreciation for the thoughtful report.

bcoconni, 2024-02-06:
Indeed, thank you for the excellent bug report @mathieuflament 👍 Clear and thorough.

I agree with your analysis, and while investigating the code history, I discovered that the error was introduced almost 13 years ago by myself via commit https://github.com/JSBSim-Team/jsbsim/commit/fdf7cf7e7ffe99c3a15ca5d48f7f3f6b91b7445f. However as you mentioned the side effects are very limited as `vPQRdot` is used in just a few places, namely:
https://github.com/JSBSim-Team/jsbsim/blob/d2d8f618b89db5bc216d035f945913032104a57c/src/input_output/FGOutputTextFile.cpp#L280
which writes the value of `vPQRdot` to a CSV file.

https://github.com/JSBSim-Team/jsbsim/blob/d2d8f618b89db5bc216d035f945913032104a57c/src/initialization/FGTrim.cpp#L436
where `vPQRdot` is used to determine the position at which the aircraft will sit on the ground at the simulation start.

https://github.com/JSBSim-Team/jsbsim/blob/d2d8f618b89db5bc216d035f945913032104a57c/src/models/FGAccelerations.cpp#L271-L273
where `vPQRdot` is used to compute the friction forces between the landing gears and the ground. Most likely the sign error has a small effect on this computation as the norm of `in.vOmegaPlanet` is very small (≈7.3e-5 rad/s).

This is most likely why the error remained unnoticed for such a long time. Anyway the bug must be fixed so you could perhaps submit a pull request (referencing this issue) to fix it ?

Thanks.

bcoconni, 2024-02-10:
The PR #1036 that fixes the bug reported in this issue has been merged in `master`. I guess that this issue can now be closed ?

bcoconni, 2024-03-02:
Closing this issue as the bug has been fixed in PR #1036.

closed 2024-03-02 by bcoconni