JSBSim-Team/jsbsim #1039: Request: turbine spindown table or property support

Octal450, 2024-02-12:
**I'm submitting a ...**
  - [ ] bug report
  - [x] feature request
  - [ ] support request => **Please do not submit support request here**, see note at the top of this template.

**Current behavior**

In the current implementation, the value of n1spindown or n2spindown can only be set in the XML. When realistic values are used.

**Requested behavior?**

Ability to edit the spindown value at runtime by using either fgfunction or property.

**What is the motivation / use case for changing the behavior?**
I have a panel state system which quickly sets the aircraft into various states. For startup, set-running + modified spool tables work great, but shutdown takes way too long when realistic values are set into spindown. These changes would allow modification of the spindown value at runtime.

**Please tell us about your environment:**

 - Windows 11 x64
 - FlightGear 2020.4 (next)

**Other information**

gallonmate, 2024-02-19:
Just a quick note until someone addresses the request: In the meanwhile you could set n1 and n2 directly. I'm not too familiar with scripts, so I'm only assuming an advanced xml script could be written if you still wanted to manually force a faster spinup/down. I do know setting any value from 0 to 100 for n1 and n2 works instantly. And it doesn't break the spinup/down, it just continues from the new values.

Octal450, 2024-02-19:
I didn’t realize that worked. I’ll give it a try. Thanks!

Kind Regards,
Josh