JSBSim-Team/jsbsim #1042: Aircraft crashed on air when rolling consecutively

nmfarras, 2024-02-16:
**I'm submitting a ...**
  - [x] bug report
  - [ ] feature request
  - [ ] support request => **Please do not submit support request here**, see note at the top of this template.

I've tried to replicate the [A DIY Flight Simulator tutorial](https://dev.epicgames.com/community/learning/tutorials/mmL/unreal-engine-a-diy-flight-simulator-tutorial) and now recreate it for fighter F16 using available F16 jsbsim model. I happen to get the simulation when still on the air.

When i fly the aircraft and roll it multiple time (like 2 or more in a row) there is a chance the simulation stop and the log message says "Aircraft Crashed" as shown below. 

[image]

Is there anyone who know why this happen even though i'am not crashing it to the ground. Is it how the jsbsim handle things or bug on jsbsim plug in on UE? I appreciate if anyone know how to handle it or what should be done to avoid this. Thanks!

**What is the motivation / use case for changing the behavior?**

I create it on:
 - Window 11
 - JSBSim v1.2.0 (Plugin version 1.01)
 - Unreal Engine 5.2.1

seanmcleod, 2024-02-16:
So the crash message you're seeing is coming from this piece of code:

https://github.com/JSBSim-Team/jsbsim/blob/92daf8953c32a0a602ab0f48240a07c4a04aadb9/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/JSBSimFlightDynamicsModel/Private/JSBSimMovementComponent.cpp#L261-L265

Which is triggered if the AGL altitude is less than -10.

https://github.com/JSBSim-Team/jsbsim/blob/92daf8953c32a0a602ab0f48240a07c4a04aadb9/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/JSBSimFlightDynamicsModel/Private/JSBSimMovementComponent.cpp#L693-L699

Now in order to calculate the AGL altitude JSBSim needs information on the terrain elevation. So my guess is that in some areas while rolling a very large terrain elevation is being reported.

Looking at your screenshot I see a message - "YOUR SCENE CONTAINS A SKYDOME MESH.... BUT IT DOES NOT COVER THAT PART..."

Is this mesh issue possibly causing an issue with the terrain elevation lookup?

gallonmate, 2024-02-16:
You're likely detecting parts of the aircraft model or the player. etc, while rolling, instead of the ground. The UE5 plugin does a line trace to find the ground mesh. When you roll your plane, something is likely rolling into the line trace and triggering it. See this comment https://github.com/JSBSim-Team/jsbsim/discussions/862#discussioncomment-5512289 and the discussion in there for your options to correct it. It's up to you how to you want to setup your collision channels for your models/game.

nmfarras, 2024-02-19:
> You're likely detecting parts of the aircraft model or the player. etc, while rolling, instead of the ground. The UE5 plugin does a line trace to find the ground mesh. When you roll your plane, something is likely rolling into the line trace and triggering it. See this comment [#862 (comment)](https://github.com/JSBSim-Team/jsbsim/discussions/862#discussioncomment-5512289) and the discussion in there for your options to correct it. It's up to you how to you want to setup your collision channels for your models/game.

All right I just come back online and will check it out.

> So the crash message you're seeing is coming from this piece of code: [...]

"YOUR SCENE CONTAINS A SKYDOME MESH.... BUT IT DOES NOT COVER THAT PART..." thing is related to environment pawn not finish rendering i guess. Cause when i fly normally and the message come out and the sky become black like that, it will will become normal after few second. But the reference to AGL and such is useful. Thank you!

nmfarras, 2024-02-19:
> You're likely detecting parts of the aircraft model or the player. etc, while rolling, instead of the ground. The UE5 plugin does a line trace to find the ground mesh. When you roll your plane, [...]

 @gallonmate I have a question. I check the issues and i am curious for line trace from which point it emit? does it use one of parameter set from xml file as the source? or it come from this "orange color" boundary that come after activating jsbsim plugin?

[image]

If i want to disabled collision from player like suggest from [this](https://github.com/JSBSim-Team/jsbsim/discussions/862#discussioncomment-5513521)  and [this](https://github.com/JSBSim-Team/jsbsim/discussions/862#discussioncomment-5514251) Where should i go? Cause i think i already set ignore collision in my blueprint and skeletal mesh as follow

[image]

[image]

gallonmate, 2024-02-19:
The code all starts here: https://github.com/JSBSim-Team/jsbsim/blob/10ac2036d8b5ac6d840a105f41062c603e952e6a/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/JSBSimFlightDynamicsModel/Private/JSBSimMovementComponent.cpp#L182

The line trace itself uses the aircraft transform + 200 units up and the end point always always straight down. I guess I should have clarified, the line trace code is already setup to ignore anything attached to the aircraft blueprint. So it might take some more work to find out what the line trace is actually hitting. If you are not spawning any players/characters and it's just the aircraft blueprint, then maybe it's colliding with some part of the cesium system? I honestly have no idea.

If you want to fiqure it out, you have two options:  You can edit or debug the plugin in c++ to fiqure out what you're hitting. For example you can roll the aircraft until you crash then check the call stack in VS and see what the line trace's Hitresult shows for the object hit. 
Or you can add some code to print the Hitresult.
Or you can delete parts of your level and anything you have spawned until the aircraft stops crashing when you roll.

nmfarras, 2024-02-20:
> ...The line trace itself uses the aircraft transform + 200 units up and the end point always straight down. I guess I should have clarified, the line trace code is already setup to ignore anything attached to the aircraft blueprint. So it might take some more work to find out what the line trace is actually hitting. If you are not spawning any players/characters and it's just the aircraft blueprint, then maybe it's colliding with some part of the cesium system that it's not supposed to.

Thanks for the reply! I will check if deactivating Cesium might stop the issue.

Also I got new info. I check it with my college and he do it in UE5.0 but got no such a error. But i notice something interesting that in both version, sometime the fighter behave like "hitting" something when rolling, and cause the roll seems unnatural. Does line tracing cause this kind of effect as well (like hitting a wall or something) or just simply checking the distance between aircraft and nearest mesh below aircraft? 

Also as in the one on UE5.0 no airplane crashed occur, does UE version might also has some factor in it as well?

nmfarras, 2024-02-20:
[image]

[image]

Quick update. I already deactivate Cesium, but i guess it still happen. I also try to remove volumetric cloud in case it will affect too. But still the same i guess...

gallonmate, 2024-02-20:
It looks like you have extra blueprints attached to airplane. Try flying with only the aircraft blueprint. As for the question about about hitting/bouncing off things, the landing gear and body can have contact points defined in the xml file and then UE5 uses line traces to send collision information at those locations to JSBSim. So you can bump/hit objects with different parts of the plane and not "crash" but instead bounce or lose stability.

[image]
I suggest removing those attached objects to the plane and seeing if that helps. 

Here you can see the results of contacting an object. In the last example moving the block under the aircraft's location changes the "ground level" and moving it high enough eventually "crashes" the aircraft.
[image]

nmfarras, 2024-02-20:
> It looks like you have extra blueprints attached to airplane. Try flying with only the aircraft blueprint. As for the question about about hitting/bouncing off things, the landing gear and body can [...]

All right i will try flying with nothing attach first. Regarding the use of UE5.0 not having crash issue, just now i try to rebuilt from scratch while leaving the pawn having extra object (it is armament attach on Blueprint pawn), and it work though the weird hitting or maybe tapping something when rolling still exist. I will update you again if i found something else.

nmfarras, 2024-02-22:
Sorry for late update. After trying, the error still persist. Just in case i try [this one ](https://github.com/JSBSim-Team/jsbsim/discussions/862#discussioncomment-5513394) and the the behavior like tapping something once when rolling and crashed on the air no longer happened. So it is indeed line tracing hitting something. 

I actually want to follow second option suggested [here](https://github.com/JSBSim-Team/jsbsim/discussions/862#discussioncomment-5513521). As i still new, can you suggest me some reference to add pawn or actor into ignore list as suggest? So far i only working with the graph side (node thingy) of unreal, therefore I'am not familiar how to check in which file my actor registered, or do they have object class relate to it that can be registered on the array?

gallonmate, 2024-02-22:
If you don't know what you're colliding with, then an ignore list will not really help. And it also requires editing the plugin in c++. The person was kind enough to share the c++ code changes here if you believe you want to try adding the change yourself: https://github.com/JSBSim-Team/jsbsim/pull/893/commits/a7e4b899aa1218c394770aa65f220c78a0ef5538

But if you're able to change code in c++, then you should be able to debug the code and check the name of the object in HitResult. Otherwise then you're going to have to still double check everything again. Check what is different between your UE5.0 and UE5.2 projects. Or start with the sample project that comes with the plugin and slowly add your changes until the error happens again.

nmfarras, 2024-02-22:
Allright maybe i will take my time for this one. For checking the HitResult does i just need to uncomment DrawDebugLine()?

>Uncommenting the debug line seen below, in the GetALevel function and also checking the name of the component in the HitResult would confirm if the aircraft is self-colliding or colliding with an unintended object.

>https://github.com/JSBSim-Team/jsbsim/blob/10ac2036d8b5ac6d840a105f41062c603e952e6a/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/JSBSimFlightDynamicsModel/Private/JSBSimMovementComponent.cpp#L218-L222

_Originally posted by @gallonmate in https://github.com/JSBSim-Team/jsbsim/discussions/862#discussioncomment-5512289_

gallonmate, 2024-02-22:
That's just for visually seeing the line trace, it won't tell you what you're hitting.
You can already view the HitResult without changing code.  

How to debug and view the HitResult:
-Open UEReferenceApp.sln with Visual Studio.
-Press the Green button labeled "Local Windows Debugger" which will open UE5 with debugging ability.
-After UE5 is loaded, start flying your aircraft as normal and get close to the moment that causes crashed/error but do not cause the crash/error yet.
-Pause game and switch back to Visual Studio.
-Open file JSBSimMovementComponent.cpp and go down to line 249. 
-Right click in the area left of the 249 line and Insert Conditional Breakpoint.
-Where the new box says `Example: x==5`, write in there `HAT < -3`, press keyboard enter or click close button. This condition means when distance is less than -3 meters (-10 feet) it will automatically pause the program so you can inspect the data values. Here's a picture for reference. 
[image]

-Now you can switch back to UE5 and press play to continue flying.
-Try rolling aircraft or however you normally can cause error/crash.
-Visual studio will then pause or "break" the program when the aircraft is less than -3 meters (-10 feet).
-When this happens you can now find the HitResult in the Locals tab. As seen in this picture:
[image]

nmfarras, 2024-02-23:
> That's just for visually seeing the line trace, it won't tell you what you're hitting. You can already view the HitResult without changing code.
> 
> How to debug and view the HitResult: -Open UEReferenceApp.sln with Visual Studio. -Press the Green button labeled "Local Windows Debugger" which will open UE5 with debugging ability. -After UE5 is loaded, start flying your aircraft as normal and get close to the moment that causes crashed/error but do not cause the crash/error yet. -Pause game and switch back to Visual Studio. -Open file JSBSimMovementComponent.cpp and go down to line 249. -Right click in the area left of the 249 line and Insert Conditional Breakpoint. -Where the new box says `Example: x==5`, write in there `HAT < -3`, press keyboard enter or click close button. This condition means when distance is less than -3 meters (-10 feet) it will automatically pause the program so you can inspect the data values. Here's a picture for reference. ![image](https://private-user-images.githubusercontent.com/28702753/307115808-ab7d89f9-c732-4e74-bc2e-6923426ecdc2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg2NTk0ODMsIm5iZiI6MTcwODY1OTE4MywicGF0aCI6Ii8yODcwMjc1My8zMDcxMTU4MDgtYWI3ZDg5ZjktYzczMi00ZTc0LWJjMmUtNjkyMzQyNmVjZGMyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjIzVDAzMzMwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTllMjI1ZTRiMDExNzJmOGE3MmM1NGQ4MjIxMjJmNDJmOWQ1MDk2NDY2MjE1ZmFiNmVlYTE0MjMwMjIyZDg5MDcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.5PsQn3Ev-UJrnr5O74uwFMYHeUhS7YtDaX_KBjCc8n0)
> 
> -Now you can switch back to UE5 and press play to continue flying. -Try rolling aircraft or however you normally can cause error/crash. -Visual studio will then pause or "break" the program when the aircraft is less than -3 meters (-10 feet). -When this happens you can now find the HitResult in the Locals tab. As seen in this picture: ![image](https://private-user-images.githubusercontent.com/28702753/307120940-81420b20-d2fc-478a-87a2-d985a01763e5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg2NTk0ODMsIm5iZiI6MTcwODY1OTE4MywicGF0aCI6Ii8yODcwMjc1My8zMDcxMjA5NDAtODE0MjBiMjAtZDJmYy00NzhhLTg3YTItZDk4NWEwMTc2M2U1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjIzVDAzMzMwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTlkNmI4ZjBlNzViNGM3N2IyOTE0OWExYTBiMjdjNWQ3MDc2ZWVkOTM0ZTE4ODVhNjEyZGM0MzNlYmQ4NTJmNWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.jEpubCiKznIMr8XHHRPm2zpwSjHgOPBMyRk11Yo1KnU)
> 
> (You can also hover the mouse pointer over the variables in the code file and it will show the same current variable data if available, in a popup window you can scroll through.) ![image](https://private-user-images.githubusercontent.com/28702753/307127256-866b38f1-2469-414d-910f-a8a43cd4816d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg2NTk0ODMsIm5iZiI6MTcwODY1OTE4MywicGF0aCI6Ii8yODcwMjc1My8zMDcxMjcyNTYtODY2YjM4ZjEtMjQ2OS00MTRkLTkxMGYtYThhNDNjZDQ4MTZkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjIzVDAzMzMwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQxM2UyMTc5NDM4NjllMWVjMDY4YTBjOGYyOWJlMzBiODg3ZTNiZWVkZjAwZTcyZmEwMjU4MjY2MGM3NTNjMmEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.to70bTfby40KHxOEQtkdc8tIrSWSsQIAe4oIxtDUy_I)

Thank  you for the guide of debugging on visual studio. It is very helpful!

nmfarras, 2024-02-26:
Considering the solution, i guess it is pretty much answer what should i do to solve my problem. I guess I will close the issue for now and maybe will open it again if the I find something else aside debugging with hit result and adding the object in ignore list. Thank you for the support!

closed 2024-02-26 by nmfarras