JSBSim-Team/jsbsim #1072: JSBSimFlightDynamicsModel could not be loaded in Unreal Engine

Neurowerk, 2024-04-24:
Hi, 
I built JSBSim plugin from source code for Unreal Engine 5.2.1 in Debian 12, but when i enable JSBSim plugin to my Unreal project. i see next error:

[2024.04.24-03.45.27:568][  0]LogCore: Warning: dlopen failed: libJSBSim.so.1: cannot open shared object file: No such file or directory
[2024.04.24-03.45.27:568][  0]LogPluginManager: Error: Plugin 'JSBSimFlightDynamicsModel' failed to load because module 'JSBSimFlightDynamicsModel' could not be loaded.  There may be an operating system error or the module may not be properly set up.

How is possible to fix this error?

gallonmate, 2024-04-24:
What steps did you follow in order to build? Did you check if the  libJSBSim.so file exists in the correct location of your project plugin's folder? Did you read this issue https://github.com/JSBSim-Team/jsbsim/issues/963 and see if the solution solves your error?

The creator of the linux build version is listed in that issue #963, you can try tagging/asking them question if you still encounter difficulty.

Neurowerk, 2024-04-24:
I ran JSBSimForUnrealLinux.sh, additional in my case i copied from src/ to  UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/ThirdParty/JSBSim/Include all headers, than i checked libJSBSim.so at UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/ThirdParty/JSBSim/Lib/Linux/, further i generated Unreal Project Files and run debug in VS code, after that i started debug in VS code, then i ran Makefile in jsbsim/UnrealEngine. 
In builded plugin, again I also checked libJSBSim.so at UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/ThirdParty/JSBSim/Lib/Linux/.
I tried to run jsbsim/UnrealEngine/UEReferenceApp.uproject, i created custom project and copied there Plugins folder - the result is same

Neurowerk, 2024-04-24:
Solved.

closed 2024-04-24 by Neurowerk

gallonmate, 2024-04-25:
What was the solution?