JSBSim-Team/jsbsim #1110: Transfer function make JSBSim close down

NikolaiVChr, 2024-07-01:
**I'm submitting a ...**
  - [x] bug report
  - [ ] feature request
  - [ ] support request => **Please do not submit support request here**, see note at the top of this template.

**Describe the issue**
It seems it does not like this component:
```xml
    <lead_lag_filter name="fcs/pitch/integrator">
            <input> fcs/pitch/integrator-input </input>
            <c1> 1 </c1>
            <c2> 5 </c2>
            <c3> 1 </c3>
            <c4> 0 </c4>
    </lead_lag_filter>
```

**What is the current behavior?**
`5.55 [INFO]:flight    C:\Jenkins\workspace\Windows-release\flightgear\src\FDM\JSBSim\JSBSim.cxx:249:   aero does not exist (you may have mis-typed the name).`

**What is the expected behavior?**
I expect JSBSim to not close down.

**What is the motivation / use case for changing the behavior?**
I have documents outlining the use of this component in the control scheme.

**Please tell us about your environment:**
 - OS=Win 10
 - JSBSim version=FG 2020.3.19

**Other information**

seanmcleod, 2024-07-01:
> `aero does not exist (you may have mis-typed the name)`

Hmm, are you sure this error is related to the `lead_lag_filter` you listed? The error specifically mentions `aero` which doesn't appear in your snippet above, sounds like an undefined property you're possibly trying to use somewhere else in your FDM?

NikolaiVChr, 2024-07-01:
Yes, sorry. I looked at the wrong log file.

closed 2024-07-01 by NikolaiVChr