JSBSim-Team/jsbsim #1116: Request: Unreal Marketplace Plugin

jamesford42, 2024-07-03:
For people just wishing to try things out, the current process leaves a lot to be desired. Please consider actually publishing the unreal engine plugin to the marketplace.

gallonmate, 2024-07-03:
A sample project is included in the UnrealEngine folder. The plugin is fairly barebones and was originally meant as a reference point for the simulation industry to use as a starting point. It should be noted that anyone wishing to truly develop with the plugin will need to be ready to learn a lot or already be an experienced developer.

Since no one owns the reference plugin and it is MIT licensed, you are free to package the plugin itself as compiled binary and submit to the Unreal Marketplace. But since the plugin relies on JSBSim, which is LGPL and GPL licensed, you have to be careful which parts of JSBSim you include on the marketplace, as GPL content is definitely not allowed on the marketplace. LPGL, might be allowed. That's something you would have to find out.

Any changes or improvements are welcome to the plugin/project. If there are parts of the read me that can use improvement or clarity, please share. You can also submit pull requests with changes you'd like to see.

jamesford42, 2024-07-03:
Considering this repository is public, doesn't that fulfill the terms of LGPL/GPL (i actually do not know), regardless of how and where binaries built from that code are packaged and distributed? Not sure if this is even the same issue exactly, but if a commercial game cannot be made which uses JSBSim code for the aircraft physics sim, that would be a big nail in the coffin of my planned use.

seanmcleod, 2024-07-03:
Take a look at - https://github.com/JSBSim-Team/jsbsim/discussions/631

jamesford42, 2024-07-03:
> Take a look at - #631

I see, looks like the most current state is described here.
https://github.com/JSBSim-Team/jsbsim/discussions/631#discussioncomment-2908489

Question posted as reply to that.

demonixis, 2024-07-10:
A marketplace plugin is too complex because we need to support 3 versions of Unreal. JSBSim works on all UE5 versions, it's not the problem. The problem is that it takes times to compile each time an engine version and deploy it to the store.