JSBSim-Team/jsbsim #1117: JSBSim can't interpolate between n dimensions

TrentConley, 2024-07-03:
I've read an implemented Jon's advice [here](https://sourceforge.net/p/jsbsim/mailman/message/31037613/), but it looks like the nester interpolation is not happening, since the `GetValue()` function called on the parameters provides a double value rather than calling interpolate again. See `src/math/FGFunction.cpp` at line 691. 

For the RC craft I am working on, I'll need at least 6 dimensions to interpolate from to produce 6 distinct aerodynamic coefficients. 

Below are two examples of how I was setting up the nested interpolation. Neither worked since they each only interpolated once:

``` <axis name="DRAG">
      <function name="aero/coefficient/CD">
        <description>CD Coefficient</description>
        <interpolate1d>
          <p>aero/beta-rad</p>
          <v>-0.3476658503988659</v>
          <interpolate1d>
            <p>aero/alpha-rad</p>
            <v>-0.063651317007977318</v>
            <interpolate1d>
              <p>fcs/rudder-pos-rad</p>
              <v>-0.4690658503988659</v>
              <interpolate1d>
                <p>fcs/left-aileron-pos-rad</p>
                <v>-0.3490658503988659</v>
                <interpolate1d>
                  <p>fcs/right-aileron-pos-rad</p>
                  <v>-0.3490658503988659</v>
                  <v>-0.02731959022</v>
                  <v>0.3490658503988659</v>
                  <v>-0.025771052</v>
                </interpolate1d>
                <v>-0.17453292519943295</v>
                <interpolate1d>
                  <p>fcs/right-aileron-pos-rad</p>
                  <v>-0.17453292519943295</v>
                  <v>-0.01767199402</v>
                  <v>0.17453292519943295</v>
                  <v>-0.01580392049</v>
                </interpolate1d>
                <v>0</v>
                <interpolate1d>
                  <p>fcs/right-aileron-pos-rad</p>
                  <v>0</v>
                  <v>-0.01419396561</v>
                  <v>-0</v>
                  <v>-0.01419396561</v>
                </interpolate1d>
                <v>0.17453292519943295</v>
                <interpolate1d>
                  <p>fcs/right-aileron-pos-rad</p>
                  <v>0.17453292519943295</v>
                  <v>-0.01857091949</v>
                  <v>-0.17453292519943295</v>
                  <v>-0.02048905171</v>
                </interpolate1d>
                <v>0.3490658503988659</v>
                <interpolate1d>
                  <p>fcs/right-aileron-pos-rad</p>
                  <v>0.3490658503988659</v>
                  <v>-0.03070267218</v>
                  <v>-0.3490658503988659</v>
                  <v>-0.03315893703</v>
                </interpolate1d>
              </interpolate1d>```

```
      <function name="aero/coefficient/CLalpha">
        <description>Lift_due_to_alpha</description>
        <interpolate1d>
          <property>aero/beta-rad</property>
          <value>0</value>
          <interpolate1d>
            <property>aero/alpha-rad</property>
[... 160 lines cut ...]
                 0.1745 -0.2124093169 -0.1853586857 -0.1301135914
                 0.3491 -0.2124093169 -0.1853586857 -0.1301135914
              </tableData>
            </table>
            <value>0.20943951023931956</value>
            <table name="aero/table/CL_aos20_aoa12">
              <independentVar lookup="row">fcs/elevator-pos-rad</independentVar>
              <independentVar lookup="column">fcs/rudder-pos-rad</independentVar>
              <tableData>
                         0.0000000000  0.1745329252  0.3490658504
                -0.3491 -0.4407664491 -0.3964551189 -0.3489947922
                -0.1745 -0.4407664491 -0.3964551189 -0.3489947922
                 0.0000 -0.4407664491 -0.3964551189 -0.3489947922
                 0.1745 -0.4407664491 -0.3964551189 -0.3489947922
                 0.3491 -0.4407664491 -0.3964551189 -0.3489947922
              </tableData>
            </table>
          </interpolate1d>
        </interpolate1d>
      </function>
```

To resolve this, I am planning on creating a nxm matrix where n is the number of dimensions I am need to interpolate from and m is the number of data points I have. Technically, I could only store n-1 dimensions in the matrix since I am planning on having the last row be the coefficient value. Then, I'll created a new function interpolateNd that can interpolate between the vectors in the matrix. 

Any thoughts on a better implementation?

Octal450, 2024-09-29:
Maybe I am missing something, but why not use the "table" lookup and breakPoint="" to create a 3D table rather than using interpolate1d?

Kind Regards,
Josh

TrentConley, 2024-09-29:
It's fine to create a 3D table. But let's say you have aerodynamic data across 7 control surfaces, now you need at least 7 dimensional data. Currently, JSBSim can't handle that to my understanding. I made a modification to a fork of JSBSim, and I could bring that into main as well. Check out this PR that introduced it:
https://github.com/machindustries/jsbsim/pull/2

Octal450, 2024-09-29:
I see,
interesting.

Kind Regards,
Josh

Octal450, 2024-09-29:
Took a quick look.

I REALLY like this. Would be awesome to see this move into JSB/FlightGear. I could definitely make use of this in my flight models.

Kind Regards,
Josh

TrentConley, 2024-09-29:
Cool! I'll work on integrating it into JSBSim then.

bcoconni, 2026-05-10:
The feature has just been incorporated in the `master` branch thanks to @agodemar's PR #1443.
This issue is now closed.