JSBSim-Team/jsbsim #1118: Robocopy step errors if spaces in path.

jamesford42, 2024-07-03:
This post build copy step (or however it is hooked in) does not properly quote its paths. It appears to break when there is a space in the folder structure.
```
>C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\Microsoft\VC\v160\Microsoft.CppBuild.targets(499,5): warning MSB8004: Output Directory does not end with a trailing slash.  This build instance will add the slash as it is required to allow proper evaluation of the Output Directory.
1>JSBSimForUnreal.vcxproj -> D:\Projects\Unreal Projects\JSBSim\UnrealEngine\Plugins\JSBSimFlightDynamicsModel\Source\ThirdParty\JSBSim\LibDebug\JSBSim.dll
1>A subdirectory or file UnrealEngine\Plugins\JSBSimFlightDynamicsModel\Source\ThirdParty\JSBSim\Include already exists.
1>
1>-------------------------------------------------------------------------------
1>   ROBOCOPY     ::     Robust File Copy for Windows
1>-------------------------------------------------------------------------------
1>
1>  Started : Wednesday, July 3, 2024 12:52:40 PM
1>   Source - D:\Projects\Unreal\
1>     Dest - D:\Projects\Unreal Projects\JSBSim\Projects\JSBSim\src\
1>
1>    Files :
1>  Options : /DCOPY:DA /COPY:DAT /R:1000000 /W:30
1>
1>------------------------------------------------------------------------------
1>
1>EXEC : error : Invalid Parameter #3 : "UnrealEngine\Plugins\JSBSimFlightDynamicsModel\Source\ThirdParty\JSBSim\Include"
1>Missing operator.
1>
1>       Simple Usage :: ROBOCOPY source destination /MIR
1>
1>             source :: Source Directory (drive:\path or \\server\share\path).
1>        destination :: Destination Dir  (drive:\path or \\server\share\path).
1>               /MIR :: Mirror a complete directory tree.
1>
1>    For more usage information run ROBOCOPY /?
```

jamesford42, 2024-07-03:
fixes #1118

bcoconni, 2024-08-16:
As the PR #1119 has been merged, I suggest that to close this issue.

closed 2024-08-19 by jamesford42