JSBSim-Team/jsbsim #1120: Keyboard/Gamepad input non-functional in Unreal Plugin Project

jamesford42, 2024-07-04:
I have gotten to the plugin project loaded in editor and playing (simulating) but asside from the D and P keys nothing seems to actually do anything. I kind of assume im doing something wrong here, but i don't see what. Actually im not sure where the bindings to keys is actually happening, i see the input action events (which dont fire) though.

demonixis, 2024-07-10:
Hey,
I think we've to update the plugin to the latest version of Unreal with EnhancedInput instead of the old InputSystem.
Input works correctly if you remap keys, but best is to use InputAction now it's stable and enabled by default.

AirshipSim, 2025-02-28:
@jamesford42 you do not say which kind of hardware you are trying to use. You may have not noticed the key binding detailed in the README section of the plugin.

As far as code is concerned, please see:
https://github.com/JSBSim-Team/jsbsim/pull/1225

@gallonmate tagging you for follow-up.

gallonmate, 2025-02-28:
Considering they never replied in the last 6 months, I hope/assume they figured it out. 😅  

The problem was likely that they only imported/copied the plugin to their own project and did not actually use the entire sample project. Meaning, they did not copy the input configuration from the sample project to their own project. So pressing a key does nothing because it's not configured in the Input section of their own project settings. Except the P and D key work because they are called directly by reference in the blueprint.

jamesford42, 2025-02-28:
My experimentation with jsbsim ended abruptly as project changed direction. Unable to further clarify anything--Sorry folks.

closed 2025-02-28 by jamesford42