JSBSim-Team/jsbsim #1125: aerodynamics_2_graphviz

yanzixiang, 2024-07-20:
hello,everyone,I am a newbie to jsbsm,
i want to analyse the aerodynamics of an aircraft using graphviz,
but jsbsim sames have NOT export this kind of information out,
so i add some function and variable,
and use the [gvpp ](https://github.com/compilin/gvpp/) to export an dot,
like this.
I am a newbie to cmake ,too,
I make two project using Virtual Studio,
one called jsbsim_static,
another one called aerodynamics_2_graphviz,
I got the result like this.

[image]
[image]
[image]
[image]
[image]
[image]

I think maybe this can help someone else, a little.
I wish someone can help to write an "CMakeLists.txt" for them.

seanmcleod, 2024-07-20:
> i want to analyse the aerodynamics of an aircraft using graphviz,

What sort of analysis are you trying to perform? And how do these graphs help in that?

yanzixiang, 2024-07-21:
> > i want to analyse the aerodynamics of an aircraft using graphviz,
> 
> What sort of analysis are you trying to perform? And how do these graphs help in that?

the full equation, which parameter is counted in the axis force calculation,
seems very hard to understand.

the pictures are made by analysing the aerodynamics of aircraft included in jsbsim,f16,
[image]
the left labels refer to the functions defined here
[image]

seanmcleod, 2024-07-21:
When I looked at the visual graphs I must say I thought they were harder to read than the actual XML in the aerodynamics section.

If you want a more compact summary wouldn't something like this be better/easier to read? Basically a summary getting rid of all the XML tags etc. and also not showing the contents of `Table` elements.

```
aero/coefficient/CDDh  (Drag_due_to_horizontal_tail_deflection)
   - aero/qbar-psf
   - metrics/Sw-sqft
   - Table
      - aero/alpha-rad
      - fcs/elevator-pos-rad

aero/coefficient/CDmach  (Drag_due_to_mach)
   - aero/qbar-psf
   - metrics/Sw-sqft
   - Table
      - velocities/mach

aero/coefficient/CDDlef  (Drag_due_to_leading_edge_flap_deflection)
   - aero/qbar-psf
   - metrics/Sw-sqft
   - fcs/lef-pos-rad
   - Table
      - aero/alpha-rad
....
....
```

Ideally maybe with an option with a link to graph renderings of tables using something like this - https://github.com/NikolaiVChr/jsb-to-png

yanzixiang, 2024-07-22:
> When I looked at the visual graphs I must say I thought they were harder to read than the actual XML in the aerodynamics section.
>
> If you want a more compact summary wouldn't something like this be [...]

your way can just follow the way from aero/coefficient/CDDh to aero/qbar-psf,metrics/Sw-sqft,aero/alpha-rad,fcs/elevator-pos-rad....
sometimes, people like me want to think in the opposite direction,
what does fcs/elevator-pos-rad change and effect the aerodynamic axis

> Ideally maybe with an option with a link to graph renderings of tables using something like this - https://github.com/NikolaiVChr/jsb-to-png

yes, that is just what i am trying to do....

seanmcleod, 2024-07-23:
> sometimes, people like me want to think in the opposite direction,

Yep, I read/looked at the graphs from left to right and followed the direction of the arrows.

> what does fcs/elevator-pos-rad change and effect the aerodynamic axis

Hmm, given the number of crisscrossing lines etc. again I must say that for me trying to read and follow the graph is harder than either scanning the summary example I gave above, or having the summary tool allow you to select a parameter, e.g. `fcs/elevator-pos-rad` and then filter the list of functions that use it etc.

Not saying that no-one will find these graphs useful, just for me personally so far I find them more difficult to read.

yanzixiang, 2024-07-24:
> > Ideally maybe with an option with a link to graph renderings of tables using something like this - https://github.com/NikolaiVChr/jsb-to-png
> 
> yes, that is just what i am trying to do....

i add the option for embbing the jsb-to-png image of an table,
and get this.

the number of crisscrossing lines is depending on the complexity of the aerodynamics,
maybe the appearance of the graphs is not so good,
but it is necessary to give people the possiblity to get the information of the aerodynamics,
and make better graphs....

[image]
[image]
[image]
[image]
[image]
[image]

agodemar, 2024-07-24:
@yanzixiang @seanmcleod 
I like the effort to transform the XML into something visual. It can become a piece of software that goes in the `utils` section, besides `aeromatic`.

seanmcleod, 2024-07-24:
@agodemar there are a some comments from @bcoconni and myself in the pull request - https://github.com/JSBSim-Team/jsbsim/pull/1126

closed 2024-07-25 by yanzixiang