JSBSim-Team/jsbsim #1139: Helicopters cannot be trimmed in flight

bcoconni, 2024-08-15:
**I'm submitting a ...**
  - [X] bug report
  - [X] feature request
  - [ ] support request => **Please do not submit support request here**, see note at the top of this template.

**Describe the issue**
As per the discussion #1132, helicopters can't be trimmed in flight.

**What is the current behavior?**
Currently, helicopters can be trimmed on ground by setting `simulation/do_simple_trim` to 2 (i.e. trim on ground).

However, helicopters cannot be trimmed in flight because helicopters produce lift by their rotor while the trim algorithm assumes that lift is produced by wings and hence adjusts the angle of attack to maintain constant altitude.

https://github.com/JSBSim-Team/jsbsim/blob/b41fcddc505baf041a3082ac3be2312e392c7792/src/initialization/FGTrim.cpp#L808-L809

**What is the expected behavior?**
JSBSim should be able to trim helicopters in flight just as it does for aircraft.

dershow, 2024-08-15:
In general helicopters are inherently unstable while most airplanes are stable.  So, trim algorithms have to handle them differently.   

--Adam

gallonmate, 2024-08-16:
So helicopters do successfully trim on ground? I'm seeing trim failure for `trim on ground` when testing in UE5 and with helicopter engine running. With engine off, trim is successful. I have not debugged or looked further in it. Just thought I would clarify. 

There are a few other JSBSim helicopters being used in Flightgear, so I'd be curious what they do for trim, if at all.

bcoconni, 2024-08-16:
> So helicopters do successfully trim on ground?

Well, this is what we inferred from the discussion mentioned in the PR description. I have to admit that I did not test helicopter trimming myself. But this statement seemed reasonable as, from an algorithmic point of view, trimming an helicopter on ground is not different than trimming an aircraft on ground: you have a mass and a list of contact points and the trim algorithm has to compute the location of the contact points such that the solid (helicopter or aircraft) is in a stable equilibrium.

https://github.com/JSBSim-Team/jsbsim/blob/0b99bf8b173fe21cd3a1481695ff199b5e4a4335/src/initialization/FGTrim.cpp#L349-L373

> I'm seeing trim failure for trim on ground when testing in UE5 and with helicopter engine running. With engine off, trim is successful. I have not debugged or looked further in it. Just thought I would clarify.

Thanks for the clarification. It's possible that this limitation applies to aircraft as well.

seanmcleod, 2024-08-16:
> There are a few other JSBSim helicopters being used in Flightgear, so I'd be curious what they do for trim, if at all.

Are you sure that they're JSBSim based as opposed to YASim based? When I looked around for helicopter JSBSim FDMs a couple of years ago the majority I found at the time were YASim based.

gallonmate, 2024-08-16:
> Thanks for the clarification. It's possible that this limitation applies to aircraft as well.

Most aircraft I've used so far, have trimmed successfully on ground with engines running. It was just something I noticed that was different with the helicopter.

> Are you sure that they're JSBSim based as opposed to YASim based? When I looked around for helicopter JSBSim FDMs a couple of years ago the majority I found at the time were YASim based.

Correct there are only a few JSBSim helicopters among the many for Flightgear. There was maybe 4 made with JSBSim that I found. The Alouette-III and the R22 are both JSBSim helicopters currently available directly through flightgear. The other ones were available from third parties but I can't find them at the moment. Anyways, it was more a curiosity to see how/if they use jsbsim trim.

andgi, 2024-08-17:
Hi all,

I have a few airships and boats that don't trim that well.. but I didn't
expect that anyway. :)

See https://github.com/andgi/ for my examples. The repositories are called
FlightGear-<name>.

Cheers,
Anders