JSBSim-Team/jsbsim #1159: Request: round tag

Octal450, 2024-09-18:
**I'm submitting a ...**
  - [x] feature request

**Describe the request**

Inside <function>, currently I am using the integer tag with a summation of 0.5 to round values. It would be great to have a tag that works like this:

`
<round>
    <property>myprop</property>
    <unit>5</unit>
</round>
`

In this case, rounding myprop to 5.

Which support rounding to any arbitrary number (10, 5, 12.5, etc) including decimal like 0.1, 0.2, etc.

**What is the motivation / use case for changing the behavior?**
Not needing to do this all over my code :D
[image]

Thanks,
Josh

bcoconni, 2024-09-28:
You could define a function template (note the hashtag `#` that will be replaced by the actual property when the template function will be applied later):
```xml
<function name="my-correct-round" type="template">
  <quotient>
    <integer>
      <sum>
        <product>
          <property> # </property>
          <value> 10.0 </value>
        </product>
        <value> 0.5 </value>
      </sum>
    </integer>
    <value> 10.0 </value>
  </quotient>
</function>
```
then apply it to the property `hydraulics/stabilizer/final-deg` using the `apply=""` attribute:
```xml
<sum>
  <property apply="my-correct-round"> hydraulics/stabilizer/final-deg </property>
  <property> some/other/property </property>
</sum>
```

seanmcleod, 2024-09-28:
Ah, I had forgotten about the support for template functions, even though I added a bunch a number of years ago 😉 

Add a set of template functions for unit conversions to SI units - https://github.com/JSBSim-Team/jsbsim/commit/cfd181b5f8a99f9ee77abe209474125ef25ca4d6

seanmcleod, 2024-09-28:
Although I'm sure the next request will be to allow parameters to be passed to the template functions 😉

Octal450, 2024-09-29:
I didn't realize you could do that. But I would have to copy it thru all the planes. Hence the request. :)

-J

seanmcleod, 2024-09-29:
Ideally you'd stick this in a single `rounding.xml` file and then include/reference it in each aircraft's FDM that requires it.

https://github.com/JSBSim-Team/jsbsim/blob/57361d0756a7a4d477d80f786e381317d8722671/scripts/737_cruise.xml#L78-L83

seanmcleod, 2024-09-29:
So I tried a quick test with the aerodynamics section.

```xml
<aerodynamics file="unitconversions">
......
......
        <axis name="DRAG">

            <function name="aero/coefficient/CD0">
                <description>Drag_at_zero_lift</description>
                <product>
                    <property apply="convert-R-to-K">aero/qbar-psf</property>
                    <property>metrics/Sw-sqft</property>
                      <table>
                          <independentVar>aero/alpha-rad</independentVar>
                          <tableData>
                             -1.57	1.5000
                             -0.26	0.0420
                              0.00	0.0210
                              0.26	0.0420
                              1.57	1.5000
                          </tableData>
                      </table>
                </product>
            </function>
```

However I got the following error. Basically none of the functions defined within the `aerodynamics` section but outside of a specific axis were found.

```sh
In file aircraft/737/737.xml: line 585
Property aero/function/kCDge does not exist
FATAL ERROR: JSBSim terminated with an exception.
The message was: FGPropertyValue::GetValue() The property aero/function/kCDge does not exist.
```

https://github.com/JSBSim-Team/jsbsim/blob/57361d0756a7a4d477d80f786e381317d8722671/aircraft/737/737.xml#L451-L516

I then commented out any references to these functions and then the script succeeded and I noticed a difference in thrust required etc., i.e. my template function was being applied. But it appears that by 'including' my template functions they effectively replace any other functions defined in the aerodynamics section outside a specific axis as opposed to ending up with a union of the functions.

I haven't looked at the source code yet. Not sure if this is intentional.

bcoconni, 2024-09-29:
> But I would have to copy it thru all the planes. Hence the request. :)

Similar requests (refer to issues #94, #126) have been noted, and JSBSim cannot accommodate every conceivable mathematical function in C++. Doing so would increase the maintenance burden for specialized use cases and contradict the purpose of employing XML to circumvent hardcoded C++ functionalities. The introduction of template functions was intended to address this very issue.

bcoconni, 2024-09-29:
> However I got the following error. Basically none of the functions defined within the aerodynamics section but outside of a specific axis were found.

This is most likely due to this code since an `<output>` section is being loaded in an `<aerodynamics>` section.
https://github.com/JSBSim-Team/jsbsim/blob/57361d0756a7a4d477d80f786e381317d8722671/src/input_output/FGModelLoader.cpp#L78-L81

I would have expected the code from `<output>` to be ignored by `<aerodynamics>` but that is obviously not the case. This looks very much like a bug…

seanmcleod, 2024-09-29:
Ah, sorry, I forgot to mention I had actually edited my local copy of `unitconversions.xml`, to:

```xml
<?xml version="1.0"?>
<aerodynamics>
  
  <!-- 
  JSBSim exported non-SI units:
  
  - Angles
  rad
```

Before I made the change I got this error:

```sh
In file aircraft/737/737.xml: line 494
 Read model 'output' while expecting model 'aerodynamics'

In file aircraft/737/737.xml: line 494
```

Octal450, 2024-09-29:
@bcoconni 
I completely understand that, and it make sense for what was mention in that issue.

However, I feel that rounding is one of the "basic" operations that should be implemented, similar to how ceiling, floor was added. 

With rounding, it is not a specific use case, but a very common need in all kinds of things. Hence I find that it should be included. 

But I completely understand why you had said about template for the others, which are more complex and specific operations.

Kind Regards,
Josh

bcoconni, 2024-09-29:
> With rounding, it is not a specific use case, but a very common need in all kinds of things. Hence I find that it should be included.

JSBSim already includes the standard rounding functions `floor`, `ceil`, and `integer`, which suffice for the great majority of use cases. Your particular requirement (to multiply by 10, add 0.5, round to an integer, and then divide by 10) is indeed specific and does not fall into the 'common need' category.

https://github.com/JSBSim-Team/jsbsim/blob/57361d0756a7a4d477d80f786e381317d8722671/src/math/FGFunction.cpp#L520-L523
https://github.com/JSBSim-Team/jsbsim/blob/57361d0756a7a4d477d80f786e381317d8722671/src/math/FGFunction.cpp#L546-L552

Octal450, 2024-09-29:
Hi @bcoconni 
I don't understand that. floor/ceil and integer do not round the number. Hence why I am doing that thing with 10 and 0.5

Actually, you don't need the * 10 thing. that's just because I'm rounding to the 10s place not the 1s place.

By rounding I mean
0.5 -> 1
0.4999 -> 0
0.8 -> 1
0.2 -> 0

Behavior with floor:
All 0.x goes to 0

Behavior with ceiling:
All 0.x goes to 1

Integer = Floor. Actually I don't understand why Floor was added, is the same as integer, no?

Kind Regards,
Josh

Octal450, 2024-09-29:
My example in the first line, rounds the stabilizer to the 0.1's place.

In the above post, I gave examples for the 1's place.

The equivalent current code would be:

```<integer> <!-- or floor -->
    <sum>
        <property>hydraulics/stabilizer/final-deg</property>
        <value>0.5</value>
    </sum>
</integer```

Bests,
Josh

Octal450, 2024-09-29:
My example in the first post, rounds the stabilizer to the 0.1's place.

In the above post, I gave examples for the 1's place.

The equivalent current code would be:

```
<integer> <!-- or floor -->
    <sum>
        <property>hydraulics/stabilizer/final-deg</property>
        <value>0.5</value>
    </sum>
</integer```

Bests,
Josh

seanmcleod, 2024-09-30:
> Integer = Floor. Actually I don't understand why Floor was added, is the same as integer, no?

`floor()` rounds down. `int()` truncates. The difference is clear when you use negative numbers:

```python
>>> import math
>>> math.floor(-3.5)
-4
>>> int(-3.5)
-3
```

Octal450, 2024-09-30:
I see. I'll have to go fix that.

Thanks.

I maintain my request though

Regards,
Josh

bcoconni, 2024-09-30:
> I don't understand that. floor/ceil and integer do not round the number. Hence why I am doing that thing with 10 and 0.5

Well, `floor` and `ceil` **do** round numbers. There are many ways to round a real number to an integer. To understand this better, consider reading the Wikipedia article on [rounding](https://en.wikipedia.org/wiki/Rounding), particularly the section on [rounding to an integer](https://en.wikipedia.org/wiki/Rounding#Rounding_to_integer), which details 10 (ten) different rounding methods  including `floor` (*rounding down*) and `ceil` (*rounding up*).

> I maintain my request though

Your request has shifted since your initial post, making it challenging to understand what exactly is that you maintain. However this discussion made me realize that the C/C++ `round` function (i.e. *rounding half away from zero* according to the Wikipedia article) is far from being trivial as is very well explained in [this post from StackOverflow](https://stackoverflow.com/a/24348037/7941251). So to settle this discussion, I propose the integration of the C/C++ `round` function into JSBSim, without further additions.

seanmcleod, 2024-09-30:
@bcoconni I think what @Octal450 was originally asking for is the following from the Wikipedia article you linked to.

https://en.wikipedia.org/wiki/Rounding#Rounding_to_a_specified_multiple

```python
>>> def roundmult(x, m):
...    return round(x/m) * m

>>> roundmult(93, 10)
90
>>> roundmult(95.1, 10)
100
>>> roundmult(95.1, 0.5)
95.0
>>> roundmult(95.2345, 0.01)
95.23
```

However, even using the built-in `round()` of Python to handle the edge cases the `roundmult()` still has some unexpected results.

```python
>>> roundmult(95.1, 0.1)
95.10000000000001
>>> roundmult(95.2345, 0.001)
95.23400000000001
```

seanmcleod, 2024-09-30:
> expected 95.2345 to round to 95.235 not 95.234

Yep, that's why I used it as one of the unexpected results.

```python
>>> roundmult(95.2345000001, 0.001)
95.235

>>> 95.2345/0.001
95234.5
>>> round(95.2345/0.001)
95234
```

seanmcleod, 2024-10-03:
I've submitted a PR, see - https://github.com/JSBSim-Team/jsbsim/pull/1170, and note my question/comment in the PR.

Octal450, 2024-10-07:
Many thanks.!

closed 2024-10-12 by seanmcleod