JSBSim-Team/jsbsim #1227: Number parsing is locale-dependent

vranki, 2025-02-05:
**I'm submitting a ...**
  - [x] bug report
  - [ ] feature request
  - [ ] support request => **Please do not submit support request here**, see note at the top of this template.

**Describe the issue**
While reading XML files, JSBSim uses locale-dependent string parsing to read numbers. This causes decimal numbers to be cut, if decimal separator is not what JSBSim expects. Locale-dependent parsing should not be used when loading formatted data. 

**What is the current behavior?**
If LC_NUMERIC uses comma as decimal separator (for example fi_FI.UTF-8 ) decimal numbers will be rounded without warning, causing values to be wrong in simulation.

**What is the expected behavior?**
LC_NUMERIC or other locale specific variables should not be used when parsing formatted data.

See for example 
https://stackoverflow.com/questions/25579637/parsing-a-float-from-a-string-in-different-locales
on how to change behavior in c++.

**What is the motivation / use case for changing the behavior?**

Quite serious bug which was not obvious as JSBSim seemed to mostly work. Can cause wrong results if unnoticed.

**Please tell us about your environment:**
 - OS Ubuntu 22.04
 - JSBSim version 1.2.0

**Other information**

Steps to reproduce:

LC_NUMERIC=fi_FI.UTF-8 ./jsbsim <- Values are rounded

LC_NUMERIC=en_US.UTF-8 ./jsbsim <- Works as expected

bcoconni, 2025-02-09:
Thanks for the bug report.

This was supposed to be fixed by the PR #799 which has been introduced in JSBSim v1.1.14+. I guess this is due to some conversion functions which do not use the function `atof_locale_c` from the PR #799.

bcoconni, 2025-02-09:
The PR #1229 has been submitted for review.
If you have the opportunity to test it, let us know if that fixes your issue.

bcoconni, 2025-02-22:
The PR #1229 has been merged in `master`.

bcoconni, 2025-03-23:
@vranki, version 1.2.2 of JSBSim has been released and includes the fixes I have mentioned above. Could you please test this new version and let us know if that fixes your issue ?

vranki, 2025-04-08:
Seems to work now. Thanks for fixing this!