JSBSim-Team/jsbsim #1239: Command Console causing unexpected Blueprint Node errors in Unreal Engine 5

SLGY, 2025-02-24:
**I'm submitting a ...**
  - [x] bug report @gallonmate 
  - [ ] feature request
  - [ ] support request => **Please do not submit support request here**, see note at the top of this template.

**Describe the issue**
After updating Unreal Engine plugin with pull #1221 - the Command Console throws the following error when attempting to read a number of JSBSim parameters at runtime (not at compiling):
`"BP_Airliner_C_1: JSBSim Command Console Blueprint Node Error: Property name must be alphanumeric and limited to these -._/ four characters. Do not use parentheses *(RW)* in your property name"`

**What is the current behavior?**
The issue is repeatable in the UEReferenceApp project. Add a Command Console node to the Tick sequence and attempt to read and print this example parameter `aero/alpha-wing-rad`. Then compile and run the project.

[image]

The following error is thrown without any reference to the specific node causing the problem (difficult to troubleshoot when many Command Consoles exist in a project):

[image]

**What is the expected behavior?**
The application runs, and the Command Console doesn't throw an error when attempting to read a parameter from JSBSim.

**Please tell us about your environment:**
Win 10
Unreal Engine 5.5.3
JSBSim v1.2.1

**Other information**

gallonmate, 2025-02-25:
I can confirm there is a bug/syntax mistake.
I only realized earlier today there was a problem.
The solution in c++ is:
`std::regex("^[a-zA-Z0-9.-_/]+$")))` needs to have `\\` added to become this:
`std::regex("^[a-zA-Z0-9.\\-_/]+$")))`

Editing these two lines will solve the bug:
https://github.com/JSBSim-Team/jsbsim/blob/e6208a337f7de3d396672da8782890430d9689c6/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/JSBSimFlightDynamicsModel/Private/JSBSimMovementComponent.cpp#L113

https://github.com/JSBSim-Team/jsbsim/blob/e6208a337f7de3d396672da8782890430d9689c6/UnrealEngine/Plugins/JSBSimFlightDynamicsModel/Source/JSBSimFlightDynamicsModel/Private/JSBSimMovementComponent.cpp#L149

I will submit the fix for the plugin soon.

SLGY, 2025-02-25:
Thankyou, that seems to have fixed the Command Consoles that have a `/` character. Do the `[` and ']' characters also have an implementation issue here? For example I'm using gear/unit[1]/WOW and it throws the error.

gallonmate, 2025-02-25:
Yes that's another mistake by me, it was not included in the check. Here's code to fix that too:
`std::regex("^[a-zA-Z0-9\\[\\]\\-._/]+$")))`

When I submit the fix to the repo, I'll also update the error message to display the property name that is causing an error.

SLGY, 2025-02-27:
@gallonmate 

Following the update I'm now getting crashes when the Command Console is used outside the primary actor blueprint. For example in the UEReferenceApp, add a Command Console to the animation blueprint (ABP_CommercialPlane, see pic below) and attempt to compile it, for me this results in an app crash.

[image]

In my project, the Command Consoles located in it's Animation Blueprint were being used for animating the landing gear compression and after updating to the latest commit, the app now crashes when loading the blueprint. Only way to fix it was to remove the Command Consoles from the Animation Blueprint.

gallonmate, 2025-02-27:
Hmm, the node was never designed for this. But there are no rules on it's usage. 

Generally with animation blueprints you grab your data from the corresponding main blueprint actor. 

The animation blueprint by default is updating the live animation preview window, which is then running the Event Update Animation and thereby calling all the code in the grab. Since JSBSim library is not always loaded before the game starts, the node crashes.

To prevent crashes in the animation blueprint, you can add a check before calling the node to see if BeginPlay has happened. Here's a basic example. You can also use the EventBeginPlay to set a boolean, instead of using the gate.

[image]

SLGY, 2025-02-27:
@gallonmate 

Thanks a lot, that really clarifies things. I guess my animation blueprint must have been set up previously in a way that just happened to work, only running after BeginPlay had been triggered at some point. Adding the gate ensures it won’t crash now. 👌

closed 2025-02-27 by SLGY