JSBSim-Team/jsbsim #1249: FGTurbine Corrections

Octal450, 2025-03-26:
**I'm submitting a ...**
  - [ ] bug report
  - [x] feature request
  - [ ] support request => **Please do not submit support request here**, see note at the top of this template.

**Describe the request**
The way FGTurbine computes thrust is not correct.

Firstly, I have been analyzing thrust vs fan speed graphs for various engines for a while now, and have found that on average, the curve that best fits is N1 fan speed to the 3.5 power. This is most accurate for high-bypass engines, but the couple of low-bypass engines I've looked at still seem to fit this definition fairly well.

FGTurbine currently does N2 squared.
https://github.com/JSBSim-Team/jsbsim/blob/e1ea4e85524e209037ec868f5a1383ec2ef3c7a1/src/models/propulsion/FGTurbine.cpp#L218

This results in thrust being "more linear" than it should be.

Example of a correct table:

[image]

Secondly, FGTurbine assumes that the rated (mil) thrust occurs at the maximum rotor speed. This is not correct. I have not encountered a single engine in which this occurs. The rated thrust always occurs below the maximum rotor speed and varies with air density (which I define with temp, alt, etc).
https://github.com/JSBSim-Team/jsbsim/blob/e1ea4e85524e209037ec868f5a1383ec2ef3c7a1/src/models/propulsion/FGTurbine.cpp#L217

Instead, we should be allowed to define an N1 table where the rated (mil) thrust (and probably aug thrust too) occurs. If the table is not defined, then I suggestion reversion to the current "max rotor speed" method for backwards compatibility. I am currently fixing this aircraft side, but it would be very nice to see implemented properly.

**What is the motivation / use case for changing the behavior?**
Being able to deliver more accurate jet turbine engine models in FlightGear Aircraft.

On another note, the EPR computation is also wrong, but it's hard to do a 1 size fits all here. This changes would "help" it be slightly less wrong, curvature wise, but it still is incorrect.

**Please tell us about your environment:**

agodemar, 2025-03-26:
@Octal450 that's a great catch. Thank you for pointing out this bug.

Another approach in modeling jet and turbofan engine would be providing lookup tables for maximum (steady-state) thrust as a function of flight Mach number and altitude (and perhaps of engine rating state); providing also time delays for transient scenarios, when changing throttle settings.
This would require modeling thrust as a generic applied force and authoring XML files inspite of hard coding engine dynamics.

Octal450, 2025-03-26:
Hi @agodemar 
Thanks for your comments.

I have considered that, but I found that with some tweaks, we can make FGTurbine work and prevent re-inventing the real for some of it.

All I need to get it working is the corrected curve and the ability to define at what N1 milthrust occurs. Other thrust limits work "intuitively" then.

See: https://github.com/Octal450/MD-11/blob/master/Systems/md-11-ge-engine-fadec.xml

Bests,
Josh

seanmcleod70, 2025-03-27:
@Octal450 can you provide links to, or copies of the thrust vs fan speed graphs that you've been using? Ideally showing both N1 and N2.

> providing lookup tables for maximum (steady-state) thrust as a function of flight Mach number and altitude 

@agodemar but isn't that pretty much what we currently have?

https://github.com/JSBSim-Team/jsbsim/blob/e1ea4e85524e209037ec868f5a1383ec2ef3c7a1/engine/CFM56.xml#L42-L57

seanmcleod70, 2025-03-28:
So taking a quick look at the turbine code, it looks like it's quite a simple model. I guess as @bcoconni has mentioned in various replies over the years, it's not a high-fidelity engineering model.

So N1 and N2 have a specified range:

https://github.com/JSBSim-Team/jsbsim/blob/e1ea4e85524e209037ec868f5a1383ec2ef3c7a1/engine/CFM56.xml#L19-L22

And effectively, ignoring the transient case which includes spool up/down time, the `ThrottlePos` directly maps to an N1/N2 value.

https://github.com/JSBSim-Team/jsbsim/blob/e1ea4e85524e209037ec868f5a1383ec2ef3c7a1/src/models/propulsion/FGTurbine.cpp#L213-L216

The following graph shows the mapping between `ThrottlePos`, `N1` and `N2` to thrust, with the red line showing a $x^2$ mapping and the blue line showing a $x^{3.5}$ mapping. With example values for min/max for N1 and N2 for the CFM56 engine file.

[image]

Idle thrust and max thrust are looked up using engine tables indexed by Mach and density altitude.

seanmcleod70, 2025-03-28:
> Example of a correct table:

@Octal450 this doesn't look like a good example of a curve matching $x^{3.5}$, rather it looks like a very good fit for $x^2$.

[image]

By the way, if you're not familiar with **Engauge Digitizer** I'd suggest taking a look since it's a very useful tool for digitizing graphs. If you know of something better please mention it.

Octal450, 2025-03-28:
Hi @seanmcleod70 
OK, I did miss the norm part (oops) in my initial evaluation.

However, what you may be missing is that you are taking the x^2 based off the 0-1 domain. This results in an incorrect curve in your chart.

In the following, I have plotted N2norm^2 for N1 values 25-110, (based on 105% being the rated thrust, and 25% being idle, and 110 being max). This maps it to 0 thru 1 domain as FGTurbine does. Then I've plotted N1 ^ 3.5 / norm factor based off 105%. Here is the comparison, you can see it's pretty close but different. I didn't bother to correct the 25% case to 0 since it's near 0 anyways, but you get the idea. Lastly, I've plotted the curve from the chart, normalized to the same values.

[image]

I should note that the graph image shown above for that engine starts at 35%, but the actual idle value for that engine is about 20% N1. Hence the curve does not have a domain of 0-1 when normalized.

You will also see that the chart value isn't quite 3.5, it matches a little closer to a slightly lower exponent, but there are two things I've noticed, the exponent that best fits is not constant, and the exact curve varies. I've seen values between ~3.2 and ~3.7 when curve fitting various tables over the years, hence my "settling" on 3.5 as a "common value". Regardless, it's quite steeper than the N2norm^2 curve which is easily seen if I skew my normalization to make it "appear" closer to the blue line, but near the high end it appears to flatten out slightly.

Kind Regards,
Josh

Octal450, 2025-03-28:
> [@agodemar](https://github.com/agodemar) but isn't that pretty much what we currently have?
> 

No. That is the milthrust varience factor. What we were talking about is the rated thrust N1 value, which varies.

Kind Regards,
Josh

seanmcleod70, 2025-03-28:
> No. That is the milthrust varience factor. What we were talking about is the rated thrust N1 value, which varies.

I'm not following.

https://github.com/JSBSim-Team/jsbsim/blob/e1ea4e85524e209037ec868f5a1383ec2ef3c7a1/engine/CFM56.xml#L42-L48

So given the above for the CFM56 on a standard ISA day at sea-level the static thrust at 100% N1 would be scaled by 1.0. But assuming a very hot day such that the density altitude is 10,000ft then at 100% N1 the engine thrust would be scaled by 0.74. So in order to achieve the same thrust you would need an N1 value of greater than 100%. Obviously with some upper limit on N1 and/or upper limits on EGT etc.

Same applies for a very cold day, say giving a density altitude of -10,000ft then you'd get to the same thrust value long before you got to 100% N1.

So isn't this the rated thrust N1 variance you're expecting to see?

seanmcleod70, 2025-03-28:
@Octal450 can you share more of the data you have for the CFM56-7B27?

This reference - https://www.aircraft-commerce.com/wp-content/uploads/aircraft-commerce-docs/Aircraft%20guides/CFM56-7B/ISSUE58_CFM56_7B_SPECS.pdf mentions a thrust rating of 27,300lbf. 

You seem to be using a figure of 105% N1 for this thrust rating? And then 110% N1 for some max thrust greater than this? Where did you find these figures?

Also note that the graph you posted above for this engine is for Mach 0.3, so the thrust values in the graph are on the order of 92.8% of the static thrust values if you assume the thrust scale factor due to Mach of the CFM56 engine in JSBSim.

Octal450, 2025-03-28:
Hi,
105% is a for example value. For various engine's I've seen rated for between 85% and 113% N1 for milthrust @ sealevel @ rated temp.

Yes you are correct that as temp decreases N1 value should drop.

I need to be able to have granular control of what N1 achieves milthrust at temp/alt. Currently I do this with scaling factors.

Example table in the Airbus CFM, reated at 45C:
 -45   -15    15    45     75
79.5  84.6  89.2   93.8  98.3

For temperatures @ sea level.

Note that the value decreases above 45C in the actual takeoff command n1 due to hitting egt limits/etc.

 -45   -15    15    45     75
79.5  84.6  89.2   93.8  89.1

Thus it can no longer reach rated thrust. 

However, the max allowable rotor speeds are 104%. so maxn1 is set to 104, but i need to be able to set the rated thrust differently.

Also this allows simulation for dual-rated engines.

Essentially, i use the milthrust table *only* to define how the thrust curve changes, because it's not linear with the n1 fan speed changes it seems. I need to be able to define both. 

Kind Regards,
Josh

seanmcleod70, 2025-03-29:
@Octal450 if you could share some of the graphs you've been using it would be a lot easier to make sure we've got the same understanding. Ideally one showing the complete range of what you're trying to duplicate, for example the one you did share, 'correct table' doesn't really show enough. A full picture is worth a thousand words 😉 

Typically, the first thing I do when a request comes in is to check the JSBSim source to see how things are currently implemented, which I've summarized above with the relevant snippets of code and the diagram I generated. So that we're all on the same page. Then check whether there is a way to implement the new request via existing JSBSim functionality.

So based on your comments, I think what you're predominately getting at is the concept of a 'flat rated engine', and wanting to be able to model it?

https://www.metabunk.org/attachments/docslide-us_a-320-engine-pdf.16733/

[image]

So in the flat rated engine case the FADEC is able to provide a fixed and consistent thrust value over a specific temperature/density altitude range, rather than the thrust varying continuously like it will for the JSBSim CFM56. I'm assuming we can interchange temperature and density altitude.

https://github.com/JSBSim-Team/jsbsim/blob/e1ea4e85524e209037ec868f5a1383ec2ef3c7a1/engine/CFM56.xml#L42-L48 

So in the flat rated engine graph example, let's assume N1 max is 104% at point B that I've annotated. Which means that the engine manufacturer has calculated what combination of fuel to inject at this ambient temp/density altitude (in this case Std Day sea level + 15C) which will produce the maximum allowable N1 fan speed and limit the EGT to it's rated maximum. This combination of fuel and air mass flow (from N1) will produce the rated thrust of 25,000lbs.

Thereafter from point B to C the FADEC will actually reduce the amount of fuel injected in order to keep EGT constant at max EGT, which will in turn mean that the fan will spin slower so N1 will reduce. The combination of less air mass flow from both a slower N1 plus the density altitude increasing will therefore mean less thrust, therefore the negative thrust slope from B to C mirroring negative slope of N1.

Which means given the calculations to achieve point B, the N1 value at point A (ISA -15C) in order to generate the rated thrust will be a fair bit lower than the max N1 value of 104%, say 95% for example. Given the higher air density the same thrust can be generated with less air mass flow, so N1 lower, and also less fuel required. But the FADEC also limits things here in that the engine could actually produce more thrust than the rated thrust at this point, but the engine is certified to only produce a maximum thrust based on the rated thrust.

Octal450, 2025-03-30:
Hi,
The is not correct. The max rotor speeds are based on mechanical limitations and are NOT in any way, shape, or form related to the rated thrust N1. The only requirement is that the rated thrust N1 occurs below the max rotor speed for obvious reasons.

On the CFM56 engine specifically the options on the A320 series, the N1 max is 104% across the whole line-up. But the rated thrust N1 occurs at a much lower number. @ rated temp of 45C, this is 93.8% N1 and at sea level, this is the highest number. As temp reduces, the N1 required reduces. And as temp increases above rated, the TOGA N1 then has to decrease due to EGT limits, but the N1 required to produce rated thrust would keep increasing.

I'm going to disregard the A320 at the moment as it doesn't yet use my latest implementations.

Lets look at the CF6-80C2D1F which I have the most info for.

The following table https://github.com/Octal450/MD-11/blob/master/Systems/md-11-ge-engine-fadec.xml#L180
Is the takeoff N1 required to reach rated thrust, vs altitude and TAT.

Max N1 on this engine is 117.5% N1. If you scroll down more, you'll see tables for MCT, Climb, and Cruise power limits. They are nowhere near the max N1. However, if you switch the FADEC into ALTN, the limitations are "removed" and moving the power lever all the way up would command 117.5% N1 - which with use would likely damage the engine, however, would produce thrust "above" the rated thrust!

This is implemented to how it works in the real airplane and these tables are directly taken from the performance manuals of the real airplane.

Hopefully this plants a better picture of how it should work. Implementing this stuff using the milthrust table itself may be possible, but it's harder to compute, as then I have to combine the milthrust reduction with thinner air from altitude + mach effects AND the rated N1 slewing tables into one table... and that's just painful especially because I maintain a multiple projects each with various engine types. For example, if I have two engines, one rated at 20K thrust, and one at 21K thrust, I can keep the milthrust table the same, and just adjust the N1 limit tables - which is what the real engine does also, different N1 limit tables but same everything else.

Kind Regards,
Josh

seanmcleod70, 2025-03-30:
@Octal450 I think you've fixated on the N1 max value so much you didn't answer my general question 😉 

> I think what you're predominately getting at is the concept of a 'flat rated engine', and wanting to be able to model it?

Again, I'm assuming/asking whether this is at least some/large part of what you're trying to model?

I think in general the issue is that the current turbine model in JSBSim is a fairly simple model which doesn't try to model FADEC behavior. So as shown above, there is a simple linear relationship mapping from Throttle position `[0 - 1]` to N1 `[N1min - N1max]` and N2 `[N2min - N2max]`, with a transient model. With a mapping then via $N2_{norm}^2$ to the rated thrust and then scaled via a combination of Mach and density altitude (MilThrust table) to produce a final thrust value.

Given your FADEC tables in your system how are you currently achieving the lines AB and BC and the associated thrust lines from the flat rated diagram above? For example if the pilot moves the thrust levers fully forward for point A (sea level, 0C) how do you limit the final output thrust and display the relevant N1 for point A in the engine displays?

Looking at your FADEC take-off table you linked to it looks like you have for this engine type:

A - N1 106.3%
B - N1 111.7%
C - N1 109.8%

> Is the takeoff N1 required to reach rated thrust, vs altitude and TAT.

So point A and B will give you the rated engine thrust, but given that this engine is only flat rated to 30C it means point C will give you less than the rated thrust.

How have you determined the thrust line for the N1 segment BC?

Octal450, 2025-04-15:
Sorry been busy. Will get back to this.

Kind Regards,
Josh

Octal450, 2025-04-17:
Hi.
I am not talk about FADEC. Same computations required in non FADEC engines. I have data for what the N1s are to produce the flat thrust setting and the gradients of that table is vary. I prefer to adjust this separately otherwise it just becomes complicated to compute.

All I am asking is the ability for the flexibility to to tell JSBSim at which N1 fan speed occurs MilThrust. I can do the rest.

Then the separate issue that the thrust curve is wrong, that why it takes so much power above idle to taxi (shouldn't) and then acts too linear in the high end. Because the the 0 to 1 is wrong.

It not about any FADEC. It's about the compressor percentage speed is not starting from 0. So the curve when modelled from 0 to 1 is producing wrong curvature. 

In engine where the fan speed idle is higher compared to mil, the gradient should not be the same in engine where it isn't.

You will notice my x^3.5 approach would fix both these issues, by making it less flat near idle, and more sharp near max --- and due to this, my EPR to N1 scalars are nearly perfect! N1 to EPRs with x^3.5 approach (ignoring other pressure factors) results in VERY close gradients to real plane, and EPR measure of the thrust, and this is more proof that it is correct. The EPR number generated by JSBSim use the same 0-1 norm ^ 2 approach like thrust compute and even with scaling factors corrected the resulting EPRs are totally wrong.

Kind Regards,
Josh

Octal450, 2025-04-17:
For section B C, the "required N1" keeps increasing, but the actual N1 the pilot or FADEC should set decreases. Thus thrust delivered < rated. The two tables line up in the flat rated section.

Kind Regards,
Josh.

Octal450, 2025-04-17:
My EPR calculation simplified is as follows:

((N1 fan speed * density/temp corrections) ^ 3.5) * scaling factors + corrective factors = EPR.

Result in very close to real numbers.

seanmcleod70, 2025-04-17:
> tell JSBSim at which N1 fan speed occurs MilThrust.

But in your example, at point C (N1 109.8%) doesn't produce the rated thrust, it produces less and is the figure you have from the aircraft performance manual. So, for this point you would have to come up with and calculate some larger N1 value which would produce the rated thrust? And how would you model the FADEC limiting the N1 to 109.8% at this point?

Octal450, 2025-04-18:
The table I linked you is the takeoff thrust table to be set either by the pilot manually, the autothrottle, or the FADEC.

A - N1 106.3%
B - N1 111.7%
C - N1 109.8%

The table which tells JSBSim what the rated thrust occurs at would be as follows:

A - N1 106.3%
B - N1 111.7%
C - N1 117.1%

That should clarify it.

Kind Regards,
Josh

Zaretto, 2025-04-18:
It's fair to say that the model in FGTurbine is simplified and the use of rotor percentages to determine thrust is a simplification that works well enough. The biggest problem is that there isn't a linear relationship between rotor speed and thrust because thrust is affected by temperatures and pressures. 

Rotor percentages are usually based on a design point RPM that can be exceeded in certain conditions resulting in more than 100%. I don't think we should change FGTurbine to model this as it's making simple code more complicated for not much improvement. 

You already can make thrust dependent on N1 without having to change any code.

e.g. 
```  <function name="MilThrust">
    <interpolate1d>
      <p>atmosphere/density-altitude</p>
      <v>0</v>
      <table>
        <independentVar lookup="row">propulsion/engine[0]/n1</independentVar>
        <independentVar lookup="column">velocities/mach</independentVar>
        <tableData>
                    0.3     0.5
             35  0.0000  0.1261
             54  0.1233  0.2709
             66  0.2648  0.4157
             75  0.4064  0.7194
             89  0.7032  0.8735
             93  0.8539  1.0230
            100  1.0000  1.0400
        </tableData>
      </table>
      <v>10000</v>
      <table>
        <independentVar lookup="row">propulsion/engine[0]/n1</independentVar>
        <independentVar lookup="column">velocities/mach</independentVar>
        <tableData>
                    0.3     0.5
             35  0.0000  0.1261
             54  0.1233  0.2709
             66  0.2648  0.4157
             75  0.4064  0.7194
             89  0.7032  0.8735
             93  0.8539  1.0230
            100  1.0000  1.0400
          </tableData>
      </table>
      <v>20000</v>
...etc...
    </interpolate1d>
  </function>
```

Octal450, 2025-04-18:
Hi Richard,
I do not mean to come across rudely, but I think you did not understand what I am asking.

Kind Regards,
Josh

seanmcleod70, 2025-04-18:
Josh, here is a table version of the flat rated graph I posted above with 11 points (including the original A, B, C points) with the N1 figures from your MD11 for take-off thrust at sea-level across a range of temperatures from -40C to +60C. Can you fill in the generated static thrust for the last 3 points?

|Point|Temp (Density Alt)|N1|Static Thrust|
|--|--|--|--|
|1|-40C -6,534ft|98.9|60,000|
|2|-30C -5,346ft|100.8|60,000|
|3|-20C -4,158ft|102.7|60,000|
|4|-10C -2,970ft|104.5|60,000|
|5 A|0C -1,782ft|106.3|60,000|
|6|10C -594ft|108.1|60,000|
|7|20C 594ft|109.9|60,000|
|8 B|30C 1,782ft|111.7|60,000|
|9 C|40C 2,970ft|109.8|?|
|10|50C 4,158ft|108.0|?|
|11|60C -5,346ft|106.2|?|

Can you write a basic Python function which will take in the following inputs and generate the N1 value (ignoring spooling up/down) and thrust output?

- Throttle pos norm [0-1]
- Density altitude (or pressure altitude and temp)
- Min N1
- Max N1
- Rated thrust

Which matches the table entries above.

Zaretto, 2025-04-18:
What I understood was that you're suggesting changing N2<sup>2</sup> because your research indicates that N2<sup>3.5</sup> is a better fit for various engines, without listing the engines and including references to the documents together with an explanation of how you came by this figure.

Without the valid data to backup your case this seems to be a case of changing one magic number for another.

Secondly my point was that we can already include N1 in the MilThrust lookup table so we don't need "to define an N1 table where the rated (mil) thrust (and probably aug thrust too) occurs".

Octal450, 2025-04-18:
@Zaretto, No, I'm suggesting replacing N2 normalized first from min/max N2 to 0-1^2 with N1 %RPM ^3.5 then normalized to 0-1 where max N1^3.5 = 1 and idle N1^3.5 = 0.

Basically, the normalization should occur after the exponent, not before. Why did the exponent change? Well naturally changing the inputs impact the gradient. So the value had to be updated. Look at the graph I put above. It shows the both approaches compared to the CFM56 real world data.

Look how close the green and yellow lines are compared to the blue, which is what JSBSim does now. 

EPR is linear to thrust, when you delete corrective factors (aka, fixing alt, temp, mach scalars in time)
EPR is not linear to N1.

By comparing the EPR to N1s, you can see how the thrust should scale with N1. 

With N1^3.5 approach, I have been able to successfully model multiple engines where the N1 to EPR comparisons are very close to what you can observe from checking YouTube videos showing engine spooling up. I did this for the IAE V2500, PW4062, PW4168, JT8D-217A, JT8D-15A, amongst others and each time it works.

Meanwhile, the normalized value^2 approach produces nonsense in every situation. JSBSim computes EPR like it does thrust (because they're linear, so that makes sense), so you can compare those EPRs and apply a scalar so the default 1-2 range matches min/max and then compare the values throughout the range.

You said the thing in FGTurbine is arbitrary, I agree. Looks like an oversimplifaction. What I said is producing results that line up pretty closely to real data. The only missing link is the thrust computation itself - which is why again - aircraft require TOO much thrust above idle to taxi, but then at the top range, are too linear, resulting in poor gradients for flexible/derated power settings.

Kind Regards,
Josh

Octal450, 2025-04-18:
@seanmcleod70 I need a clarification.

You want me to tell you what the thrust should be at those power settings?

What I'm pointing out here is,

I have 2 sets of data.

I have data telling me how the rated thrust scales with density altitude and mach.

And I have data telling me how the N1 setting for the rated thrust scales with absolute altitude (not density) and temperature.

I think you can imagine how annoying it would be to combine all of this into one thing.

I'll I'm asking is for the ability to have this done behind the scenes so that developers can input data tables directly and not have to do all the math themselves.

Essentially there are 2 issues at hand here which we need to clarify because they are *different*.

Issue 1: The thrust curvature is wrong. This is due to the erroneous norm^2 approach which matches nothing.
Issue 2: It is not possible to tell JSBSim at what N1 (or N2... if y'all are fixed on N2, I can make that work via conversions) milthrust should occur at.

In order to make the simulation correct, both issues should be corrected... I really don't want to have to make the whole engine in a JSBSim system just to get it right... 

Kind Regards,
Josh

seanmcleod70, 2025-04-18:
> N1 setting for the rated thrust scales with absolute altitude (not density) and temperature.

But altitude (pressure) and temperature define the air density, and the air mass flow rate is proportional to the density. Now an aircraft doesn't have a gauge displaying density to the pilots, so when they need to look up an N1 setting for take-off it makes sense to show them a table which they can read using the gauges they have, altimeter (pressure) and an OAT gauge.

> erroneous norm^2 approach which matches nothing.

That may be a bit of an exaggeration. You've only shared one graph depicting Thrust versus N1 over a range from close to idle to close to full N1 and as I pointed out when I looked at that graph and digitized it, for the data presented in the graph it's a very close match to $norm^2$. 

The FADEC data you linked to in your MD11 model showing the N1 setting for take-off based on altitude and temperature gives us the N1 for the segment AB and the associated thrust based on the rated thrust and the N1 values for segment BC but without the associated thrust for the segment BC.

> And I have data telling me how the N1 setting for the rated thrust scales with absolute altitude (not density) and temperature.

But it doesn't for segment BC. 

> I have data telling me how the rated thrust scales with density altitude and mach.

Are you referring to the fairly standard `MilThrust` table?

https://github.com/JSBSim-Team/jsbsim/blob/20066483eb1d1386fa43f3ed85f9fde6d6682f83/engine/CFM56.xml#L42-L57

> You want me to tell you what the thrust should be at those power settings?

Yes. Simply, how would JSBSim calculate the thrust for these points? Show via a simple Python (or computer language of your choice).

Effectively this is the current JSBSim implementation ignoring spooling. 

```python
rated_thrust = 60000

def calc_thrust(throttle_pos_norm, density_alt, mach):
      return throttle_pos_norm**2 * rated_thrust * DensityAltMachLookupScaleFactor(density_alt, mach)
```

So what would the equivalent look like based on your suggestions? Which would match points 1-8 in the table above and also output thrust values for points 9-11.

Plus would output thrust for some arbitrary input, e.g. standard ISA day at 10,000ft with the throttle pos = 0.5.

Ideally we'd also have one or more curves like the one above for a couple of conditions which display thrust on the one axis versus N1 on the other axis covering a large part if not the full N1 range to plot against from an engine/aircraft manufacturer. I've seen detailed performance graphs from piston engine manufacturers, like the one below, but haven't come across so far the equivalent level of performance data from jet turbine manufacturers.

[image]

seanmcleod70, 2025-04-18:
Josh, I only saw your last comment above "Re Issue 2:" after I had posted my comment above. Although it turns out I did make some comments regarding density altitude versus altitude and temperature in my reply above. Here are a couple more.

> So if density altitude drops with lower temps, more thrust than 60,000lbs is produced.

Not if it is a flat rated engine. 

> The temperature and density of the ambient air vary inversely.
>
>Lower temperature = > Higher Density
>
>Higher Temperature = >Lower Density
>
>The amount of airflow (lbs/sec) through the engine is a function of compressor speed and air density. It is greater when the compressor speed and density are high.
>
>The compressor speed is a function of the energy available to the compressors turbine. That energy comes from the combustion or air and fuel, so the turbine turns faster when the fuel flow rate is greater. 
>
>Flat rating is a concept by which an engine, as stated above is restricted, in terms of power or thrust output by the fuel control unit to produce a uniform power or thrust output as the aircraft climbs or descends until the ambient air conditions will not allow the engine to operate at or beyond the maximum thermodynamic output of the engine. Like any other heat engine, the power or thrust produced by an aviation gas turbine (the gas core) depends on the available mass flow rate of working fluid (air in this case) as well as the operating temperature limits of the engine's hot section. 

You mentioned:

> That is why the N1 limits are based on actual altitude, not density altitude.

Pretty sure they're not, as I mentioned above I'm pretty sure the reason they are published in a table format with altitude and temperature is because those are the gauges available to the pilot, they don't have a density gauge or density altitude gauge. Nor does the FADEC unit have an air density input.

> Apples and oranges.

I don't think so. They're just two different proxies for the same thing, i.e. the actual air density.

Octal450, 2025-04-19:
> That may be a bit of an exaggeration. You've only shared one graph depicting Thrust versus N1 over a range from close to idle to close to full N1 and as I pointed out when I looked at that graph and digitized it, for the data presented in the graph it's a very close match to 

Again, that's because you did so erroneously (no offense). The data there is only a subset of the actual full thrust curve over the idle-max N1 range and it doesn't doubt me that some areas of the curve may appear similarily to quadratic. I showed you the full graphs of each approach and illustrated the issues. Too much thrust due to much linearly in the high end and too flat of a curve at the beginning. I uploaded the numbers and graphs in Excel in the above replies.

I also used the EPR values (on non-GE engines, they do not use EPR so no data avail) as a secondary line of reasoning. Any combination of anything^2 will result in EPR curve of the wrong curvature.

If you try to justify any conclusion, I am sure you can find some set of solutions for it - you can justify nearly any claim by doing that. But look at the bigger picture and it starts to fall apart. 

> Not if it is a flat rated engine.
You're right, I poorly explained my point above. I meant at the same N1 values, the thrust would increase above rated.

My implementation isn't perfect, the thrust varies a little bit, I need to sort that out but that's secondary.

Also - don't focus on FADEC here. Assume this is a manual engine and pilot setting these N1 values manually.

I can tell you for a 100% fact that the N1 tables I am telling you are calibrated in terms of corrected altitude. Aka, not density altitude. That is why there is a TAT axis as well in addition to altitude. Same reason the MilThrust table has no TAT value, it's not needed as it's factored into density.

TBH, my concern here is, how do YOU want me to take N1 tables calibrated off corrected altitude and mach and TAT, and instruction JSBSim where MilThrust should occur? That's the feature I am asking you to add. I need the ability to tell JSBSim, hey, at *this* N1 value, you should be at MilThrust. All the other computations of this factor are airplane specific.

Then, I would like the thrust gradients to be fixed because they're objectively wrong as I explained. So that taxiing can occur at sensible levels of thrust and thrust reductions/derates no longer are overpowered when the correct fan speeds are used (from the manuals).

I've been working on this issue for a while now, and while I don't have written down every single source/document/graph that I've used. I am sorry but I do not keep records of every single finding I have! Maybe that's my failing.

But, the tables in the MilThrust came from a General Electric document on the CF6 engine to produce. So I don't fancy modifying those because they came directly from GE, describing the thrust gradients for density altitude and mach @milthrust. The difference here is I need to define *where* milthrust occurs, and I need a corrected fan speed to thrust gradient.

That is the goal of this feature request. I really don't want to have to make my own engine system to be able to simulate the engines properly... so please tell me what you need from me in order for you to justify this changes.

Kind Regards,
Josh

seanmcleod70, 2025-04-19:
> Again, that's because you did so erroneously (no offense).

Well you stated it was a **Correct table** and didn't mention that it wasn't showing the full range of N1 vs thrust. So if you withhold some relevant information is it really my error?

Having said that, I'm not sure what the provenance is of that particular graph. You generally have not provided links to 'official' looking documents. As @Zaretto also mentioned:

> is a better fit for various engines, without listing the engines and including references to the documents together with an explanation of how you came by this figure.
Without the valid data to backup your case this seems to be a case of changing one magic number for another.

When I did some internet searching for thrust vs N1 graphs I came across that exact graph image in some online forum, again with no link to the original source. Given the rendering etc. it's not clear to me that it wasn't generated by some flight sim enthusiast as opposed to coming from some 'official' source.

For example, that specific engine model is flat rated to 30C. The graph title mentions Sea level ISA which means sea level and 15C. As you've seen with the N1 take-off settings you use with your MD-11 that means the rated thrust of 27,300 would be at an N1 value a fair bit below the max N1 value for that engine of 104%. So even after correcting for the reduction based on Mach=0.3 in the graph it doesn't look like the graph would match the rated thrust. 

> I don't have written down every single source/document/graph that I've used. I am sorry but I do not keep records of every single finding I have! Maybe that's my failing.

You mention having a lot of data but as @Zaretto mentioned it's difficult for us to double-check if you don't provide references etc. as opposed to mostly providing 'confident/definitive' statements.

> I can tell you for a 100% fact that the N1 tables I am telling you are calibrated in terms of corrected altitude. Aka, not density altitude.

I never said that the data you supplied, e.g. the MD11 N1 settings for take-off thrust **weren't** based on pressure altitude and temperature. I mentioned a couple of times why I think they're presented in that format. In particular for the static thrust case for take-off where the engine will produce the rated thrust up to 30C at sea-level there is no difference between TAT and OAT given it's static thrust. Even at say on the order of 150KIAS around take-off speeds there will be a tiny difference between OAT and TAT.

>  Dry air behaves almost like an ideal gas, meaning it obeys the ideal gas law P/DT = R, where P is pressure, D is density, T is temperature, and R is a constant. 

So as I mentioned, given the pressure from the altitude in your N1 take-off tables plus the temperature you can trivially convert that to a density/density altitude. 

> so please tell me what you need from me in order for you to justify this changes.

I'll say it for the third time 😉 Present some basic code, even if bits are pseudo code.

As I mentioned above, this is effectively the code in JSBSim currently. Also not shown in the code, but as mentioned much earlier in my description of the JSBSim implementation the other assumption is that there is a linear mapping between throttle position and the N1 and N2 value. 

```python
rated_thrust = 60000

def calc_thrust(throttle_pos_norm, density_alt, mach):
      return throttle_pos_norm**2 * rated_thrust * DensityAltMachLookupScaleFactor(density_alt, mach)
```

When we have that then we can double-check it for any corner cases, whether it works for all the example data you've provided, if there any gotchas we may spot etc. 

For example, given the climb thrust table you've shared, what is the code that will produce the correct amount of thrust?

Let's say I'm at 5,000ft at Mach 0.4 TAT 0C and I move the thrust lever until I see an N1 of 99.11 as per the table. How does JSBSim calculate the correct thrust? What is the expected thrust for this particular combination?

seanmcleod70, 2025-04-19:
I've been going on about density, since I think it could simplify things potentially. So, in my mind at least to first order the thrust is proportional to the air density and the N1 fan speed, with that combination determining the airflow mass rate.

So I took the N1 take-off data Josh provided and calculated the density for each table entry based on the pressure altitude of the row and the OAT of the column. Then knowing the expected flat rate thrust for segment AB, 60,690lbf I took a look to see what sort of relationship might fit that portion of the data. Looking for something along these lines:

$$ Thrust = k \rho N1^x $$

I've done this so far for sea-level, 5,000ft and 10,000ft.

Here are the results for sea-level:

[image]

Which shows that the combination of an exponent of 1.67 and using $N1Norm^{1.67}$ results in matching the flat rated thrust of 60,690lbf within -46lbf to a +53lbf variation across the range from -60C to +30C.

The entries for +40C, +50C, +60C represent segment BC and indicate the thrust that would be produced using the scale factor k and the 1.67 exponent. It would be ideal if we could find some documented thrust values for this segment BC to see how closely they match.

For k, I used the average of the **N1 Norm Factor** column, i.e. 29,904,030 or normalized by the rated thrust, 492.73.

The results for the 5,000ft case were a variation ranging from -478lbf to +496lbf and for the 10,000ft case from -649lbf to +421lbf. So in both cases a variation slightly less than 1% of the rated thrust. There was an increase of 11% of the value of k from sea-level to 5,000ft and 13% from 5,000ft to 10,000ft. So, that would need to be factored in.

The data covers an N1 norm range from 75% to 98%. So it would be ideal to get official thrust figures covering the 0% to 75% N1 norm range to see how well this formula matches in that lower range.

In terms of using an exponent of 1.67 compared to 2, here is a graph showing the difference, green line being 1.67. 

[image]

So assuming this sort of formula did hold up, say within a fairly small % of actual thrust values across the whole envelope then as @Zaretto mentioned you could code this up within the **MilThrust** function without having to make any changes to the JSBSim source code.

It would calculate **N1Norm**, use `atmosphere/rho-slugs_ft3`, scale by Mach, use the normalized k (since JSBSim code multiplies the return from MilThrust function by rated thrust) and scale by a factor of N1Norm^1.67/N1Norm^2.

Zaretto, 2025-04-20:
So lets use so actual data to compare against real numbers.

The data is for the F100[1] and I'm comparing this against the current method and the proposed new method; that looks like this

[image]

Notes: 

1. I converted N1 to N2 using `(n1-n2min)*n2r` 
2. Thrust is already a percentage so I just divided this down and adjusted for sea level on the basis that 70% becomes 100%. 
3. I've attached the spreadsheet[2] in case I got the new formula wrong; it's in cells E25:E37
4. For the new method I've used the formula `FN idle+(N1 normalized ^ 3.5)*FN range` based on the description _"replacing N2 normalized first from min/max N2 to 0-1^2 with N1 %RPM ^3.5 then normalized to 0-1 where max N1^3.5 = 1 and idle N1^3.5 = 0."_

Figure 9 is from reference[1] is 
[image]

-----
[1]  NASA TP-1228, figure 9 "Corrected gross thrust as function of corrected fan speed with and without uniform inlet flow. Ram pressure ratio, 1.66; altitude, 7380 meters (24 200 ft). "
[2] [jsbsim-thrust-calc-n1-method-compared-current-and-F100-data.ods](https://github.com/user-attachments/files/19824617/jsbsim-thrust-calc-n1-method-compared-current-and-F100-data.ods)

Octal450, 2025-04-20:
Hi guys,
Firstly thanks again for your comments and time. I do really appreciate your inputs! I hope I am not coming across rudely.

@seanmcleod70 

> So if you withhold some relevant information is it really my error?

I didn't at first, true, but I did in this comment: [this](https://github.com/JSBSim-Team/jsbsim/issues/1249#issuecomment-2762240081)

> I should note that the graph image shown above for that engine starts at 35%, but the actual idle value for that engine is about 20% N1. Hence the curve does not have a domain of 0-1 when normalized.

> Having said that, I'm not sure what the provenance is of that particular graph

You're right of course, the graph I provided was created by a table. It is not original. Most of the information is in table form, which doesn't clearly illustrate the point, hence why I chose that one.

Of course, it is my failing that I've been less than optimal about sources. I tend to just save info after I've validated it into my giant folder for each engine/plane/etc.

> In terms of using an exponent of 1.67 compared to 2, here is a graph showing the difference, green line being 1.67.

Okay, but you're still making a rather drastic error here. The "actual" thrust from the engine starts at 0% compressor speed and up. The moment it starts turning, it's pushing some amount of air, no matter how miniscule. So that is why you *CANNOT* use normalized values from idle-max n1. But, since JSBSim needs MilThrust scalar as not including below Idle, since there is IdleThrust scalar, we need to add a little extra correction.

In addition, the gradient of the curve drastically changes based on the input number. That is the nature of exponential curves. Thus, given that the provided data as input in every situation I have provided is N1 in %, that is what the calculation/tables should use. 

I've put together another Microsoft Excel spreadsheet (attached) which shows this. You can see how I applied the scaling, correction etc.

I've included N1^2 here to indicate that lower exponentials produce very different behavior when N1% is input as opposed to normalized values. But notice how N1^3.5 increases more steadly at the low end and then is more steep at the higher end.

[image]

If you want, I can happily put together a bit of code that computes the thrust the way I want it to. 

Re climb:
It does not matter, as engines I've never seen having a rated "climb" thrust. I don't entirely know what goes into choosing those values. The only thing I know is there is a rated thrust and sometimes a few rated derates and sometimes rated MCT.

None of that is needed, you only need JSBSim to implemented rated thrust. The rest can be easily computed.

@Zaretto I must admit, I don't understand what is going on your graph, none of those lines make any sense. I can't open your ODS in Excel properly so I'll have to install OpenOffice (is that what you're using?). I'll get back to you on this, because I don't understand how any of your lines look like that.

Kind Regards,
Josh

[ThrustJSBSimStuff.xlsx](https://github.com/user-attachments/files/19824980/ThrustJSBSimStuff.xlsx)

Zaretto, 2025-04-20:
@Octal450 my plot compares the existing JSBSim against real data and your proposed changes.

> Issue 1: The thrust curvature is wrong. This is due to the erroneous norm^2 approach which matches nothing.

My plot shows that this isn't the case for the F100-PW-100 and that the equations in FGTurbine appear to work well for this engine.

> Issue 2: It is not possible to tell JSBSim at what N1 milthrust should occur at.

My view here is that the way we have that the maximum power is at the maximum RPM is fairly close to reality, what you're finding is a discrepancy in scaling from your data sources; i.e. what the aircraft calls 100% N1 isn't necessarily the same as the engineering value for maximum RPM but because you haven't posted your data sources, or even a detailed summary of workings it isn't possible for us to follow your workings.

Octal450, 2025-04-20:
I'll get back to you on Issue 1.

Yeah, I get that, but those curves look totally wrong. I was expecting to see an exponential curve here. But now I just noticed your X axis is not linear. So I'm going to have to download your spreadsheet and work with it. Given how significantly it diverges from my calculations, something is going on and I don't yet know what.

For a turbojet/turbofan engine, this is *never* true as far as I have ever seen.

It has nothing to do with 100% or higher/lower.

There seems to be at minimum 5% buffer of higher N1 allowable before maximum N1 occurs from rated N1. This is due to inconsistencies on how power scales for N1. If max (rated) power occurred at max N1, then you would not be able to set the N1 for all power settings.

Look in any FCTM for any aircraft that uses N1 (or N2) to set power, not EPR, and you'll find drastically varying N1 tables. Like the takeoff table in the MD-11 which you can find here (same as in the XML I linked). This image comes directly from a MD-11 FCTM.

[image]

Meanwhile, max N1 for the CF6-80C2D1F is according to the type certificate data sheet and confirmed by the FTCM, 117.5% N1.

[image]
[image]

You can see it's nowhere close. Highest rated N1 is about 114.

OK, lets check another random engine I have data for, the CFM56-5B4:

Here's the takeoff table:

[image]

And here's the maximum N1:

[image]

I've seen the exact same in every other turbofan engine that I have ever worked on.

So I am very sorry, but rated thrust does not occur near the maximum N1 hence this approach is wrong, and that's why we got all the complaints about underpowered airplanes when I implemented proper tables for takeoff thrust... and ever since applying my corrective logic that forces FGTurbine to apply rated N1 at the N1 that I tell it, it works correctly. And because this table is separate from the takeoff table, the behavior above the rated temp means the tables diverge, and thus the takeoff thrust reduces above the corner temp when set by the takeoff table, but otherwise is flat-rated.

Kind Regards,
Josh

Zaretto, 2025-04-20:
@Octal450 I think that the problem is related to using the data incorrectly that is causing your model to have the wrong figures.

Firstly here's some real data[1];  thrust for the CF6-50; which I believe is almost the same as the CF6-80

[image]

Notice that the chart goes up to 4200 RPM and that it has the maximum thrust at the maximum RPM. 

Therefore it is exactly a case that the data you refer to defines 100% N1 in a different way to the way JSBSim defines MaximumRPM and thus it is your `<MilThrust>` section that is wrong.

To fix your MilThrust you should adjust your data so that you are getting max thrust at 100% because that is what JSBSim requires. To do this you will need to correct the data from pressure altitude and temperature to instead use density altitude.

-----
[1] NASA CR-159699 "CF6 JET ENGINE PERFORMANCE IMPROVEMENT - NEW FAN"

Octal450, 2025-04-20:
OK,
Issue 1:
That graph makes no sense to me either. Every piece of literature I've seen is that there is an upward curving exponential on thrust. What is meant by "corrected" fan speed? Maybe they are applying some linearization or something.

Because what you're showing is a downward curving overall. If I tried to implement that then the performance would be totally and completely wrong. As I have it done now, it's pretty close to FCTM performance data, with just the few issues I mentioned.

Re the second part which is Issue 2:
Yes, that is right, if I *knew* what thrust was being computed by maxn1, I could do that. The issue is, in 99.9% of cases, I don't. I've never seen any literature on any of the engines I've worked on that tells what the thrust is at max n1 (+ corrections, density, mach, etc).

So I am forced to use the rated thrust and let the model define the rest. That is the purpose behind my feature request. So I don't have to copy and paste the same code into every single plane I develop in order to define rated thrust at rated RPM - which is the data I DO have.

I am not really sure why you are against adding this? If you have an engine where you have the data, then just don't use it and it will use default behavior where max = milthrust.

Kind Regards,
Josh

seanmcleod70, 2025-04-20:
Josh, so far the only data with some real provenance I've seen from what you've shared so far are the N1 take-off tables. And within the N1 take-off tables the only entries for which we know the exact thrust are those within the flat rated section before the OAT corner point. Given that subset of ground truth data I took a look at what sort of model could potentially match those entries, i.e. given (N1, density (pressure alt, OAT)).

For the entries in the N1 take-off tables that are after the corner point, I calculated the projected thrust using the model parameters, but as I said we don't have the ground truth for what the actual thrust values are, unless you have some document confirming the output thrust for these entries?

As I mentioned, the N1 tables don't cover the full range of N1 values for the engine. For the data I processed as I mentioned it covered an N1-norm range from 0.75 to 0.98. Do you agree that for the given ground truth data processed that the model I presented is a good fit?

I'm not making any claims about it's accuracy for the N1-norm range of 0.0 to 0.75 until I can get some truth thrust data for this range.

It's also useful that you've now also posted some snippets of the actual N1 takeoff tables. I had assumed that your FADEC N1 takeoff table in the MD11 was accurate, barring the odd typo, and it's what I used for my model fitting.

I noticed however that the table snippets you've now posted, for both the MD-11 and the CFM engine use OAT and not TAT that you have in your MD-11's takeoff table.

Can you post the CFM table that covers sea-level to 6,000ft, so that I can test out my model on it and to compare to the GE engine of the MD-11?

> If you want, I can happily put together a bit of code that computes the thrust the way I want it to.

Asking for the 4th time now 😉

Octal450, 2025-04-20:
Hi,
Sorry I don't understand this:

> As I mentioned, the N1 tables don't cover the full range of N1 values for the engine. For the data I processed as I mentioned it covered an N1-norm range from 0.75 to 0.98. Do you agree that for the given ground truth data processed that the model I presented is a good fit?

> Can you post the CFM table that covers sea-level to 6,000ft, so that I can test out my model on it and to compare to the GE engine of the MD-11?

Sure.

[image]

Well the issue is I don't have an easy reference to share you. That's why it's taken me years before suggesting such a change. I've had to connect different data points together to come to this conclusion. Unfortunately. But I do know that with the model, performance wise, everything has fallen into place. I tune my flight models without engine thrust, so I expect the engine to perform right when the correct data is used.

It also all lines up EPR wise.

> Asking for the 4th time now 😉
Yeah sorry, just I'm doing this juggling university work so that may take me a bit to do, cause I'll have to figure out all the factors to make sure I give you something helpful.

Kind Regards,
Josh

Octal450, 2025-04-20:
> For the data I processed as I mentioned it covered an N1-norm range from 0.75 to 0.98.

I don't understand this. These tables do not help in seeing the thrust factor vs fan speed in any way. That curve must be defined at a static altitude/temp/mach.

Kind Regards,
Josh

seanmcleod70, 2025-04-20:
> These tables do not help in seeing the thrust factor vs fan speed in any way. 

Why not, they're the only ground truth data that you've shared so far, with individual data points (below the OAT corner) for (N1, density (altitude, OAT), mach=0, thrust).

At the end of the day JSBSim needs a function that takes as input (throttle position, density, mach) and outputs a thrust value, and if we assume some mapping between throttle position and N1 then (N1, density, mach) -> thrust.

So why not use the ground truth data to see what sort of function mapping we can derive? There could be multiple ways of doing the mapping, so let's see how they compare against the data we have. Your code function would be one such mapping.

Octal450, 2025-04-20:
Because they do NOT tell you how the N1/N2 compare to thrust.

They only tell you how at which N1 the rated thrust occurs at for various alt/temp. So thus it is impossible to get thrust vs N1 data from them. Any attempt to use that data will result in wrong.

For each alt/mach/temp, there are different values. But these tables only provide ONE data point and thus cannot make a curve from him.

My code would not take this into account at all. I would choose an alt/mach/tat and then create the thrust factor based on that. And then it applies when that applies. Think of a plane on the ground, stopped, at 15C. It would give you the thrust for each N1 from idle through n1 max.

Kind Regards,
Josh

Octal450, 2025-04-20:
Btw the -5B4 is rated to 46C. I don't know the reason. But that's what it is, apparently.

seanmcleod70, 2025-04-20:
Ah, but if you have some physical model of the process, even if it's only tested on a very sparse set of data points it can potentially predict thrust values for the full range.

So in my case think of the input space as a 2-dimensional plane of (N1, density) (constant mach) and the height (z-coordinate) as the thrust value.

So yes, currently the only ground truth values I have all have the same thrust value for a sparse set of (N1, density) pairs. But given the assumption in the physical model it can predict thrust values for all the other (N1, density) pairs. In my spreadsheet example above you can see some of the predictions for some (N1, density) pairs after the OAT corner. What thrust values will your approach produce for those data points in comparison?

> Think of a plane on the ground, stopped, at 15C. It would give you the thrust for each N1 from idle through n1 max.

Sure, my function above does produce thrust for each N1 value. 

Question is how well it will do once we have more ground truth data to test it against.

> My code would not take this into account at all.

Sure, as I said, there are multiple ways to do this. Let's try them out and compare.

Octal450, 2025-04-20:
What do you mean "my function"? Which function?

And I am really sorry, but this table only shows what N1 is rated at various alt/temp.

It doesn't show anything else. It doesn't show how N1 and thrust vary due to differential rotor speed aka throttle position.

It is important to understand what the data is telling you.

Kind Regards,
Josh

seanmcleod70, 2025-04-21:
> What do you mean "my function"? Which function?

The one I described above.

$$ Thrust = k \rho N1^x $$

> And I am really sorry, but this table only shows what N1 is rated at various alt/temp.

Don't be sorry. Correct, as I've mentioned, I know that. In fact, only a portion of the N1 take-off table has entries that output the rated thrust, i.e. the portion within the flat rated section. The entries after the OAT corner point do not tell us what the ground truth thrust value is for those entries. Like I've mentioned a couple of times.

> It doesn't show anything else. It doesn't show how N1 and thrust vary due to differential rotor speed aka throttle position, only how the air density affects thrust. That's not at issue here.

Not completely, what you're missing is that there is **some** useful N1 knowledge baked into the tables.

(N1, density (alt, OAT), rated thrust)

(94.9, 0.0032136, 60,690)
(111.7, 0.0022596, 60,690)

So, yes, there aren't a range of different N1 values for the same density producing different thrust, but given some first order physics knowledge my hypothesis is that if we can generate/learn a function that matches (N1, density, 60,690) for a number of ground truth points varying in N1 and density, that it **could potentially** be extrapolated over the full range of (N1, density). Now maybe it won't end up being a good match at all, or maybe it will end up being a fairly decent match for some or all of the operating envelope.  

So for example:

[image]

Now if/when we can get hold of some additional ground truth data points we can compare how well my function does. I'm not claiming that it is definitely a good match. 

> It is important to understand what the data is telling you.

Sure, and if you read my replies more closely you should've picked up that I mentioned a couple of times what the data in the N1 take-off tables tells us.

You seem to immediately pick up some particular part of a comment and focus so much on discrediting that particular part that you don't seem to really comprehend all the other parts. Sure, some parts may not be completely valid, but it appears that you are so set on achieving the goal using your particular approach that you're not open to alternatives, that sure may involve a bit more work in terms of taking some of the existing data and processing it etc. as opposed to using it pretty much "as-is" with the approach you have in mind.

seanmcleod70, 2025-04-21:
Using my function to plot the static thrust curves for varying OAT at sea-level. As noted, the only ground-truth data points are those shown on the graph from the N1 take-off tables.

[image]

seanmcleod70, 2025-04-21:
I've added in the 3 post OAT corner points at sea-level  (40C, 50C, 60C). I haven't labelled these as 'ground-truth' since the N1 take-off tables don't specify the thrust values for these points. The N1 value is the ground-truth, but the thrust value is based on my formula.

[image]