JSBSim-Team/jsbsim #1257: Build of JSBSimForUnreal.sln always fails.

SLGY, 2025-04-10:
**I'm submitting a ...**
  - [X] bug report
  - [ ] feature request
  - [ ] support request => **Please do not submit support request here**, see note at the top of this template.

**Describe the issue**
In a fresh download and extraction of JSBSim, attempting to build JSBSimForUnreal.sln fails in Visual Studio Community 2022.

**What is the current behavior?**
Previously I have been able to build JSBSimForUnreal.sln. But no matter the options I select when first opening the solution (see below pics), the build fails.

[image]

[image]

[image]

And the build always gives this error at the end:

[image]

I am following the directions in the Unreal Engine folder readme closely and here is my installed modules:

[image]

**What is the expected behavior?**
I expect to see "1 succeeded, 0 failed, x skipped" and able to launch the UEReferenceApp.uproject file, but currently unable to.

**What is the motivation / use case for changing the behavior?**
N/A

**Please tell us about your environment:**
Windows 10 (latest updates), latest JSBSim

**Other information**
This problem appears to have coincided with Windows 10 updates ominously mentioning .NET framework updates (March 25, 2025-KB5055063 Cumulative Update for .NET Framework 3.5, 4.8 and 4.8.1 for Windows 10 Version 22H2) ... Could this be related?

bcoconni, 2025-04-10:
The PR for enabling C++17 during the compilation of the UE plugin has just been merged.
Could you please try pulling the last version of the `master` branch and retry ?

SLGY, 2025-04-10:
> The PR for enabling C++17 during the compilation of the UE plugin has just been merged. Could you please try pulling the last version of the `master` branch and retry ?

That's fixed it, thankyou!

closed 2025-04-10 by SLGY