JSBSim-Team/jsbsim #1268: Unreal Plugin: Ground Reactions on Cesium World Terrain

Scavanger, 2025-04-23:
JSBSim/Unreal Plgin: latest from Source
Unreal Engine: 5.5.4

I'm trying to get JSBSim running with Unreal 5.5.4 and JSBSim, the reference app is working so far. 

But apparently ground reactions do not work on the generated “Cesium World Terrain”.  When the airplane is taxiing on a static mesh (like the runway in the reference app), everything is fine. Flight also worked without any problems. 

But as soon as the plane is on the ground and on the “Cesium World Terrain” the physics flips out completely, usually the plane just falls through the ground, “Airplane crashed”, in my own app with lighter planes it can happen that the plane bounces around like crazy (height a few hundred meters) and at some point falls through the ground or even drives in any direction.

Reproduce in Unreal Referenece app:
Delete the default terrain, activate Cesium plugin, add “Cesium World Terrain and Bing maps”, calibrate Cesium Georefence with Unreal's Georefence (Origin locaction) Adjust runway/airplane altitude.
Start simuation, Taxi, leave runway -> "Airplane crashed"

Just a setting I forget or a Bug?

bcoconni, 2025-04-23:
@gallonmate , any idea?

gallonmate, 2025-04-23:
No idea.
@demonixis Did Cesium terrain collision work with Unreal JSBSim?

There are other users who have used Cesium with JSBSim, but I don't see any references that the Cesium collision with Unreal JSBSim ever worked by default. I read one comment from 2023 that said Cesium collision caused airplane crash.

To see where the issue might be, here's a general way to debug the airplane crash using Visual Studio: https://github.com/JSBSim-Team/jsbsim/issues/1042#issuecomment-1960105583

demonixis, 2025-04-23:
Collisions are working well. You've to be sure to use the GeoReference actor with the same parameters than the one from Cesium. 

Also I've my own JSBSim fork with some changes about collisions. But I don't think you need this to make it works.

Scavanger, 2025-04-23:
I found the solution myself:

"Tile Culling" (Outliner/Cesium World Terrain/TileCulling) must be deactivated, otherwise tiles are unloaded outside the field of view and the collision mesh is also "gone".
Activated tile culling leads to the funny effect if you use the external camera it works first, if you switch to the internal camera the tiles behind the camera are unloaded and the airplane falls through the ground.

Another tip for those who don't want to use a static mesh as a starting point: Since the cesium terrain is streamed and therefore loads with a delay, the airplane falls through the ground right at the beginning (since it's not there yet), so place the airplane on an (invisible) static mesh.

closed 2025-04-24 by Scavanger