JSBSim-Team/jsbsim #1279: JSBSim VTOL/thrust vectoring feature

DuanzzhiWang, 2025-05-18:
Recently I was working on VTOL feature for V-22 Osprey in jsbsim, and want to test it out inside Unreal Engine. 
From FGThruster.cpp, I assume this is the part that supports thrust vectoring but I didn't see any existing aircraft implenting it:

[image]

therefore, I add a line under FGThruster.h  to calculate the thrust vectoring angle:

[image]

for the aircraft script file, I also delete the orientation part, since the thrust is no longer a fixed value:

[image]

and added a new channel called thrust vectoring under the FCS part：

[image]

with these changes, UnrealEngine crashes with following report：

[image]

So Is there anything I missed while working on the VTOL function? Need some advice plsss

seanmcleod70, 2025-05-18:
There should be no need to modify the JSBSim source code to implement thrust vectoring.

Here is a simple example of thrust vectoring;

```python
import jsbsim
import math

AIRCRAFT_NAME="737"
# Path to JSBSim files, location of the folders "aircraft", "engines" and "systems"
PATH_TO_JSBSIM_FILES="../.."

# Avoid flooding the console with log messages
#jsbsim.FGJSBBase().debug_lvl = 0

fdm = jsbsim.FGFDMExec(PATH_TO_JSBSIM_FILES)

# Load the aircraft model
fdm.load_model(AIRCRAFT_NAME)

# Set engines running
fdm['propulsion/set-running'] = -1

# Initial conditions
fdm['ic/h-sl-ft'] = 1000
fdm['ic/vc-kts'] = 250
fdm['ic/gamma-deg'] = 0

# Initialize the aircraft with initial conditions
fdm.run_ic() 

# Trim
try:
    fdm['simulation/do_simple_trim'] = 1

except jsbsim.TrimFailureError:
    print("Trim failed....")
    pass  # Ignore trim failure

print('')

print(f'fbx: {fdm["forces/fbx-prop-lbs"]}  fby: {fdm["forces/fby-prop-lbs"]}  fbz: {fdm["forces/fbz-prop-lbs"]}')
print(f'Engine: {fdm["propulsion/engine[0]/thrust-lbs"]*2}')
print(f'Thruster pitch: {fdm["propulsion/engine[0]/pitch-angle-rad"]}')

# Modify thruster pitch angle for some thrust vectoring
fdm["propulsion/engine[0]/pitch-angle-rad"] = math.radians(30)
fdm["propulsion/engine[1]/pitch-angle-rad"] = math.radians(30)

fdm.run()

print('')

print(f'fbx: {fdm["forces/fbx-prop-lbs"]}  fby: {fdm["forces/fby-prop-lbs"]}  fbz: {fdm["forces/fbz-prop-lbs"]}')
print(f'Engine: {fdm["propulsion/engine[0]/thrust-lbs"]*2}')
print(f'Thruster pitch: {math.degrees(fdm["propulsion/engine[0]/pitch-angle-rad"]):.1f}')
```

With the following output:

```sh
End of vehicle configuration loading.
-------------------------------------------------------------------------------
  Full Trim

  Trim successful
  Trim Results:
       Angle of Attack:   3.17  wdot:  5.47e-05 Tolerance: 1e-03  Passed
              Throttle:   0.61  udot: -5.10e-05 Tolerance: 1e-03  Passed
            Pitch Trim:  -0.20  qdot: -7.49e-11 Tolerance: 1e-04  Passed
            Roll Angle:  -0.00  vdot: -3.63e-14 Tolerance: 1e-03  Passed
              Ailerons:   0.00  pdot:  2.98e-16 Tolerance: 1e-04  Passed
                Rudder:   0.00  rdot: -9.75e-32 Tolerance: 1e-04  Passed

fbx: 12859.608467405447  fby: 0.0  fbz: 0.0
Engine: 12859.608467405447
Thruster pitch: 0.0

fbx: 11136.747616295526  fby: 0.0  fbz: -6429.8042341651435
Engine: 12859.608468330289
Thruster pitch: 30.0
```

seanmcleod70, 2025-05-18:
Depending on the level of fidelity you're looking to achieve you may also want to model a reduction in thrust based on the angle of the thrust vectoring nozzle. May or may not be relevant for the V-22 Osprey.

seanmcleod70, 2025-05-18:
I came across this NASA report - [ Optimal Pitch Thrust-Vector Angle and Benefits for all Flight Regimes](https://ntrs.nasa.gov/api/citations/20000034897/downloads/20000034897.pdf)

Here are their results for cruise for a generic widebody airliner.

[image]

So I couldn't resist testing this out in JSBSim, using the 737 model in the repo.

[image]

Which shows very similar results. In fact the overall thrust reduction is almost identical at 40lbf.

Optimisation, although this was simply a brute force approach, is finding the lowest thrust value in order to minimize fuel burn for a given altitude and speed. Trade-offs being that a greater thrust angle means less lift needed from the wings, so for the same airspeed you can fly at a smaller AoA which reduces drag, smaller AoA also means you need a smaller moment from the horizontal stabilizer which means less negative lift from the stabilizer and less drag from the stabilizer. But at some point as the thrust vector angle increases you get to a point where there isn't enough horizontal thrust to match the drag, so overall thrust needs to increase etc.

Not sure what is happening with the slight discontinuity around 1.8deg thrust vector angle.

DuanzzhiWang, 2025-05-18:
> I came across this NASA report - [ Optimal Pitch Thrust-Vector Angle and Benefits for all Flight Regimes](https://ntrs.nasa.gov/api/citations/20000034897/downloads/20000034897.pdf)
>
> Here are their [...]

thank you for the info! maybe I didn't think as far as thrust reduction yet (gonna do that with f-35 later). I am right now trying to visualize it using the unreal engine plugin, where the plugin (or jsbsimsimmovementcomponent.cpp) read the data from aircraft.xml. the part I am struggling with is that inside aircraft.xml file, the thrust orientation is always fixed, and I tried to implement a control axis that controls the direction of the thrust. I don't know did you try the thrust vectoring B737 script inside Unreal Engine? I really appreciate some advice on that. Thank you!!!

seanmcleod70, 2025-05-18:
I was trying to point out with the 737 example that all I needed to do in order to get thrust vectoring to work, with no changes what so ever to the 737.xml was to simply update the `pitch-angle-rad` property of the thruster.

> Modify thruster pitch angle for some thrust vectoring
fdm["propulsion/engine[0]/pitch-angle-rad"] = math.radians(30)
fdm["propulsion/engine[1]/pitch-angle-rad"] = math.radians(30)

I really doubt that the Unreal plug-in is updating/overwriting the `pitch-angle-rad` property each time during the simulation loop.

bcoconni, 2025-07-26:
Since this topic has not seen any progress in months and it is not addressing either a bug report or a feature request, it is moved to discussions.