JSBSim-Team/jsbsim #1281: Building python module from source

Dobid, 2025-05-19:
If you need help or want to ask questions, please use [Discussions](https://github.com/JSBSim-Team/jsbsim/discussions).

This repository's issues are reserved for feature requests and bug reports. Issues that are neither bug reports or feature requests will be transferred to [Discussions](https://github.com/JSBSim-Team/jsbsim/discussions) by the team and the issue will be locked and closed.

***

**I'm submitting a ...**
  - [x] bug report
  - [ ] feature request
  - [ ] support request => **Please do not submit support request here**, see note at the top of this template.

**Describe the issue**
I'm trying to build the jsbsim python module from source and following the current developer docs.
- I cloned the jsbsim repository `git clone https://github.com/JSBSim-Team/jsbsim.git jsbsim-code`
- Created the build dir `mkdir build && cd build`
- then ``cmake -DINSTALL_JSBSIM_PYTHON_MODULE=ON ..``
- and ``make -j4``
But when I do : ``python python/setup.py install --skip-build``, I have this error:
``
Traceback (most recent call last):
  File "/home/david/sandbox/jsbsim/jsbsim/build/python/setup.py", line 155, in <module>
    for d in os.scandir('jsbsim/aircraft'):
FileNotFoundError: [Errno 2] No such file or directory: 'jsbsim/aircraft'
``
I've tried multiple things like changing the directory in the setup.py script. I've managed to circumvent this error last month, but now I can't find what I did to successfully install the jsbsim module.

**Please tell us about your environment:**
 Python 3.9

bcoconni, 2025-05-22:
Direct calls to `setup.py` are now discouraged (see [Why you shouldn't invoke setup.py directly](https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html) or ['setup.py install is deprecated' warning shows up every time I open a terminal in VSCode](https://stackoverflow.com/questions/73257839/setup-py-install-is-deprecated-warning-shows-up-every-time-i-open-a-terminal-i)). The correct way to proceed is to use the `build`package:
```bash
> mkdir build && cd build
> cmake -DINSTALL_JSBSIM_PYTHON_MODULE=ON ..
> make -j4
> cd python
> pip install build
> python -m build --wheel
> pip install jsbsim --no-index -f dist
```
Note that:
1. The command `python -m build --wheel` is calling `setup.py` under the hood. `setup.py` will automatically detect that the library has already been built and should only compile and link `_jsbsim.cxx` (which is the code generated by Cython).
2. The wheel package will be built in the directory `build/python/dist` hence the flag `-f dist` that is passed to the last `pip install` command.

Dobid, 2025-05-23:
Ok I understood. Thank you for your reply! Then I would kindly advise to update the [Developers Docs](https://github.com/JSBSim-Team/jsbsim/blob/master/doc/DevelopersDocs.md#installing-the-python-module), so newcomers don't run into the same issue.

bcoconni, 2025-05-24:
> Then I would kindly advise to update the [Developers Docs](https://github.com/JSBSim-Team/jsbsim/blob/master/doc/DevelopersDocs.md#installing-the-python-module), so newcomers don't run into the same issue.

Oops ! You are right, our docs are outdated. I have submitted the PR #1286 to fix that.
Thanks for reporting this error.

bcoconni, 2025-05-24:
The pull request #1286 has just been merged and the docs are now updated.

Dobid, 2025-05-24:
Great ! Thank you !

closed 2025-05-24 by Dobid