JSBSim-Team/jsbsim #1300: Unexpected crash after ~70 minutes of flight

alexisnov, 2025-07-03:
**I'm submitting a ...**

- [x] bug report

- [ ] feature request

- [ ] support request => Please do not submit support request here, see note at the top of this template.

**Describe the issue**
The JSBSim simulation unexpectedly chrashes after ~70 minute flight. When run with the "Rascal" aircraft model via ArduPilot SITL, repeatedly logs *CRASH DETECTED* messages for all landing gear components (LEFT_MLG, RIGHT_MLG, TAIL_LG) for an extended period of time after an initial impact. The messages are logged in rapid succession, suggesting an issue with how the crash event is processed or reported.

**What is the current behavior?**
The simulation runs normally for a significant duration (over 4300 simulated seconds), including a successful takeoff. However, at approximately 4393.24 seconds, the aircraft crashes. Following this crash, the log is flooded with repetitive *CRASH DETECTED* messages, with each landing gear component being reported as crashed multiple times within milliseconds of each other. This excessive logging makes it difficult to ascertain the exact point of impact or to analyze subsequent simulation state.

**Steps to reproduce:**

Navigate to your ArduPilot SITL directory.

Run the following command:
```

./build/sitl/bin/arduplane -w -S --model jsbsim:Rascal  --defaults ./Tools/autotest/default_params/plane-jsbsim.parm
```
Observe the console output after the aircraft has been flying for some time (e.g., around 4393 simulated seconds). The repetitive crash messages will appear.

**What is the expected behavior?**
Flying without unexpected crashes for a long period of time.

**Please tell us about your environment:**

OS: Ubuntu 22.04, WSL, 5.15.167.4-microsoft-standard-WSL2

JSBSim version: v1.2.1.dev1 May 6 2025 20:49:28

**Console log example**
[ArdupilotLogCrashAfterHour.txt](https://github.com/user-attachments/files/21038526/ArdupilotLogCrashAfterHour.txt)

alexisnov, 2025-07-03:
Thank you @seanmcleod70 ! 
There are following important aspects of the flight in question: the flight follows a trajectory that eliminates any possibility of a collision with the ground, as well as any reasonable explanations for the UAV's crash.  However, the UAV consistently crashes after approximately 70 minutes of flight, which should not be the case.

seanmcleod70, 2025-07-04:
> the flight follows a trajectory that **eliminates any possibility** of a collision with the ground,

I did consider asking if you wanted to wager a bet on that statement 😉 

So just before 4300s someone is setting all the control inputs (throttle included) to 0.

closed 2025-07-04 by seanmcleod70